espressif/arduino-esp32 #10784: opi psram: PSRAM ID read error: 0x00000008, PSRAM chip not found or not supported, or wrong PSRAM line mode

lmabbe-lm, 2024-12-26:
### Board

esp32-s3-wroom-1u-n16r8

### Device Description

Use ov7670 camera to connect to the motherboard

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

clion + platform io

### Operating System

macos 

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

pio config
---
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
upload_speed = 115200
monitor_speed = 115200

board_build.arduino.memory_type = qio_opi
board_build.flash_mode = qio
board_build.psram_type = opi
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.extra_flags =
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
build_flags =
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue

### Sketch

```cpp
#include <WiFi.h>
#include <Arduino.h>
#include "esp_camera.h"

// 摄像头引脚配置
#define CAM_PIN_PWDN    -1
#define CAM_PIN_RESET   5

#define CAM_PIN_SIOD    36
#define CAM_PIN_SIOC    8

#define CAM_PIN_VSYNC   18
#define CAM_PIN_HREF    37

#define CAM_PIN_XCLK    38
#define CAM_PIN_PCLK    17

#define CAM_PIN_D7      16
#define CAM_PIN_D6      39
#define CAM_PIN_D5      15
#define CAM_PIN_D4      40
#define CAM_PIN_D3      7
#define CAM_PIN_D2      41
#define CAM_PIN_D1      6
#define CAM_PIN_D0      42

void setup() {
    Serial.begin(115200);
    while (!Serial) delay(100); // 等待串口连接

    delay(5000);

    Serial.println("ESP32-S3 is running...");

    // 连接WiFi
    WiFi.begin("lmabbe", "hello_world");
    Serial.print("Connecting to WiFi");
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println("\nWiFi connected!");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());

    // 摄像头初始化
    camera_config_t camera_config = {
        .pin_pwdn = CAM_PIN_PWDN,
        .pin_reset = CAM_PIN_RESET,
        .pin_xclk = CAM_PIN_XCLK,
        .pin_sccb_sda = CAM_PIN_SIOD,
        .pin_sccb_scl = CAM_PIN_SIOC,
        .pin_d7 = CAM_PIN_D7,
        .pin_d6 = CAM_PIN_D6,
        .pin_d5 = CAM_PIN_D5,
        .pin_d4 = CAM_PIN_D4,
        .pin_d3 = CAM_PIN_D3,
        .pin_d2 = CAM_PIN_D2,
        .pin_d1 = CAM_PIN_D1,
        .pin_d0 = CAM_PIN_D0,
        .pin_vsync = CAM_PIN_VSYNC,
        .pin_href = CAM_PIN_HREF,
        .pin_pclk = CAM_PIN_PCLK,
        .xclk_freq_hz = 20000000,
        .ledc_timer = LEDC_TIMER_0,
        .ledc_channel = LEDC_CHANNEL_0,
        .pixel_format = PIXFORMAT_GRAYSCALE,
        .frame_size = FRAMESIZE_QVGA,
        .jpeg_quality = 12,
        .fb_count = 1,
        .fb_location = CAMERA_FB_IN_PSRAM,
        .grab_mode = CAMERA_GRAB_WHEN_EMPTY
    };

    esp_err_t err = esp_camera_init(&camera_config);
    if (err != ESP_OK) {
        Serial.println("Camera initialization failed");
        return; // 如果摄像头初始化失败，退出setup
    }
    Serial.println("Camera initialized successfully");
}

void loop() {
    Serial.println("Looping...");
    delay(1000); // 每秒打印一次
}
```

### Debug Message

```plain
/Users/lmabbe/.platformio/penv/bin/pio run -t upload -t monitor -e esp32-s3-devkitc-1
Processing esp32-s3-devkitc-1 (platform: espressif32; board: esp32-s3-devkitc-1; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.9.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio/build/esp32-s3-devkitc-1/src/main.cpp.o
Linking .pio/build/esp32-s3-devkitc-1/firmware.elf
Retrieving maximum program size .pio/build/esp32-s3-devkitc-1/firmware.elf
Checking size .pio/build/esp32-s3-devkitc-1/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  14.5% (used 47504 bytes from 327680 bytes)
Flash: [==        ]  22.2% (used 740641 bytes from 3342336 bytes)
Building .pio/build/esp32-s3-devkitc-1/firmware.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.wchusbserial210
Uploading .pio/build/esp32-s3-devkitc-1/firmware.bin
esptool.py v4.5.1
Serial port /dev/cu.wchusbserial210
Connecting....
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: cc:ba:97:05:04:b8
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000c4fff...
Compressed 15104 bytes to 10429...
Writing at 0x00000000... (100 %)
Wrote 15104 bytes (10429 compressed) at 0x00000000 in 1.1 seconds (effective 105.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 400.6 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 834.7 kbit/s)...
Hash of data verified.
Compressed 741008 bytes to 484278...
Writing at 0x00010000... (3 %)
Writing at 0x0001c14f... (6 %)
Writing at 0x00025177... (10 %)
Writing at 0x00031214... (13 %)
Writing at 0x00036ea5... (16 %)
Writing at 0x0003c8bf... (20 %)
Writing at 0x00041f15... (23 %)
Writing at 0x00047068... (26 %)
Writing at 0x0004c16d... (30 %)
Writing at 0x00050fe7... (33 %)
Writing at 0x00055f08... (36 %)
Writing at 0x0005ae53... (40 %)
Writing at 0x0006022b... (43 %)
Writing at 0x00066aa2... (46 %)
Writing at 0x0006bbfb... (50 %)
Writing at 0x00070e8d... (53 %)
Writing at 0x00076bd6... (56 %)
Writing at 0x0007b914... (60 %)
Writing at 0x000808d3... (63 %)
Writing at 0x00085925... (66 %)
Writing at 0x0008ad75... (70 %)
Writing at 0x000901ea... (73 %)
Writing at 0x00095868... (76 %)
Writing at 0x0009b239... (80 %)
Writing at 0x000a091a... (83 %)
Writing at 0x000a8e1d... (86 %)
Writing at 0x000b1182... (90 %)
Writing at 0x000b65c5... (93 %)
Writing at 0x000bc0ef... (96 %)
Writing at 0x000c1c32... (100 %)
Wrote 741008 bytes (484278 compressed) at 0x00010000 in 42.6 seconds (effective 139.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
--- Terminal on /dev/cu.wchusbserial210 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP32-S3 is running...
Connecting to WiFi.
WiFi connected!
IP address: 192.168.0.111
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403773f9
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
E (157) opi psram: PSRAM ID read error: 0x00000008, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (158) spiram: SPI RAM enabled but initialization failed. Bailing out.
ESP32-S3 is running...
Connecting to WiFi.
WiFi connected!
IP address: 192.168.0.111
E (5702) cam_hal: cam_dma_config(301): frame buffer malloc failed
E (5703) cam_hal: cam_config(385): cam_dma_config failed
E (5703) gdma: gdma_disconnect(299): no peripheral is connected to the channel
E (5710) camera: Camera config failed with error 0xffffffff
Camera initialization failed
Looping...
Looping...
Looping...
Looping...
Looping...
Looping...
Looping...
Looping...
Looping...
Looping...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lmabbe-lm, 2024-12-26:
**Update the log and code and print the psram size. If you see it has been printed in the log, it means that psram has been registered successfully.**

**the code**
--- 

#include <WiFi.h>
#include <Arduino.h>
#include "esp_camera.h"

// 摄像头引脚配置
#define CAM_PIN_PWDN    -1
#define CAM_PIN_RESET   5

#define CAM_PIN_SIOD    36
#define CAM_PIN_SIOC    8

#define CAM_PIN_VSYNC   18
#define CAM_PIN_HREF    37

#define CAM_PIN_XCLK    38
#define CAM_PIN_PCLK    17

#define CAM_PIN_D7      16
#define CAM_PIN_D6      39
#define CAM_PIN_D5      15
#define CAM_PIN_D4      40
#define CAM_PIN_D3      7
#define CAM_PIN_D2      41
#define CAM_PIN_D1      6
#define CAM_PIN_D0      42

void setup() {
    Serial.begin(115200);
    while (!Serial) delay(100); // 等待串口连接

    delay(5000);

    Serial.println("ESP32-S3 is running...");

    // 连接WiFi
    WiFi.begin("lmabbe", "hello_world");
    Serial.print("Connecting to WiFi");
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println("\nWiFi connected!");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());

    Serial.printf("getPsramSize: %d \n",ESP.getPsramSize());

    Serial.printf("getFreePsram: %d \n",ESP.getFreePsram());

    // 摄像头初始化
    camera_config_t camera_config = {
        .pin_pwdn = CAM_PIN_PWDN,
        .pin_reset = CAM_PIN_RESET,
        .pin_xclk = CAM_PIN_XCLK,
        .pin_sccb_sda = CAM_PIN_SIOD,
        .pin_sccb_scl = CAM_PIN_SIOC,
        .pin_d7 = CAM_PIN_D7,
        .pin_d6 = CAM_PIN_D6,
        .pin_d5 = CAM_PIN_D5,
        .pin_d4 = CAM_PIN_D4,
        .pin_d3 = CAM_PIN_D3,
        .pin_d2 = CAM_PIN_D2,
        .pin_d1 = CAM_PIN_D1,
        .pin_d0 = CAM_PIN_D0,
        .pin_vsync = CAM_PIN_VSYNC,
[... 145 lines cut ...]
rst:0x8 (TG1WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403773f9
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
E (158) opi psram: PSRAM ID read error: 0x00000008, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (158) spiram: SPI RAM enabled but initialization failed. Bailing out.
ESP32-S3 is running...
Connecting to WiFi.
WiFi connected!
IP address: 192.168.0.111
getPsramSize: 0 
getFreePsram: 0 
E (5727) cam_hal: cam_dma_config(301): frame buffer malloc failed
E (5728) cam_hal: cam_config(385): cam_dma_config failed
E (5728) gdma: gdma_disconnect(299): no peripheral is connected to the channel
E (5735) camera: Camera config failed with error 0xffffffff
Camera initialization failed
Looping...
Looping...
Looping...
Looping...

me-no-dev, 2024-12-26:
This is an unexpected error, especially given that OPI PSRAM is initialized in the bootloader by ESP-IDF. There are a few issues with this submission though:
- You are using PlatformIO, which we do not support ourselves.
- The version of the core that is provided by PlatformIO is now very old and not supported anymore.
- N16R8 should be OPI-OPI module (both Flash ad PSRAM are OPI) if it is supplied by Espressif

I suggest that you have a look at `pioarduino` to get newer version of the core through the PlatformIO plugin.

Use this in your config for v3.0.7
```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.07/platform-espressif32.zip
```

Or this for v3.1.0
```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10/platform-espressif32.zip
```

Try with those and see if the behavior changes

SuGlider, 2024-12-27:
@Jason2866 - PTAL. Thanks!

Jason2866, 2024-12-27:
@lmabbe-lm   The Platformio config is not correct. Can you link the board you are using?
With the boards info, we can make a valid platformio board config.

Jason2866, 2024-12-28:
Here is a great overview for the espressif32 modules how to setup with Platformio 

https://github.com/sivar2311/ESP32-S3-PlatformIO-Flash-and-PSRAM-configurations

closed 2025-01-06 by rftafas