espressif/arduino-esp32 #10787: Failed to load code into Esp32P4

Vineetha1705, 2024-12-28:
### Board

Esp32p4

### Device Description

ESP32-P4-Function_EV_Board

### Hardware Configuration

USB cable is connected to USB-UART port.

### Version

latest development Release Candidate (RC-X)

### IDE Name

Espressif IDE

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I have started working with the ESP32P4-Function-EV Board, and the IDE being used is Espressif IDE (v5.3.1). I successfully created and built the project, but when I attempted to flash the code into the ESP32P4 device, it displayed the error.

### Sketch

```cpp
/*
 * SPDX-FileCopyrightText: 2010-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

#include <stdio.h>
#include <inttypes.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_chip_info.h"
#include "esp_flash.h"
#include "esp_system.h"

void app_main(void)
{
    printf("Hello world!\n");

    /* Print chip information */
    esp_chip_info_t chip_info;
    uint32_t flash_size;
    esp_chip_info(&chip_info);
    printf("This is %s chip with %d CPU core(s), %s%s%s%s, ",
           CONFIG_IDF_TARGET,
           chip_info.cores,
           (chip_info.features & CHIP_FEATURE_WIFI_BGN) ? "WiFi/" : "",
           (chip_info.features & CHIP_FEATURE_BT) ? "BT" : "",
           (chip_info.features & CHIP_FEATURE_BLE) ? "BLE" : "",
           (chip_info.features & CHIP_FEATURE_IEEE802154) ? ", 802.15.4 (Zigbee/Thread)" : "");

    unsigned major_rev = chip_info.revision / 100;
    unsigned minor_rev = chip_info.revision % 100;
    printf("silicon revision v%d.%d, ", major_rev, minor_rev);
    if(esp_flash_get_size(NULL, &flash_size) != ESP_OK) {
        printf("Get flash size failed");
        return;
    }

    printf("%" PRIu32 "MB %s flash\n", flash_size / (uint32_t)(1024 * 1024),
           (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");

    printf("Minimum free heap size: %" PRIu32 " bytes\n", esp_get_minimum_free_heap_size());

    for (int i = 10; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
}
```

### Debug Message

```plain
Executing action: flash
Running ninja in directory D:\Vineetha\Risc-V\Esp32p4\Software\Esp32p4_Test\hello_world\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D D:\Vineetha\Risc-V\Esp32p4\Software\Esp32p4_Test\hello_world\build\esp-idf\esptool_py && C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v5.3.1/components/partition_table/check_sizes.py --offset 0x8000 partition --type app D:/Vineetha/Risc-V/Esp32p4/Software/Esp32p4_Test/hello_world/build/partition_table/partition-table.bin D:/Vineetha/Risc-V/Esp32p4/Software/Esp32p4_Test/hello_world/build/hello_world.bin"
hello_world.bin binary size 0x2fec0 bytes. Smallest app partition is 0x100000 bytes. 0xd0140 bytes (81%) free.

[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D D:\Vineetha\Risc-V\Esp32p4\Software\Esp32p4_Test\hello_world\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v5.3.1/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x2000 D:/Vineetha/Risc-V/Esp32p4/Software/Esp32p4_Test/hello_world/build/bootloader/bootloader.bin"

Bootloader binary size 0x5450 bytes. 0xbb0 bytes (12%) free.

[2/3] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v5.3.1\components\esptool_py && C:\Espressif\tools\cmake\3.24.0\bin\cmake.exe -D IDF_PATH=C:/Espressif/frameworks/esp-idf-v5.3.1 -D SERIAL_TOOL=C:/Espressif/python_env/idf5.3_py3.11_env/Scripts/python.exe;;C:/Espressif/frameworks/esp-idf-v5.3.1/components/esptool_py/esptool/esptool.py;--chip;esp32p4 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_argsesptool.py --chip esp32p4 -p COM25 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x2000 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin

esptool.py v4.8.1

Serial port COM25

Connecting.....

Chip is ESP32-P4 (revision v1.0)

Features: High-Performance MCU

Crystal is 40MHz

MAC: 60:55:f9:fc:00:3d

Uploading stub...

Running stub...

Stub running...

Changing baud rate to 460800

Changed.

Configuring flash size...

A fatal error occurred: bootloader/bootloader.bin requires chip revision in range [v0.0 - v0.99] (this chip is revision v1.0). Use --force to flash anyway.

CMake Error at run_serial_tool.cmake:66 (message):
  
  C:/Espressif/python_env/idf5.3_py3.11_env/Scripts/python.exe;;C:/Espressif/frameworks/esp-idf-v5.3.1/components/esptool_py/esptool/esptool.py;--chip;esp32p4
  failed.

 -D WORKING_DIRECTORY=D:/Vineetha/Risc-V/Esp32p4/Software/Esp32p4_Test/hello_world/build -P C:/Espressif/frameworks/esp-idf-v5.3.1/components/esptool_py/run_serial_tool.cmake"

FAILED: CMakeFiles/flash D:/Vineetha/Risc-V/Esp32p4/Software/Esp32p4_Test/hello_world/build/CMakeFiles/flash 

cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v5.3.1\components\esptool_py && C:\Espressif\tools\cmake\3.24.0\bin\cmake.exe -D IDF_PATH=C:/Espressif/frameworks/esp-idf-v5.3.1 -D SERIAL_TOOL=C:/Espressif/python_env/idf5.3_py3.11_env/Scripts/python.exe;;C:/Espressif/frameworks/esp-idf-v5.3.1/components/esptool_py/esptool/esptool.py;--chip;esp32p4 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args -D WORKING_DIRECTORY=D:/Vineetha/Risc-V/Esp32p4/Software/Esp32p4_Test/hello_world/build -P C:/Espressif/frameworks/esp-idf-v5.3.1/components/esptool_py/run_serial_tool.cmake"

ninja: build stopped: subcommand failed.

ninja failed with exit code 1, output of the command is in the D:\Vineetha\Risc-V\Esp32p4\Software\Esp32p4_Test\hello_world\build\log\idf_py_stderr_output_16232 and D:\Vineetha\Risc-V\Esp32p4\Software\Esp32p4_Test\hello_world\build\log\idf_py_stdout_output_16232
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-12-28:
This issue is not related to this repository. You are using ESP-IDF directly and not Arduino. Anyway, the issue says that you need to select in menuconfig what chip revisions to support. Currently you have selected REV0 and you have a board with REV1

closed 2024-12-28 by me-no-dev

Vineetha1705, 2024-12-30:
Thanks for the response,

The problem got solved.

How do I start debugging on the ESP32-P4 Function-EV-Board?
Can you provide any documents related to debugging?

Regards,
D. Vineetha.

engineerarea, 2025-01-07:
@Vineetha1705

Hello , Vineetha 1705 i have also same problem , how did you solved it. Can you help me?

Best Regards
BD

jrlanglois, 2025-01-15:
You can have a look here for help: https://docs.espressif.com/projects/esp-idf/en/stable/esp32p4/api-reference/system/chip_revision.html#troubleshooting

In other words, you could set `CONFIG_ESP32P4_REV_MIN` to a compatible value, like `1`, in your `sdkconfig.defaults` (if you have one). In my project, I've a file named `sdkconfig.defaults.esp32p4` for this purpose.

FWIW, this issue was solved for me when using the latest IDF.

engineerarea, 2025-01-16:
> You can have a look here for help: https://docs.espressif.com/projects/esp-idf/en/stable/esp32p4/api-reference/system/chip_revision.html#troubleshooting
>
> In other words, you could set [...]

Thank you for your help , that is worked and also caused by idf revision

Vineetha1705, 2025-01-17:
Hello,

I am working with an Esp32p4-Function-EV-Board and the tool is Espress if IDE(v5.3.2). I am successfully flashing the code into Esp32p4 device but when I strat the debugging it is not entering into debug mode instead of it is showing errors like which are mentioned below:

**************************************************************************************************************************************************************************
Open On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:17)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'gdb port', not 'gdb_port'
DEPRECATED! use 'tcl port' not 'tcl_port'
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Flashing "C:/Users/Vineetha/Desktop/Test/hello_world/build/bootloader/bootloader.bin" at 0x2000
** program_esp input args <0x2000 verify> **
Error: esp_usb_jtag: could not find or open device!
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
** OpenOCD init failed **
shutdown command invoked
Error: [esp32p4.hp.cpu0] dmi_write is not implemented for this target.
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\share\openocd\scripts/target/esp_common.cfg:9: Error:
in procedure 'program_esp_bins'
in procedure 'program_esp'
in procedure 'program_error' called at file "C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 443
at file "C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 9
** Flashing Failed **
-1
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: [esp32p4.hp.cpu0] Unsupported DTM version: -1
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: [esp32p4.hp.cpu0] Could not identify target type.
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: [esp32p4.hp.cpu1] Unsupported DTM version: -1
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: [esp32p4.hp.cpu1] Could not identify target type.
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
**************************************************************************************************************************************

Please help me to solve this problem.

Regards,
D. Vineetha.

jrlanglois, 2025-01-17:
To be fair to the developers I highly recommend a new issue for your case.