espressif/arduino-esp32 #10791: Zigbee library - change definition of addBoundDevice function (ZigbeeEP.h)

lsroka76, 2024-12-30:
### Related area

Arduino Zigbee library

### Hardware specification

ESP32C6/H2

### Is your feature request related to a problem?

Changing definition of void addBoundDevice (protected) to virtual void addBoundDevice (public) will allow for override it in child classes.

### Describe the solution you'd like

protected:
void addBoundDevice...

-->>

public:
virtual void addBoundDevice...

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

P-R-O-C-H-Y, 2025-01-06:
Hi @lsroka76, why would you need to overwrite the `addBoundDevice`?

lsroka76, 2025-01-07:
Hi @P-R-O-C-H-Y If I could overwrite addBoundDevice in my code I can i.e. match the device address with my IoT components subsystem and activate related component. In fact I've already done it in my local copy of Zigbee library ;-)

P-R-O-C-H-Y, 2025-01-07:
> Hi @P-R-O-C-H-Y If I could overwrite addBoundDevice in my code I can i.e. match the device address with my IoT components subsystem and activate related component. In fact I've already done it in my local copy of Zigbee library ;-)

Feel free to also open the PR with the change :) Thanks for explaining

P-R-O-C-H-Y, 2025-02-25:
Closing this as it have been already solved by PR #10821

closed 2025-02-25 by P-R-O-C-H-Y