espressif/arduino-esp32 #10792: ESP32-S3-WROOM CAM development board is just showing a blank video on OV5640 camera.

SaadKhan10000, 2024-12-31:
### Board

ESP32-S3-WROOM-1-N16R8 CAM development board

### Device Description

A camera interface and sd card interface is attached to the development board.
[image]

### Hardware Configuration

Only OV5640 camera is attached

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

The module is working good with OV2640 camera but not working with OV5640 canmera, even I have checked 2 new cameras. In the streaming of camera example only blank image/video is shown

### Sketch

```cpp
/**********************************************************************
  Filename    : Camera Web Server
  Description : The camera images captured by the ESP32S3 are displayed on the web page.
  Auther      : www.freenove.com
  Modification: 2024/07/01
**********************************************************************/
#include "esp_camera.h"
#include <WiFi.h>

// ===================
// Select camera model
// ===================
//#define CAMERA_MODEL_WROVER_KIT // Has PSRAM
//#define CAMERA_MODEL_ESP_EYE // Has PSRAM
#define CAMERA_MODEL_ESP32S3_EYE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_PSRAM // Has PSRAM
//#define CAMERA_MODEL_M5STACK_V2_PSRAM // M5Camera version B Has PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_ESP32CAM // No PSRAM
//#define CAMERA_MODEL_M5STACK_UNITCAM // No PSRAM
//#define CAMERA_MODEL_AI_THINKER // Has PSRAM
//#define CAMERA_MODEL_TTGO_T_JOURNAL // No PSRAM
// ** Espressif Internal Boards **
//#define CAMERA_MODEL_ESP32_CAM_BOARD
//#define CAMERA_MODEL_ESP32S2_CAM_BOARD
//#define CAMERA_MODEL_ESP32S3_CAM_LCD

#include "camera_pins.h"

// ===========================
// Enter your AP credentials
// ===========================
const char* ssid = "ESP32S3-CAM-AP";
const char* password = "123456789";

void startCameraServer();

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sccb_sda = SIOD_GPIO_NUM;
  config.pin_sccb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.frame_size = FRAMESIZE_UXGA;
  config.pixel_format = PIXFORMAT_JPEG; // for streaming
  config.grab_mode = CAMERA_GRAB_WHEN_EMPTY;
  config.fb_location = CAMERA_FB_IN_PSRAM;
  config.jpeg_quality = 12;
  config.fb_count = 1;
  
  if(psramFound()){
    config.jpeg_quality = 10;
    config.fb_count = 2;
    config.grab_mode = CAMERA_GRAB_LATEST;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.fb_location = CAMERA_FB_IN_DRAM;
  }

  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  s->set_vflip(s, 1);
  s->set_brightness(s, 1);
  s->set_saturation(s, 0);

  WiFi.softAP(ssid, password);
  IPAddress IP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(IP);

  startCameraServer();

  Serial.println("Camera Ready! Connect to the AP and use the IP to access the stream.");
}

void loop() {
  delay(10000);
}
```

### Debug Message

```plain
No any debug message or error, the camera initializes but blank image/video is shown
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-12-31:
Wrong camera type choosen 
`#define CAMERA_MODEL_ESP32S3_EYE // Has PSRAM`
does not match your board.

SaadKhan10000, 2025-01-01:
So which one should I match, there is no option for my board, this is also done by www.freenove.com.
And this code is working for OV2640 camera perfectly but no OV5640

SuGlider, 2025-01-01:
>The module is working good with OV2640 camera but not working with OV5640 canmera, even I have checked 2 new cameras. In the streaming of camera example only blank image/video is shown.

This project seems to come from https://github.com/Freenove/Freenove_Development_Kit_for_ESP32_S3/tree/main/Sketches/Sketch_03_CameraWebServer

Have you considered posting an issue there?

The only difference when using OV5640 is at https://github.com/Freenove/Freenove_Development_Kit_for_ESP32_S3/blob/main/Sketches/Sketch_03_CameraWebServer/camera_index.h#L998

This is used in 
https://github.com/Freenove/Freenove_Development_Kit_for_ESP32_S3/blob/main/Sketches/Sketch_03_CameraWebServer/app_httpd.cpp#L1186-L1203

SaadKhan10000, 2025-01-08:
Still stuck

me-no-dev, 2025-01-08:
@SaadKhan10000 I tested the camera driver with ov5640 and everything was OK. It must be something on your end that is causing it

SaadKhan10000, 2025-01-10:
> @me-no-dev Are you using ESP32-S3-WROOM-1-N16R8 CAM development board?
Did you get photo/streaming?
I have checked 2 ESP32-S3-WROOM-1-N16R8 CAM development boards and OV5640 cameras.

me-no-dev, 2025-01-10:
yes, I am using a board with that same module on it. Works as expected. Images, video all. I would guess that maybe some of your pins are not correct or that the sensors are bad or something along these lines.

SaadKhan10000, 2025-01-10:
Sir @me-no-dev I am using these pins are they correct for my board:
#defined(CAMERA_MODEL_ESP32S3_EYE)
#define PWDN_GPIO_NUM -1
#define RESET_GPIO_NUM -1
#define XCLK_GPIO_NUM 15
#define SIOD_GPIO_NUM 4
#define SIOC_GPIO_NUM 5

#define Y2_GPIO_NUM 11
#define Y3_GPIO_NUM 9
#define Y4_GPIO_NUM 8
#define Y5_GPIO_NUM 10
#define Y6_GPIO_NUM 12
#define Y7_GPIO_NUM 18
#define Y8_GPIO_NUM 17
#define Y9_GPIO_NUM 16

#define VSYNC_GPIO_NUM 6
#define HREF_GPIO_NUM 7
#define PCLK_GPIO_NUM 13

me-no-dev, 2025-01-10:
We have an example here. Maybe try that: https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Camera/CameraWebServer

The board you show in the picture above is not ESP32S3-EYE, but maybe it uses the same pins... do not know. Please try our example and see what it says in the serial logs

SaadKhan10000, 2025-01-11:
@me-no-dev in the example you shared CAMERA_MODEL_ESP_EYE is selected and its pins are different from my board
[image]
So my board restarts continuously after uploading this code.

me-no-dev, 2025-01-11:
You can change in the sketch which pis are used. Comment the ESP_EYE and uncomment the ESP32S3_EYE

SaadKhan10000, 2025-01-11:
@me-no-dev now the same issue : blank stream

me-no-dev, 2025-01-11:
then it must be the sensors you have. Everything is fine on my end. Same sketch and pins

rjsachse, 2025-03-08:
I just bought a OV5640 too and have the same issue. ov2640 works, tried on esp32-cam and esp32-s3 dev using a waveshare module. connects but the data on screen is blank but log shows it is sending something.

grzesiekn, 2025-03-14:
Hi, I connected OV5640 cam to the same board running Tasmota 14.5.0.2
After restarting the image is fine for a second, and then it gets very dark, but it is displayed.

I changed some options to different values, and it seems to me that the values themselves don't significantly increase the brightness, but the change itself makes the image brighter.
Tinker a little with this options: “Saturation”:-2, “Brightness”:1, “Contrast”:2, “AWB”:1, “WBMode”:0, “AWBGain”:1, “AEC”:1