espressif/arduino-esp32 #10794: Zigbee library - ZigbeeHandlers enhancement (IAS Zone Notification message)

lsroka76, 2024-12-31:
### Related area

Arduino Zigbee library 

### Hardware specification

ESP32 C6/H2

### Is your feature request related to a problem?

Current implementation of ZigbeeHandlers doesn't handle properly IAS Zone Change Notification Message 

### Describe the solution you'd like

add to Zigbee Handlers:
// forward declaration of all implemented handlers
(...)
static esp_err_t zb_cmd_ias_zone_status_change_handler(const esp_zb_zcl_ias_zone_status_change_notification_message_t *message);
(....)

log_i("Receive Zigbee action(0x%x) callback", callback_id);
switch (callback_id) {
 (...)
    case ESP_ZB_CORE_CMD_IAS_ZONE_ZONE_STATUS_CHANGE_NOT_ID: ret = zb_cmd_ias_zone_status_change_handler((esp_zb_zcl_ias_zone_status_change_notification_message_t *)message); break;

(...)
static esp_err_t zb_cmd_ias_zone_status_change_handler(const esp_zb_zcl_ias_zone_status_change_notification_message_t *message) {
  if (!message) {
    log_e("Empty message");
  }
  if (message->info.status != ESP_ZB_ZCL_STATUS_SUCCESS) {
    log_e("Received message: error status(%d)", message->info.status);
  }
  log_v(
    "IAS Zone Status Notification: from address(0x%x) src endpoint(%d) to dst endpoint(%d) cluster(0x%x)", message->info.src_address.u.short_addr,
    message->info.src_endpoint, message->info.dst_endpoint, message->info.cluster
  );

  for (std::list<ZigbeeEP *>::iterator it = Zigbee.ep_objects.begin(); it != Zigbee.ep_objects.end(); ++it) {
    if (message->info.dst_endpoint == (*it)->getEndpoint()) {
        log_v(
          "IAS Zone Status Notification: status(%d), cluster(0x%x) ", message->zone_status, message->info.cluster);
        log_v("calling (*it)->zbIASZoneStatusChangeNotification(message)");
	(*it)->zbIASZoneStatusChangeNotification(message);
	//(*it)->zbAttributeRead(message->info.cluster, NULL);
    }
  }
  return ESP_OK;
}
/*****************/

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

P-R-O-C-H-Y, 2025-01-06:
@lsroka76 Hi, if you have a complete change you want to add, feel free to open a PR with the proposed changes and we can discuss/review it there.

lsroka76, 2025-01-07:
@P-R-O-C-H-Y Hi, PR done.

P-R-O-C-H-Y, 2025-01-07:
@lsroka76 Please provide a description in the PR, its empty now :) Thanks

lsroka76, 2025-01-07:
@P-R-O-C-H-Y Now is OK?

PR #10821 (2025-01-27): Add IAS Zone Notification Message service to ZigbeeHandlers and ZigbeeEP.h
## Description of Change
This change will allow to handle SP_ZB_CORE_CMD_IAS_ZONE_ZONE_STATUS_CHANGE_NOT_ID in zb_action_handler and dispatch it to destination endpoint for further processing.

## Tests scenarios
Tested on ESP32 C6 with Tuya open window sensors (TS0203). 

## Related links

eg. Closes #10794  of issue
closed 2025-01-27 by me-no-dev