espressif/arduino-esp32 #10795: Use RTC Watchdog on ESP32 with Arduino IDE

ViniP8, 2025-01-01:
### Board

ESP32

### Device Description

DEVKIT

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Ubuntu 22.04

### Flash frequency

Default

### PSRAM enabled

yes

### Upload speed

115200

### Description

I'm trying to configure and use the RTC watchdog on the ESP32 using the Arduino IDE. My goal is to have the RTC watchdog reset the system if it isn't fed within a specified timeout. However, despite my attempts, the RTC watchdog doesn't seem to trigger the reset as expected.

I specifically want to use the RTC watchdog and not the Task Watchdog or any interrupts. My focus is on configuring the RTC watchdog purely via registers to achieve a system reset without involving the FreeRTOS task-level watchdog or handling interrupts.

Steps I've Taken

I've enabled the RTC watchdog using the following register configurations:

`
WRITE_PERI_REG(RTC_CNTL_WDTCONFIG0_REG, RTC_CNTL_WDT_EN | RTC_CNTL_WDT_FLASHBOOT_MOD_EN);
WRITE_PERI_REG(RTC_CNTL_WDTCONFIG1_REG, (3 << RTC_CNTL_WDT_STG0_S)); // Stage 0: Reset system
WRITE_PERI_REG(RTC_CNTL_WDTCONFIG2_REG, 8000); // Timeout of ~53 ms
WRITE_PERI_REG(RTC_CNTL_WDTFEED_REG, 1); // Feed watchdog
`
I attempted to stop feeding the watchdog after a few loops in my loop() function to let it expire:

`
if (count > 5) {
  while (true) { /* Simulating a crash */ }
} else {
  WRITE_PERI_REG(RTC_CNTL_WDTFEED_REG, 1); // Feed watchdog
}
`
Despite this, the system never resets.

### Sketch

```cpp
#include "soc/rtc_cntl_reg.h"
#include "soc/soc.h"

void setup() {
  Serial.begin(115200);

  WRITE_PERI_REG(RTC_CNTL_WDTCONFIG0_REG, RTC_CNTL_WDT_EN | RTC_CNTL_WDT_FLASHBOOT_MOD_EN);
  WRITE_PERI_REG(RTC_CNTL_WDTCONFIG1_REG, (3 << RTC_CNTL_WDT_STG0_S));
  WRITE_PERI_REG(RTC_CNTL_WDTCONFIG2_REG, 8000);
  WRITE_PERI_REG(RTC_CNTL_WDTFEED_REG, 1);

  Serial.println("RTC Watchdog initialized!");
}

void loop() {
  static int count = 0;
  count++;

  if (count > 5) {
    Serial.println("Leaving RTC Watchdog to expire...");
    while (true) { /* Simulate system hang */ }
  } else {
    Serial.println("Feeding RTC Watchdog...");
    WRITE_PERI_REG(RTC_CNTL_WDTFEED_REG, 1);
  }

  delay(1000);
}
```

### Debug Message

```plain
-
```

### Other Steps to Reproduce

I also tried using functions like `rtc_wdt_enable()` and similar ones (`wdt_enable(WDTO_2S)`), but the code does not compile.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-01-01:
This question is not related to Arduino API or to any Arduino functionality.
ESP32 Arduino uses RTC WDT just in the boot process.

### Based on the [WDT documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/wdts.html#hardware-watchdog-timers):

RTC Watchdog Timer (RTC_WDT) - used to track the boot time from power-up until the user's main function (by default RTC Watchdog is disabled immediately before the user's main function).

Refer to the [Watchdog](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/bootloader.html#bootloader-watchdog) section to understand how watchdogs are utilized in the bootloader.

The app's behaviour can be adjusted so the RTC Watchdog remains enabled after app startup. The Watchdog would need to be explicitly reset (i.e., fed) or disabled by the app to avoid the chip reset. To do this, set the [CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html#config-bootloader-wdt-disable-in-user-code) option, modify the app as needed, and then recompile the app. In this case, the following APIs should be used:

- `wdt_hal_disable()`: see [To Disable RTC_WDT](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/hardware-abstraction.html#hw-abstraction-hal-layer-disable-rtc-wdt)
- `wdt_hal_feed()`: see [To Reset the RTC_WDT Counter](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/hardware-abstraction.html#hw-abstraction-hal-layer-feed-rtc-wdt)
- `rtc_wdt_feed()`
- `rtc_wdt_disable()
`

If RTC_WDT is not reset/disabled in time, the chip will be automatically reset. See [RTC Watchdog Timeout](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/fatal-errors.html#rtc-watchdog-timeout) for more information.

-------
### Default ESP32 sdkconfig RTC WTD settings:

```
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
```

------
### Conclusion:

RTC WDT is configured in Arduino sdkconfig to only monitor the boot timeout. 
In order to use it within the Arduino Application, it is necessary to create a Arduino as IDF Component project and enable `CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE` using menuconfig.

The RTC WDT Timeout shall also be configured using ` CONFIG_BOOTLOADER_WDT_TIME_MS` within menuconfig.

For the specific question from this issue, I'd suggest not using REGISTER direct configuration, but instead use HAL or LL functions.
Reading the IDF Boot source code may bring some good insights about how to use it.
https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/src/bootloader_init.c#L78-L87
https://github.com/espressif/esp-idf/blob/master/components/esp_system/startup_funcs.c#L151-L160
https://github.com/espressif/esp-idf/blob/master/components/esp_system/port/soc/esp32/clk.c#L136-L149
https://github.com/espressif/esp-idf/blob/master/components/esp_system/port/soc/esp32/clk.c#L161-L168

SuGlider, 2025-01-02:
@ViniP8 - I have created an Arduino Sketch thas can configure RTC WDT, count time, and reset due to the RTC Watchdog.

--------

### You can try it with the ESP32:

``` cpp
#include "soc/rtc.h"
#include "hal/wdt_hal.h"
#include "rtc_wdt.h"

// Define WTC Watchdog Timer in milliseconds
#define RTC_WDT_TIME_MS (1300)

wdt_hal_context_t rwdt_ctx;

void setup() {
  Serial.begin(115200);
  Serial.print("Enabling RTCWDT(");
  Serial.print(RTC_WDT_TIME_MS);
  Serial.println(" ms)");

  // restating RTC WDT after disabled by IDF Bootloader which 
  rwdt_ctx.inst = WDT_RWDT; // sets RTC WDT type
  rwdt_ctx.rwdt_dev = RWDT_DEV_GET(); // sets the WDT Device Get function

  wdt_hal_init(&rwdt_ctx, WDT_RWDT, 0, false);
  uint32_t stage_timeout_ticks = (uint32_t)((uint64_t)RTC_WDT_TIME_MS * rtc_clk_slow_freq_get_hz() / 1000);
  wdt_hal_write_protect_disable(&rwdt_ctx);
  wdt_hal_config_stage(&rwdt_ctx, WDT_STAGE0, stage_timeout_ticks, WDT_STAGE_ACTION_RESET_RTC);
  wdt_hal_enable(&rwdt_ctx);
  wdt_hal_write_protect_enable(&rwdt_ctx);
}

void loop() {
  // loop parameters
  const uint32_t loopTimer = 250;
  static uint32_t loopCounter = 0;
  
  // prints interactor and past time, example: "#4  :: loop() ==> 0.50 seconds"
  Serial.print("#");
  Serial.print(loopCounter);
  Serial.print("  :: loop() ==> ");
  Serial.print((float) ((loopCounter * loopTimer)/1000.0));
  Serial.println(" seconds");
  
  delay(loopTimer);
  loopCounter++;

  if (loopCounter <= 10) {
    Serial.println("RTC WDT was fed.");
    rtc_wdt_feed();
  } else {
    Serial.println("RTC WDT NOT FED!!");
  }
}
```

--------

### Serial Output:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4832
load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
entry 0x400805cc
Enabling RTCWDT(1300 ms)
#0  :: loop() ==> 0.00 seconds
RTC WDT was fed.
#1  :: loop() ==> 0.25 seconds
RTC WDT was fed.
#2  :: loop() ==> 0.50 seconds
RTC WDT was fed.
#3  :: loop() ==> 0.75 seconds
RTC WDT was fed.
#4  :: loop() ==> 1.00 seconds
RTC WDT was fed.
#5  :: loop() ==> 1.25 seconds
RTC WDT was fed.
#6  :: loop() ==> 1.50 seconds
RTC WDT was fed.
#7  :: loop() ==> 1.75 seconds
RTC WDT was fed.
#8  :: loop() ==> 2.00 seconds
RTC WDT was fed.
#9  :: loop() ==> 2.25 seconds
RTC WDT was fed.
#10  :: loop() ==> 2.50 seconds
RTC WDT NOT FED!!
#11  :: loop() ==> 2.75 seconds
RTC WDT NOT FED!!
#12  :: loop() ==> 3.00 seconds
RTC WDT NOT FED!!
#13  :: loop() ==> 3.25 seconds
RTC WDT NOT FED!!
#14  :: loop() ==> 3.50 seconds
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4832
load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
entry 0x400805cc
Enabling RTCWDT(1300 ms)
#0  :: loop() ==> 0.00 seconds
RTC WDT was fed.
#1  :: loop() ==> 0.25 seconds
RTC WDT was fed.
#2  :: loop() ==> 0.50 seconds
RTC WDT was fed.
#3  :: loop() ==> 0.75 seconds
RTC WDT was fed.
#4  :: loop() ==> 1.00 seconds
RTC WDT was fed.
#5  :: loop() ==> 1.25 seconds
RTC WDT was fed.
#6  :: loop() ==> 1.50 seconds
RTC WDT was fed.
#7  :: loop() ==> 1.75 seconds
RTC WDT was fed.
#8  :: loop() ==> 2.00 seconds
RTC WDT was fed.
#9  :: loop() ==> 2.25 seconds
RTC WDT was fed.
#10  :: loop() ==> 2.50 seconds
RTC WDT NOT FED!!
#11  :: loop() ==> 2.75 seconds
RTC WDT NOT FED!!
#12  :: loop() ==> 3.00 seconds
RTC WDT NOT FED!!
#13  :: loop() ==> 3.25 seconds
RTC WDT NOT FED!!
#14  :: loop() ==> 3.50 seconds
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee

```

ViniP8, 2025-01-03:
Thank you so much, it was exactly what I needed. 😊

closed 2025-01-03 by ViniP8

SuGlider, 2025-09-26:
@amsova - Yes, There is a IDF way to feed WDT that works for all SoC (ESP32, ESP32-S3, ESP32-C3...).
Replace `loop()` function with this:

``` cpp
void loop() {
  // loop parameters
  const uint32_t loopTimer = 250;
  static uint32_t loopCounter = 0;

  // prints interactor and past time, example: "#4  :: loop() ==> 0.50 seconds"
  Serial.print("#");
  Serial.print(loopCounter);
  Serial.print("  :: loop() ==> ");
  Serial.print((float)((loopCounter * loopTimer) / 1000.0));
  Serial.println(" seconds");

  delay(loopTimer);
  loopCounter++;

  if (loopCounter <= 10) {
    Serial.println("RTC WDT was fed.");
    // This works for ESP32-S3, ESP32, etc
    wdt_hal_write_protect_disable(&rwdt_ctx);
    wdt_hal_feed(&rwdt_ctx);
    wdt_hal_write_protect_enable(&rwdt_ctx);
  } else {
    Serial.println("RTC WDT NOT FED!!");
  }
}
```
#### Output:
``` 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1240
load:0x403c8700,len:0xc6c
load:0x403cb700,len:0x3134
entry 0x403c88c4
Enabling RTCWDT(1300 ms)
#0  :: loop() ==> 0.00 seconds
RTC WDT was fed.
#1  :: loop() ==> 0.25 seconds
RTC WDT was fed.
#2  :: loop() ==> 0.50 seconds
RTC WDT was fed.
#3  :: loop() ==> 0.75 seconds
RTC WDT was fed.
#4  :: loop() ==> 1.00 seconds
RTC WDT was fed.
#5  :: loop() ==> 1.25 seconds
RTC WDT was fed.
#6  :: loop() ==> 1.50 seconds
RTC WDT was fed.
#7  :: loop() ==> 1.75 seconds
RTC WDT was fed.
#8  :: loop() ==> 2.00 seconds
RTC WDT was fed.
#9  :: loop() ==> 2.25 seconds
RTC WDT was fed.
#10  :: loop() ==> 2.50 seconds
RTC WDT NOT FED!!
#11  :: loop() ==> 2.75 seconds
RTC WDT NOT FED!!
#12  :: loop() ==> 3.00 seconds
RTC WDT NOT FED!!
#13  :: loop() ==> 3.25 seconds
RTC WDT NOT FED!!
#14  :: loop() ==> 3.50 seconds
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1240
```

amsova, 2025-09-28:
Thanks SuGlider, it works fine, thank you.