espressif/arduino-esp32 #10798: Hostname Resolution in Async UDP

Witty-Wizard, 2025-01-02:
### Related area

AsyncUDP

### Hardware specification

ESP32 Pico Kit

### Is your feature request related to a problem?

The current implementation of `AsyncUDP` does not support Hostname resolution.

### Describe the solution you'd like

I would like to add a function overload that accepts a hostname and using `DNS` resolve, then start the communication form the resolved IP address

### Additional context

I have an implementation in mind:
```cpp
bool AsyncUDP::listen(const char *host, uint16_t port) {
    struct hostent *server;
    server = gethostbyname(host);
    if (server == NULL) {
        log_e("could not resolve host: %s (errno: %d)", host, errno);
        return 0;
    }

    IPAddress resolvedIP((const uint8_t *)(server->h_addr_list[0]));
    // use resolved IP to start the UDP communication
}
```

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

vortigont, 2025-01-06:
this does not makes sense since AsyncUDP is `async`.
Such code won't work: `server = gethostbyname(host);`
Function `gethostbyname` should block unless hostname is resolved or timeouts, thus using async call won' give you any benefit at all.

Witty-Wizard, 2025-01-06:
Now that I think about it, that makes a lot of sense. Its just better to resolve the hostname before calling the function

closed 2025-01-06 by Witty-Wizard