espressif/arduino-esp32 #10799: Fail to compile if your windows usernam contains "&"

darrepac, 2025-01-02:
### Board

ESP32 WROOM

### Device Description

plain module, USB

### Hardware Configuration

No HW configuration yet

### Version

v3.1.0

### IDE Name

Arduino IDE

### Operating System

W11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Can't compile any sketch, end up quickly with the following error:
```
& was unexpected at this time.
exit status 1

Compilation error: exit status 1
```
Seems related to the fact that my Windows username contains a "&" which was not causing problem so far with other sw (more than 10 years)

### Sketch

```cpp
any
```

### Debug Message

```plain
& was unexpected at this time.
exit status 1

Compilation error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-01-02:
@darrepac - Was it working before with the same Arduino IDE + previous ESP32 Arduino Core version?

darrepac, 2025-01-02:
First time I work on ESP32...but I can say that it was neither working with previous Arduino IDE and previous ESP32 Arduino Core Version. So seems to be an old problem (but still there)

Jason2866, 2025-03-12:
Closing since not an espressif32 Arduino core issue

closed 2025-03-12 by Jason2866