espressif/arduino-esp32 #10803: HTTPClient Freezes on Core 3.1.0

beslagic, 2025-01-04:
### Board

XIAO ESP32S3 Sense

### Device Description

XIAO ESP32S3 Sense

### Hardware Configuration

OLED I2C 0.96''display

### Version

v3.1.0

### IDE Name

Arduino IDE 2.3.4

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

# ESP32 HTTPClient Issue with Core 3.1.0

This repository demonstrates an issue with the ESP32 Arduino Core 3.1.0 where HTTPClient calls freeze under specific conditions.

## Reproducing the Issue

1. Connect to a WiFi network by updating `ssid` and `password` in the `main.ino` file.
2. Upload the sketch to an ESP32 board using Arduino IDE.
3. Monitor the serial output to observe freezing behavior during HTTP requests.

## Hardware and Software Details

- **Board**: XIAO ESP32S3 Sense
- **Arduino Core Version**: 3.1.0
- **HTTPClient Library Version**: Default with ESP32 Core
- **Tested API Endpoints**: JSONPlaceholder (https://jsonplaceholder.typicode.com)

## Code

The code is provided in the `main.ino` file.

### Sketch

```cpp
#include <WiFi.h>
#include <HTTPClient.h>
#include <ArduinoJson.h>

// WiFi settings
const char* ssid = "YourWiFiSSID";
const char* password = "YourWiFiPassword";

// Global settings
const int ListSize = 3; // Simulate a list
const char* apiEndpoints[] = {
    "https://jsonplaceholder.typicode.com/todos/1",
    "https://jsonplaceholder.typicode.com/todos/2",
    "https://jsonplaceholder.typicode.com/todos/3"
};

bool isSTAConnected = false;

void setup() {
    Serial.begin(115200);

    // Connect to WiFi
    WiFi.begin(ssid, password);
    Serial.print("Connecting to WiFi");
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println("\nWiFi connected!");
    Serial.print("IP Address: ");
    Serial.println(WiFi.localIP());
    isSTAConnected = true;

    // Start task for fetching data
    xTaskCreatePinnedToCore(
        displayStocksTask,    // Task function
        "DisplayStocksTask",  // Task name
        8192,                 // Stack size
        NULL,                 // Parameters
        1,                    // Priority
        NULL,                 // Task handle
        1                     // Core
    );
}

void loop() {
    // Main loop remains empty as we are using RTOS tasks
}

void displayStocksTask(void *pvParameters) {
    while (1) {
        if (isSTAConnected && WiFi.status() == WL_CONNECTED) {
            for (int i = 0; i < ListSize; i++) {
                readData(i);
                vTaskDelay(5000 / portTICK_PERIOD_MS);
            }
        }
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

void readData(int index) {
    if (index < 0 || index >= ListSize) return;

    String httpRequestAddress = apiEndpoints[index];
    HTTPClient http;
    http.begin(httpRequestAddress);

    int httpCode = http.GET();

    if (httpCode > 0) {
        // Successful request
        String payload = http.getString();
        Serial.printf("Data for index %d: %s\n", index, payload.c_str());
    } else {
        // Error
        Serial.printf("HTTP request error for index %d: %d\n", index, httpCode);
    }

    http.end();
}
```

### Debug Message

```plain
01:19:20.924 -> [125069][D][HTTPClient.cpp:378] disconnect(): tcp keep open for reuse
01:19:25.942 -> [130076][D][HTTPClient.cpp:253] beginInternal(): unexpected protocol: https, expected http
01:19:25.942 -> [130077][D][HTTPClient.cpp:293] beginInternal(): protocol: https, host: jsonplaceholder.typicode.com port: 443 url: /todos/3
01:19:25.942 -> [130084][D][HTTPClient.cpp:574] sendRequest(): request type: 'GET' redirCount: 0
01:19:25.942 -> 
01:19:25.942 -> [130093][D][NetworkManager.cpp:127] hostByName(): DNS found IPv4 104.21.59.19
01:19:25.981 -> [130138][D][ssl_client.cpp:196] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
01:19:26.944 -> [131086][D][HTTPClient.cpp:1112] connect():  connected to jsonplaceholder.typicode.com:443
01:19:27.024 -> [131168][D][HTTPClient.cpp:1257] handleHeaderResponse(): code: 200
01:19:27.024 -> [131170][D][HTTPClient.cpp:1260] handleHeaderResponse(): size: 84
01:19:27.024 -> [131170][D][HTTPClient.cpp:618] sendRequest(): sendRequest code=200
01:19:27.024 -> 
01:19:27.024 -> [131178][D][HTTPClient.cpp:378] disconnect(): tcp keep open for reuse
01:19:27.067 -> Data for index 2: {
01:19:27.067 ->   "userId": 1,
01:19:27.067 ->   "id": 3,
01:19:27.067 ->   "title": "fugiat veniam minus",
01:19:27.067 ->   "completed": false
01:19:27.067 -> }
01:19:27.067 -> [131182][D][HTTPClient.cpp:378] disconnect(): tcp keep open for reuse
01:19:33.042 -> [137189][D][HTTPClient.cpp:253] beginInternal(): unexpected protocol: https, expected http
01:19:33.042 -> [137190][D][HTTPClient.cpp:293] beginInternal(): protocol: https, host: jsonplaceholder.typicode.com port: 443 url: /todos/1
01:19:33.089 -> [137197][D][HTTPClient.cpp:574] sendRequest(): request type: 'GET' redirCount: 0
01:19:33.089 -> 
01:19:33.089 -> [137206][D][NetworkManager.cpp:127] hostByName(): DNS found IPv4 104.21.59.19
01:19:33.122 -> [137253][D][ssl_client.cpp:196] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
---> HERE IS FREEZE!!!!
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-01-04:
I am unable to reproduce.  Perhaps the server or a firewall is blocking you. Set appropriate timeouts:
```
http.setConnectTimeout(4000);
http.setTimeout(4000);
```

beslagic, 2025-01-04:
Btw. ESP version 3.0.7 works perfectly.

lbernstone, 2025-01-04:
I am really unable to reproduce.  Perhaps `if (isSTAConnected && WiFi.status() == WL_CONNECTED)` is no longer true.
Add this to your code to see if you are really frozen
```
#include <Ticker.h>
Ticker tk30sec;
void alive() {
  Serial.println("alive");
}

void setup() {
    tk30sec.attach(30, alive);
...
```

beslagic, 2025-01-04:
Thank you for your response. I want to clarify that the issue I am experiencing is not related to the application freezing entirely. The application remains responsive and continues to execute other tasks. For example, your provided Ticker-based alive() function continues to run and output "alive" messages.

The problem specifically lies with the HTTPClient during the http.GET() call. When this call is made, the HTTP communication halts indefinitely under certain conditions, particularly when trying to fetch data from an API.

**Observations:**
Task Behavior: The rest of the application (e.g., other tasks, timers, and serial logs) continues to function.
**Code Execution Halt:** The http.GET() call never returns, and the program flow is stuck at this line .
Version Dependency: When using ESP32 core version 3.1.0, the issue occurs consistently. However, when reverting to ESP32 core version 3.0.7, the same code functions correctly without freezing.
**Suggested Reproduction Steps:**
Use the provided code from the repository, which includes an RTOS task that periodically makes HTTP requests.
Test the behavior on both ESP32 core version 3.1.0 (where the issue is present) and version 3.0.7 (where the issue is absent).
Observe the serial output; you will see the alive message, but the HTTP request fails to return on 3.1.0.
I hope this explanation provides more clarity. Let me know if additional details are required.

Best regards,
Beslagic

SuGlider, 2025-01-05:
> Thank you for your response. I want to clarify that the issue I am experiencing is not related to the application freezing entirely. The application remains responsive and continues to execute other tasks. For example, your provided Ticker-based alive() function continues to run and output "alive" messages.
> 
> The problem specifically lies with the HTTPClient during the http.GET() call. When this call is made, the HTTP communication halts indefinitely under certain conditions, particularly when trying to fetch data from an API.
> 
> _**Observations:**_ **Task Behavior:** The rest of the application (e.g., other tasks, timers, and serial logs) continues to function. **Code Execution Halt:** The http.GET() call never returns, and the program flow is stuck at this line . **Version Dependency:** When using ESP32 core version 3.1.0, the issue occurs consistently. However, when reverting to ESP32 core version 3.0.7, the same code functions correctly without freezing. **Suggested Reproduction Steps:** Use the provided code from the repository, which includes an RTOS task that periodically makes HTTP requests. Test the behavior on both ESP32 core version 3.1.0 (where the issue is present) and version 3.0.7 (where the issue is absent). Observe the serial output; you will see the alive message, but the HTTP request fails to return on 3.1.0. I hope this explanation provides more clarity.
> 
> Best regards, Beslagic

Maybe @me-no-dev can take a look and quickly find a potential reason for this issue.

me-no-dev, 2025-01-05:
The only change in our code that has anything to do with the HTTP Client is this: https://github.com/espressif/arduino-esp32/commit/1712fb5eeee9069ad4e016c5fd6659b28ca397b0
@beslagic maybe you can try to reverse it and see if it will make things work. If this does not work, then it must be something on ESP-IDF's side and we will need to reproduce it to debug further

me-no-dev, 2025-01-05:
I am able to reproduce the issue with the provided example code. The reason for failure is that the `loop()` is spinning empty and with the same priority as the HTTP task. With that, the scheduler does not provide enough execution time to the HTTP task and execution either halts, or goes so slow that the client is dropped. There are two ways to overcome this and the best one is to just add `delay(1000);` in the `loop()`. This will in effect make the empty loop task to always sleep and not interfere with the other tasks. The second option is to raise the priority level of the HTTP task and with that tell the scheduler to give more execution time to it. That will continue to busy-loop the `loop()` task and waste CPU cycles.

The suggested change reversal above does not help in any way, so no need to do it.

You say that it works fine on 3.0.7, but I would not classify this case as "fine" because of the busy loop. I will also guess that there were some changes to the FreeRTOS kernel in ESP-IDF that slightly changed the behavior and cause that problem in cases like yours

beslagic, 2025-01-05:
Thank you for identifying the root cause of the problem. I initially tested your suggestion to add delay(1000); in the loop() function, but it didn’t feel like the optimal solution for my use case. Instead, I resolved the issue by adding vTaskDelete(NULL); in the loop() function. This ensures that the default loop task is deleted and does not consume unnecessary CPU time, allowing my RTOS tasks to run smoothly without interference.

I appreciate your guidance—it was instrumental in helping me find the right fix. Thank you for your support! 😊

me-no-dev, 2025-01-05:
Sure. Deleting the loop task is also an option

closed 2025-01-05 by me-no-dev

mikesoaps, 2025-04-07:
Not sure if anyone has since seen this issue again but I am experiencing the same issue.
I am attempting a POST request with a File (stream) as the body. On my server I am getting the full payload and returning 200 but the task that runs sendRequest is hanging just after [HTTPClient.cpp:1112] connect():  connected to xxxxxxxxs

I have made sure that this task is running on a different core to the only other task running in my application code (besides for loop, which has a delay(100) in it.

Any pointers?

Also running the 3.1.3 on platformio.

mikesoaps, 2025-04-07:
`extern const uint8_t x509_crt_imported_bundle_bin_start[] asm("_binary_x509_crt_bundle_start");
extern const uint8_t x509_crt_imported_bundle_bin_end[] asm("_binary_x509_crt_bundle_end");
NetworkClientSecure https_client;
bool client_initialized = false;

bool PostDataToRecoveryEndpoint(String tableName, String fileName) {
    bool success = false;

    if (!client_initialized) {
        https_client.setCACertBundle(x509_crt_imported_bundle_bin_start, x509_crt_imported_bundle_bin_end - x509_crt_imported_bundle_bin_start);
        https_client.setTimeout(10);
        client_initialized = true;
    }

    HTTPClient https;
    https.setConnectTimeout(10000);
    https.setReuse(false); // Don't keep connection alive

    Serial.println("Connecting to server...");
    esp_task_wdt_reset(); // Reset watchdog to prevent reset
    
    Serial.print("Opening File: ");
    Serial.println(fileName);
    File file = SD_MMC.open(fileName, FILE_READ);
    if (!file) {
        Serial.println("Failed to open file for reading");
        return false;
    }
    
    String fn = file.name();
    size_t fileSize = file.size();

    if (https.begin(https_client, server_domain, 443, DataRecoveryPath)) {

        https.addHeader("Content-Type", "application/txt");
        https.addHeader("Content-Disposition", "attachment; filename=" + fn);
        https.addHeader("Content-Length", String(fileSize));
        https.addHeader("deviceType", tableName);
        https.addHeader("serialNumber", LILYGO_SERIAL);
        https.addHeader("tableName", "telemetry");
        
        Serial.println("Sending file...");
        
        
        esp_task_wdt_delete(NULL); // Temporarily disable watchdog for test purposes
        
        int httpResponseCode = -1;
        try {
            Serial.println("Sending HTTP request...");
            httpResponseCode = https.sendRequest("POST", &file);
            Serial.println("HTTP request completed with code: " + String(httpResponseCode));
        } catch (...) {
            Serial.println("Exception during HTTP request");
        }
        
        esp_task_wdt_add(NULL);
        esp_task_wdt_reset();
        
        file.close();
        
        if (httpResponseCode > 0) {
            Serial.println(httpResponseCode);
            if (httpResponseCode == HTTP_CODE_OK) {
                Serial.println("File sent successfully!");
                success = true;
            } else {
                Serial.printf("Error: %s\n", https.errorToString(httpResponseCode).c_str());
            }
        } else {
            Serial.printf("HTTP error: %d - %s\n", httpResponseCode, https.errorToString(httpResponseCode).c_str());
        }
        
        https.end();
        return success;
    } else {
        Serial.println("Failed to start HTTP connection");
        file.close();
        return false;
    }
}`