espressif/arduino-esp32 #10804: Name conflict with attribute_t when using M5Unified library alongside Matter library

kyjibato, 2025-01-04:
### Board

M5Stack Atom

### Device Description

M5Stack Atom Lite

### Hardware Configuration

No additional hardware attached besides the integrated sensors (M5Unit-ENV)

### Version

v3.1.0

### IDE Name

Arduino IDE

### Operating System

macOS

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

When compiling an Arduino project that uses the Matter library (v3.1.0) together with M5Unified (v0.2.2) on M5Stack Atom, I encounter a compilation error caused by an ambiguous reference to attribute_t. It appears that the Matter library defines attribute_t in esp_matter_core.h, and M5Unified (via M5GFX) also declares attribute_t as an enum in enum.hpp.

### Sketch

```cpp
#include <M5Unified.h>
#include <Matter.h>

void setup() {
  M5.begin();
  Matter.begin();

  // ...
}

void loop() {
  M5.update();
  // ...
}
```

### Debug Message

```plain
In file included from .../MatterGenericSwitch.h:20,
                 from .../Matter.h:22,
                 from src/main.cpp:23:
/.../MatterEndPoint.h:41:3: error: reference to 'attribute_t' is ambiguous
   41 |   attribute_t *getAttribute(uint32_t cluster_id, uint32_t attribute_id) {
      |   ^~~~~~~~~~~

... (multiple lines with similar 'attribute_t' is ambiguous errors) ...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-01-04:
Please post this to the M5Unified repository. We can not make changes to matter on our end.

SuGlider, 2025-01-05:
It is possible to change the function signature to return `esp_matter::attribute_t` using the proper qualifier in order to solve the ambiguity.

I'll add a PR and @kyjibato would be able to test it.

SuGlider, 2025-01-05:
@kyjibato - please test the PR https://github.com/espressif/arduino-esp32/pull/10807 
It solves this issue.

kyjibato, 2025-01-06:
@SuGlider Thank you for your hard work on PR #10807. I tested it in my environment, and it resolves the reported issue. I really appreciate your help.

PR #10807 (2025-01-07): feat(matter): adds esp_matter:: namespace to attribute_t
## Description of Change

When compiling an Arduino project that uses the Matter library (v3.1.0) together with M5Unified (v0.2.2) on M5Stack Atom, I encounter a compilation error caused by an ambiguous reference to attribute_t. It appears that the Matter library defines attribute_t in esp_matter_core.h, and M5Unified (via M5GFX) also declares attribute_t as an enum in enum.hpp.

This PR fixes this issue by forcing the namespace `esp_matter::` into `attribute_t` matter type name.

## Tests scenarios
Using M5Stack Libraries:

``` cpp
#inc [... cut]
closed 2025-01-06 by kyjibato