espressif/arduino-esp32 #10806: UART_HW_FIFO_LEN macro not defined

s60sc, 2025-01-05:
### Board

n/a

### Device Description

n/a

### Hardware Configuration

n/a

### Version

v3.1.0

### IDE Name

Arduino IDE

### Operating System

n/a

### Flash frequency

n/a

### PSRAM enabled

no

### Upload speed

n/a

### Description

Compiling existing code in v3.1.0 causes:

`warning: 'UART_FIFO_LEN' macro is deprecated, please use 'UART_HW_FIFO_LEN' instead`

replacing macro causes:

`error: 'UART_HW_FIFO_LEN' was not declared in this scope`

### Sketch

```cpp
#include "driver/uart.h"
void setup(){char buff[UART_FIFO_LEN];}
void loop(){}

#include "driver/uart.h"
void setup(){char buff[UART_HW_FIFO_LEN];}
void loop(){}
```

### Debug Message

```plain
n/a
```

### Other Steps to Reproduce

n/a

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-01-05:
That is rather strange, given that `UART_HW_FIFO_LEN` is defined right after the deprecation of `UART_FIFO_LEN`
https://github.com/espressif/esp-idf/blob/release/v5.3/components/esp_driver_uart/include/driver/uart.h#L25-L30

me-no-dev, 2025-01-05:
You need to use `UART_HW_FIFO_LEN(uart_num)`

closed 2025-01-05 by me-no-dev

s60sc, 2025-01-05:
maybe change deprecation message to clarify

me-no-dev, 2025-01-05:
Suggestion is not bad, but is part of ESP-IDF and we have no control over it. We will try to notify them about it.