espressif/arduino-esp32 #10810: Fresh project build errors - ESP Insights Project commit: HEAD-HASH-NOTFOUND & ESP RainMaker Project commit: HEAD-HASH-NOTFOUND (IDFGH-14314)

ae-5th, 2025-01-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build without errors.

### What is the actual behavior?

Build with errors.

### Steps to reproduce.

Go to New Project Wizard - Choose Template arduino-as-component.
Then CTRL+Shift+P and type/choose ESP-IDF: Add Arduino ESP32 as ESP-IDF Component.

### Build or installation Logs.

```plain
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
ESP Insights Project commit: HEAD-HASH-NOTFOUND
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
ESP RainMaker Project commit: HEAD-HASH-NOTFOUND
CMake Error at components/arduino/CMakeLists.txt:362 (message):
  esp32-arduino requires CONFIG_FREERTOS_HZ=1000 (currently 100)
```

### More Information.

Fresh project with Arduino component does not build.

me-no-dev, 2025-01-05:
The actual issue is `esp32-arduino requires CONFIG_FREERTOS_HZ=1000 (currently 100)`.
Try adding `CONFIG_FREERTOS_HZ=1000` to `sdkconfig.defaults` inside the project (create the file if not there) and try to build again

ae-5th, 2025-01-08:
That did not helped. Still same build errors.