espressif/arduino-esp32 #10811: SPI related core panic (IDFGH-13789)

y-nk, 2025-01-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

latest

### Espressif SoC revision.

ESP32-D0WD-V3

### Operating System used.

macOS

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

I'm streaming wav files from sdcard to a bluetooth sink using arduino-audio-tools. For this, I want to keep a dedicated bus for the sdcard reader (for performance reason) so it is wired on a custom set of gpio (not 5/18/19/21) but rather 25/26/27/33 (i had tried 1st 12/13/14/27 but using 12 crashes my board at startup). 

This is the sketch I'm using:

```ino
#include "AudioTools.h"
#include "AudioLibs/A2DPStream.h"
#include "AudioLibs/AudioSourceSD.h"
#include "AudioCodecs/CodecWAV.h"

#define HSPI_SCLK 26
#define HSPI_MISO 33
#define HSPI_MOSI 25
#define HSPI_SS   27

const char *startFilePath="/";
const char *ext="wav";

SPIClass hspi(HSPI);
AudioSourceSD source(startFilePath, ext, HSPI_SS, hspi);
A2DPStream out;
WAVDecoder decoder;
AudioPlayer player(source, out, decoder);

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Debug);

  hspi.begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI, HSPI_SS);

  // setup output
  auto cfg = out.defaultConfig(TX_MODE);
  cfg.name = "Bose Color II SoundLink";
  cfg.auto_reconnect = true;
  out.begin(cfg);

  // setup player
  player.begin();
}

void loop() {
  player.copy();
}
```

This will crash the board at some point, invoking an SPI related error. Decoded stack trace is:

```log
PC: 0x40083d80 EXCVADDR: 0x00000000  Decoding stack results
  0x400e71fa: pthread_key_create at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/pthread/pthread_local_storage.c line 56
  0x401a2e0d: mbedtls_ecdsa_read_signature_restartable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecdsa.c line 762
  0x400df265: SPIClass::endTransaction() at /$HOME/Library/Arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/SPI/src/SPI.cpp line 196
  0x400dec09: VFSImpl::mkdir(char const*) at /$HOME/Library/Arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/FS/src/vfs_api.cpp line 215
  0x400deb0d: VFSImpl::remove(char const*) at /$HOME/Library/Arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/FS/src/vfs_api.cpp line 438
  0x40099171: prvTimerTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/timers.c line 587
  0x40099bae: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 162
```

I thought first to report to arduino-audio-tools but since the stack reveals no relationship with it, i thought this is much more related to an error here rather than there. I'm happy to move this report to the relevant place if found that it's not a core issue.

### What is the actual behavior?

Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 

### Steps to reproduce.

1. Have my board
2. Have my components and wiring
4. Prepare an SD card
5. Build the sketch
6. Should crash after some time

me-no-dev, 2025-01-05:
there is nothing wrong with SPI. The point where it says SPI is failing is after it had released it's semaphore. It has nothing to do with actual SPI peripheral and the semaphore had been previously taken, so it can not be NULL. The whole backtrace looks funky. It mentions `remove`, `mkdir`, `mbedtls` and `pthread` where those should not be even used in this case. Code would be reading, not removing files or doing anything with mbedtls. Probably the wrong `elf` was used to decode the backtrace.

y-nk, 2025-01-06:
@me-no-dev i guess those `remove`/`mkdir` and other calls are part of arduino-audio-tools then, i should move the report to them?

me-no-dev, 2025-01-06:
You could ask them and see what they would say. I kinda expect they are not doing it at all and the backtrace is actually wrong.

y-nk, 2025-01-06:
@me-no-dev i'll close this here then ; since i opened this about 3 months ago i will try to see if i can reproduce again and if i can produce a log then i'll ask them.

closed 2025-01-06 by y-nk

me-no-dev, 2025-01-06:
@y-nk sorry for the late response, but the issue was transferred from IDF to us last night. Hope it's not too late or that the problem has fixed itself

y-nk, 2025-01-06:
@me-no-dev no problem at all, i know you guys are busy. fingers crossed for no reproduction :)