espressif/arduino-esp32 #10812: ETH_LAN8720/Ethernet sample not working when using with Arduino as ESP IDF

dzungpv, 2025-01-06:
### Board

ESP32 Dev Module

### Device Description

ESP32 Dev Module wire to LAN8720 module

### Hardware Configuration

Standard LAN 8720 config with ETH_CLK_MODE  ETH_CLOCK_GPIO17_OUT

### Version

v3.1.0

### IDE Name

VS, ESP IDF

### Operating System

Mac OSX 14

### Flash frequency

40mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Working when using Arduino IDE but not working when using as ESP IDF project

### Sketch

```cpp
#include "Arduino.h"

static bool eth_connected = false;
// Important to be defined BEFORE including ETH.h for ETH.begin() to work.
// Example RMII LAN8720 (Olimex, etc.)

#ifndef ETH_PHY_MDC
#define ETH_PHY_TYPE ETH_PHY_LAN8720
#endif

#include <ETH.h>
#define ETH_PHY_ADDR  1
#define ETH_PHY_MDC   23
#define ETH_PHY_MDIO  18
#define ETH_PHY_POWER -1
#define ETH_CLK_MODE  ETH_CLOCK_GPIO17_OUT

// #include "config.h"
// WARNING: onEvent is called from a separate FreeRTOS task (thread)!
void onEvent(arduino_event_id_t event) {
  switch (event) {
    case ARDUINO_EVENT_ETH_START:
      Serial.println("ETH Started");
      // The hostname must be set after the interface is started, but needs
      // to be set before DHCP, so set it from the event handler thread.
      ETH.setHostname("esp32-ethernet");
      break;
    case ARDUINO_EVENT_ETH_CONNECTED: Serial.println("ETH Connected"); break;
    case ARDUINO_EVENT_ETH_GOT_IP:
      Serial.println("ETH Got IP");
      Serial.println(ETH);
      eth_connected = true;
      break;
    case ARDUINO_EVENT_ETH_LOST_IP:
      Serial.println("ETH Lost IP");
      eth_connected = false;
      break;
    case ARDUINO_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case ARDUINO_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default: break;
  }
}

void testClient(const char *host, uint16_t port) {
  Serial.print("\nconnecting to ");
  Serial.println(host);

  NetworkClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup() {
  Serial.begin(115200);
  Network.onEvent(onEvent);
  ETH.begin();
}

void loop() {
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}
```

### Debug Message

```plain
I (31) boot: ESP-IDF v5.3.2 2nd stage bootloader
I (31) boot: compile time Jan  6 2025 13:26:13
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.1
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1319ch ( 78236) map
I (131) esp_image: segment 1: paddr=000231c4 vaddr=3ffb0000 size=024dch (  9436) load
I (135) esp_image: segment 2: paddr=000256a8 vaddr=40080000 size=0a970h ( 43376) load
I (154) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=38af8h (232184) map
I (234) esp_image: segment 4: paddr=00068b20 vaddr=4008a970 size=04448h ( 17480) load
I (248) boot: Loaded app from partition at offset 0x10000
I (248) boot: Disabling RNG early entropy source...
I (261) cpu_start: Multicore app
I (269) cpu_start: Pro cpu start user code
I (269) cpu_start: cpu freq: 160000000 Hz
I (269) app_init: Application information:
I (272) app_init: Project name:     main
I (277) app_init: App version:      02ae9e6-dirty
I (282) app_init: Compile time:     Jan  6 2025 13:26:01
I (288) app_init: ELF file SHA256:  34e6c5c8b...
I (293) app_init: ESP-IDF:          v5.3.2
I (298) efuse_init: Min chip rev:     v0.0
I (303) efuse_init: Max chip rev:     v3.99 
I (308) efuse_init: Chip rev:         v3.1
I (313) heap_init: Initializing. RAM available for dynamic allocation:
I (320) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (326) heap_init: At 3FFB4110 len 0002BEF0 (175 KiB): DRAM
I (332) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (338) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (345) heap_init: At 4008EDB8 len 00011248 (68 KiB): IRAM
I (353) spi_flash: detected chip: generic
I (356) spi_flash: flash io: dio
W (360) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (374) main_task: Started on CPU0
I (378) main_task: Calling app_main()
```

### Other Steps to Reproduce

I test https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino example with only change:
```
#define ETH_CLK_MODE  ETH_CLOCK_GPIO0_IN
```
to
```
#define ETH_CLK_MODE  ETH_CLOCK_GPIO17_OUT
```
On the Arduino IDE, I can build flash and it work, show IP and html code of google.com
But when using Arduino as ESP IDF it not working. Logs above. I user hello_world esp idf sample change the sketch above.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-01-06:
The config that is used in Arduino for ETH is the following:
```
CONFIG_ETH_ENABLED=y
CONFIG_ETH_USE_ESP32_EMAC=y
CONFIG_ETH_PHY_INTERFACE_RMII=y
CONFIG_ETH_RMII_CLK_INPUT=y
CONFIG_ETH_RMII_CLK_IN_GPIO=0
CONFIG_ETH_DMA_BUFFER_SIZE=512
CONFIG_ETH_DMA_RX_BUFFER_NUM=10
CONFIG_ETH_DMA_TX_BUFFER_NUM=10
```

Try with that instead

dzungpv, 2025-01-06:
> The config that is used in Arduino for ETH is the following:
>
> ```
> CONFIG_ETH_ENABLED=y
> CONFIG_ETH_USE_ESP32_EMAC=y
> CONFIG_ETH_PHY_INTERFACE_RMII=y
> CONFIG_ETH_RMII_CLK_INPUT=y [...]

But my custom hardware working with internal clock of ESP32 pin 17.

So it must be 

```
CONFIG_ETH_RMII_CLK_OUTPUT=y
CONFIG_ETH_RMII_CLK_OUT_GPIO=17
```

Instead of

```
CONFIG_ETH_RMII_CLK_INPUT=y
CONFIG_ETH_RMII_CLK_IN_GPIO=0
```

And the 
```
CONFIG_ETH_DMA_BUFFER_SIZE=512
CONFIG_ETH_DMA_RX_BUFFER_NUM=10
CONFIG_ETH_DMA_TX_BUFFER_NUM=10
```
Are default value, still not working when building with ESP IDF. As mention above, I modified default Arduino sample with my hardware and it working with Arduino IDE and the config work fine with native IDF example.

dzungpv, 2025-01-06:
I turn debug level and the code show this logs, may be you can find what cause the error

```
D (622) cpu_start: calling init function: 0x400d58d4 on core: 0
--- 0x400d58d4: __esp_system_init_fn_esp_timer_init_nonos at /Users/vietdzung/esp/esp-idf-v5.3.x/components/esp_timer/src/esp_timer_init.c:28

D (628) cpu_start: calling init function: 0x400d518c on core: 0
--- 0x400d518c: __esp_system_init_fn_init_newlib at /Users/vietdzung/esp/esp-idf-v5.3.x/components/newlib/newlib_init.c:162

D (634) cpu_start: calling init function: 0x400d2898 on core: 0
--- 0x400d2898: __esp_system_init_fn_init_brownout at /Users/vietdzung/esp/esp-idf-v5.3.x/components/esp_system/startup_funcs.c:67

D (640) intr_alloc: Connected src 46 to int 1 (cpu 0)
D (645) cpu_start: calling init function: 0x400d28a4 on core: 0
--- 0x400d28a4: __esp_system_init_fn_init_newlib_time at /Users/vietdzung/esp/esp-idf-v5.3.x/components/esp_system/startup_funcs.c:81

D (651) cpu_start: calling init function: 0x400d6adc on core: 0
--- 0x400d6adc: __esp_system_init_fn_init_vfs_uart at /Users/vietdzung/esp/esp-idf-v5.3.x/components/esp_driver_uart/src/uart_vfs.c:1080

D (657) cpu_start: calling init function: 0x400d90e8 on core: 0
--- 0x400d90e8: __esp_system_init_fn_init_vfs_console at /Users/vietdzung/esp/esp-idf-v5.3.x/components/esp_vfs_console/vfs_console.c:222

D (663) cpu_start: calling init function: 0x400d51f8 on core: 0
--- 0x400d51f8: __esp_system_init_fn_init_newlib_stdio at /Users/vietdzung/esp/esp-idf-v5.3.x/components/newlib/newlib_init.c:207

D (669) cpu_start: calling init function: 0x400d55b0 on core: 0
--- 0x400d55b0: __esp_system_init_fn_init_pthread at /Users/vietdzung/esp/esp-idf-v5.3.x/components/pthread/pthread.c:76

D (675) cpu_start: calling init function: 0x400d28b0 on core: 0
--- 0x400d28b0: __esp_system_init_fn_init_flash at /Users/vietdzung/esp/esp-idf-v5.3.x/components/esp_system/startup_funcs.c:88

D (681) spi_flash: trying chip: issi
D (684) spi_flash: trying chip: gd
D (688) spi_flash: trying chip: mxic
D (692) spi_flash: trying chip: winbond
D (695) spi_flash: trying chip: generic
I (699) spi_flash: detected chip: generic
I (704) spi_flash: flash io: dio
W (708) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
D (721) cpu_start: calling init function: 0x400d27cc on core: 0
--- 0x400d27cc: __esp_system_init_fn_init_efuse at /Users/vietdzung/esp/esp-idf-v5.3.x/components/efuse/src/esp_efuse_startup.c:137

D (727) cpu_start: calling init function: 0x400f8ae8
--- 0x400f8ae8: _GLOBAL__sub_I__ZN17__eh_globals_init7_S_initE at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_globals.cc:162

D (732) cpu_start: calling init function: 0x400f8898
--- 0x400f8898: _GLOBAL__sub_I__ZN9__gnu_cxx9__freeresEv at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_alloc.cc:462

D (737) C++ init: Setting C++ exception emergency pool to 0.
D (743) cpu_start: calling init function: 0x400f7f90
--- 0x400f7f90: esp_clock_output_pin_ctrl_init at /Users/vietdzung/esp/esp-idf-v5.3.x/components/esp_hw_support/esp_clock_output.c:237

D (748) cpu_start: calling init function: 0x400e0998
--- 0x400e0998: _GLOBAL__sub_I__ZN9IPAddressC2Ev at /Users/vietdzung/dev/projects/esp32/lan8720_test/managed_components/espressif__arduino-esp32/cores/esp32/IPAddress.cpp:453

D (753) cpu_start: calling init function: 0x400e0330
--- 0x400e0330: _GLOBAL__sub_I__ZN8SPIClassC2Eh at /Users/vietdzung/dev/projects/esp32/lan8720_test/managed_components/espressif__arduino-esp32/libraries/SPI/src/SPI.cpp:334

D (758) cpu_start: calling init function: 0x400df4bc
--- 0x400df4bc: _GLOBAL__sub_I__ZN14NetworkManagerC2Ev at /Users/vietdzung/dev/projects/esp32/lan8720_test/managed_components/espressif__arduino-esp32/libraries/Network/src/NetworkManager.cpp:200

D (763) cpu_start: calling init function: 0x400de5d8
--- 0x400de5d8: _GLOBAL__sub_I__ZNK8ETHClass6handleEv at /Users/vietdzung/dev/projects/esp32/lan8720_test/managed_components/espressif__arduino-esp32/libraries/Ethernet/src/ETH.cpp:1059

D (768) cpu_start: calling init function: 0x400de2bc
--- 0x400de2bc: _GLOBAL__sub_I__ZN6StringC2EPKc at /Users/vietdzung/dev/projects/esp32/lan8720_test/managed_components/espressif__arduino-esp32/cores/esp32/WString.cpp:938

D (773) cpu_start: calling init function: 0x400ddd34
--- 0x400ddd34: _GLOBAL__sub_I_Serial0 at /Users/vietdzung/dev/projects/esp32/lan8720_test/managed_components/espressif__arduino-esp32/cores/esp32/HardwareSerial.cpp:616

[... 15 lines cut ...]

D (811) intr_alloc: Connected src 24 to int 3 (cpu 0)
D (815) app_start: Starting scheduler on CPU0
D (820) intr_alloc: Connected src 25 to int 1 (cpu 1)
D (825) app_start: Starting scheduler on CPU1
I (820) main_task: Started on CPU0
D (834) heap_init: New heap initialised at 0x3ffe0440
D (838) heap_init: New heap initialised at 0x3ffe4350
D (844) intr_alloc: Connected src 16 to int 5 (cpu 0)
I (849) main_task: Calling app_main()
```
I have a Waveshare module using external crystal and it work with default setting of Arduino, you send above

```
CONFIG_ETH_ENABLED=y
CONFIG_ETH_USE_ESP32_EMAC=y
CONFIG_ETH_PHY_INTERFACE_RMII=y
CONFIG_ETH_RMII_CLK_INPUT=y
CONFIG_ETH_RMII_CLK_IN_GPIO=0
CONFIG_ETH_DMA_BUFFER_SIZE=512
CONFIG_ETH_DMA_RX_BUFFER_NUM=10
CONFIG_ETH_DMA_TX_BUFFER_NUM=10
```

But it only work in Arduino IDE, not as ESP IDF component, same with my custom hardware.

me-no-dev, 2025-01-06:
can you please add some serial prints to setup and enable Arduino debug level to "Verbose" ?

dzungpv, 2025-01-06:
> can you please add some serial prints to setup and enable Arduino debug level to "Verbose" ?

This is verbose log level on Arduino IDE:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4916
load:0x40078000,len:16436
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3524
entry 0x400805b8
[     1][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d99b4
[    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d9984
[    26][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d9954
[    39][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d9924
[    53][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d99b4
[    66][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d9984
[    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d9954
[    93][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d9924
[   107][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d99b4
[   120][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d9984
[   134][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d9954
[   147][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d9924
[   163][D][esp32-hal-cpu.c:263] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   177][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 3 successfully set to type UART_RX (2) with bus 0x3ffbdb60
[   187][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 1 successfully set to type UART_TX (3) with bus 0x3ffbdb60
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q5
  Revision          : 3.01
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000032
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   372148 B ( 363.4 KB)
  Free Bytes        :   332708 B ( 324.9 KB)
  Allocated Bytes   :    32296 B (  31.5 KB)
  Minimum Free Bytes:   327196 B ( 319.5 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
[... 449 lines cut ...]
D (1311) app_start: Starting scheduler on CPU0
V (1315) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (1321) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x40E
 (1329) intr_alloc: Connected src 25 to int 1 (cpu 1)
D (1335) app_start: Starting scheduler on CPU1
;32mI (1315) main_task: Started on CPU0
D (1344) heap_init: New heap initialised at 0x3ffe0440
D (1348) heap_init: New heap initialised at 0x3ffe4350
V (1353) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1360) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (1368) intr_alloc: Connected src 16 to int 5 (cpu 0)
I (1373) main_task: Calling app_main()
V (1378) partition: Loading the partition table
V (1383) mmap: actual_mapped_len is 0x10000
V (1387) calculated md5: 0x3ffb62e8   f4 ad 4f 45 38 56 4b 5d  74 35 b6 2c 75 b6 95 24  |..OE8VK]t5.,u..$|
V (1396) stored md5: 0x3f428070   f4 ad 4f 45 38 56 4b 5d  74 35 b6 2c 75 b6 95 24  |..OE8VK]t5.,u..$|
V (1406) partition: Partition table MD5 verified
I (1420) main_task: Returned from app_main()
I (1421) uart: queue free spaces: 20
V (1422) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (1426) intr_alloc: esp_in

```

Seem ETH driver not init.

me-no-dev, 2025-01-06:
In IDF enable Arduino's debug log too: https://github.com/espressif/arduino-esp32/blob/master/Kconfig.projbuild#L170

If log does not change, then Arduino is not even starting

dzungpv, 2025-01-06:
> In IDF enable Arduino's debug log too: https://github.com/espressif/arduino-esp32/blob/master/Kconfig.projbuild#L170
> 
> If log does not change, then Arduino is not even starting

Here the log, seem it hang:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1532
ho 0 tail 12 room 4
load:0x40078000,len:14480
load:0x40080400,len:4
--- 0x40080400: _init at ??:?

load:0x40080404,len:3348
entry 0x40080588
V (147) mmap: after coalescing, 3 regions are left
I (147) cpu_start: Multicore app
D (147) cpu_start: Pro cpu up
D (147) cpu_start: Starting app cpu, entry point is 0x4008129c
--- 0x4008129c: call_start_cpu1 at /Users/vietdzung/esp/esp-idf-v5.3.x/components/esp_system/port/cpu_start.c:191

D (0) cpu_start: App cpu up
D (161) clk: RTC_SLOW_CLK calibration value: 3153830
D (173) cpu_start: calling init function: 0x400d28b0 on core: 0
--- 0x400d28b0: __esp_system_init_fn_init_efuse_check at /Users/vietdzung/esp/esp-idf-v5.3.x/components/efuse/src/esp_efuse_startup.c:31

D (173) cpu_start: calling init function: 0x400d29d8 on core: 0
--- 0x400d29d8: __esp_system_init_fn_init_show_cpu_freq at /Users/vietdzung/esp/esp-idf-v5.3.x/components/esp_system/startup_funcs.c:58

I (175) cpu_start: Pro cpu start user code
I (180) cpu_start: cpu freq: 240000000 Hz
D (185) cpu_start: calling init function: 0x400d27f0 on core: 0
--- 0x400d27f0: __esp_system_init_fn_init_show_app_info at /Users/vietdzung/esp/esp-idf-v5.3.x/components/esp_app_format/esp_app_desc.c:106

I (191) app_init: Application information:
I (196) app_init: Project name:     main
I (200) app_init: App version:      02ae9e6-dirty
I (206) app_init: Compile time:     Jan  6 2025 17:08:04
I (212) app_init: ELF file SHA256:  da92bebd4...
I (217) app_init: ESP-IDF:          v5.3.2-282-gcfea4f7c98
D (223) cpu_start: calling init function: 0x400d28c4 on core: 0
--- 0x400d28c4: __esp_system_init_fn_init_efuse_show_app_info at /Users/vietdzung/esp/esp-idf-v5.3.x/components/efuse/src/esp_efuse_startup.c:41

I (229) efuse_init: Min chip rev:     v0.0
I (234) efuse_init: Max chip rev:     v3.99 
I (239) efuse_init: Chip rev:         v3.1
D (244) cpu_start: calling init function: 0x400d4404 on core: 0
--- 0x400d4404: __esp_system_init_fn_init_heap at /Users/vietdzung/esp/esp-idf-v5.3.x/components/heap/heap_caps_init.c:25

V (250) memory_layout: reserved range is 0x3f414b8c - 0x3f414bd4
D (256) memory_layout: Checking 9 reserved memory ranges:
D (261) memory_layout: Reserved memory range 0x3ff82000 - 0x3ff82000
D (267) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (274) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb41f8
D (280) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (287) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (293) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (300) memory_layout: Reserved memory range 0x40078000 - 0x40080000
--- 0x40080000: _WindowOverflow4 at /Users/vietdzung/esp/esp-idf-v5.3.x/components/xtensa/xtensa_vectors.S:2027

D (306) memory_layout: Reserved memory range 0x40080000 - 0x4008ef68
--- 0x40080000: _WindowOverflow4 at /Users/vietdzung/esp/esp-idf-v5.3.x/components/xtensa/xtensa_vectors.S:2027

D (312) memory_layout: Reserved memory range 0x50001fe8 - 0x50002000
D (319) memory_layout: Building list of available memory regions:
V (325) memory_layout: Examining memory region 0x3ffae000 - 0x3ffb0000
[... 376 lines cut ...]
  Arduino Events on : 1
  CDC On Boot       : 0
============ Before Setup End ============
[  1028][V][esp32-hal-uart.c:421] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[  1037][V][esp32-hal-uart.c:510] uartBegin(): UART0 not installed. Starting installation
I (2078) uart: queue free spaces: 20
V (2078) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (2078) intr_alloc: esp_[  1062][V][esp32-hal-uart.c:575] uartBegin(): UART0 initialization done.
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   384848 B ( 375.8 KB)
  Free Bytes        :   355684 B ( 347.3 KB)
  Allocated Bytes   :    25352 B (  24.8 KB)
  Minimum Free Bytes:   353440 B ( 345.2 KB)
  Largest Free Block:   155648 B ( 152.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     1 : UART_TX[0]
     3 : UART_RX[0]
============ After Setup End =============
```

me-no-dev, 2025-01-06:
OK. The problem is in your example code. Pin definitions MUST be BEFORE include of ETH.h
```cpp
#define ETH_PHY_TYPE ETH_PHY_LAN8720
#define ETH_PHY_ADDR  1
#define ETH_PHY_MDC   23
#define ETH_PHY_MDIO  18
#define ETH_PHY_POWER -1
#define ETH_CLK_MODE  ETH_CLOCK_GPIO17_OUT
#include <ETH.h>
//...
```

me-no-dev, 2025-01-06:
This is also wrong, because MDC is defined in the file, so makes the type undefined
```cpp
#ifndef ETH_PHY_MDC
#define ETH_PHY_TYPE ETH_PHY_LAN8720
#endif
```

dzungpv, 2025-01-06:
> OK. The problem is in your example code. Pin definitions MUST be BEFORE include of ETH.h
> 
> ```c++
> #define ETH_PHY_TYPE ETH_PHY_LAN8720
> #define ETH_PHY_ADDR  1
> #define ETH_PHY_MDC   23
> #define ETH_PHY_MDIO  18
> #define ETH_PHY_POWER -1
> #define ETH_CLK_MODE  ETH_CLOCK_GPIO17_OUT
> #include <ETH.h>
> //...
> ```

Yes, it work now. It is your example, not mine, but it optimize for Arduino IDE only not ESP IDF.

me-no-dev, 2025-01-06:
>  It is your example, not mine

Maybe check again?
https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino#L8C1-L25C17

closed 2025-01-06 by me-no-dev

dzungpv, 2025-01-06:
> > It is your example, not mine
> 
> Maybe check again? https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino#L8C1-L25C17

Yes, I just change the clock mode, "your" I mean your company code not your personal code. If using directive like this:

```
#define ETH_PHY_TYPE ETH_PHY_LAN8720

#if CONFIG_IDF_TARGET_ESP32
#define ETH_PHY_ADDR  1
#define ETH_PHY_MDC   23
#define ETH_PHY_MDIO  18
#define ETH_PHY_POWER -1
#define ETH_CLK_MODE  ETH_CLOCK_GPIO17_OUT
#elif CONFIG_IDF_TARGET_ESP32P4
#define ETH_PHY_ADDR  0
#define ETH_PHY_MDC   31
#define ETH_PHY_MDIO  52
#define ETH_PHY_POWER 51
#define ETH_CLK_MODE  EMAC_CLK_EXT_IN
#endif
#include <ETH.h>
```
Will not work in ESP IDF. Must be:

```
#define ETH_PHY_TYPE ETH_PHY_LAN8720
#define ETH_PHY_ADDR  1
#define ETH_PHY_MDC   23
#define ETH_PHY_MDIO  18
#define ETH_PHY_POWER -1
#define ETH_CLK_MODE  ETH_CLOCK_GPIO17_OUT
#include <ETH.h>
```

I still don't know why, but it just work.

dzungpv, 2025-01-06:
Update, seem some cache problem, now it work with directive too. Thank your for the support