espressif/arduino-esp32 #10822: Redefined define with lolin_s3_mini

BCsabaEngine, 2025-01-07:
### Board

ESp32 Lolin S3 mini

### Device Description

default

### Hardware Configuration

default

### Version

v3.1.0

### IDE Name

VScode, PlatformIO

### Operating System

macOS

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I got a warning (and can build without -Werror, but I like this option) when use a board that contains RGB led:

```
.platformio/packages/framework-arduinoespressif32/variants/lolin_s3_mini/pins_arduino.h:17: error: "RGB_BUILTIN_LED_COLOR_ORDER" redefined

.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-rgb-led.h:15
```
The cause of the error is [pins_arduino.h](https://github.com/espressif/arduino-esp32/blob/master/variants/lolin_s3_mini/pins_arduino.h) for the lolin_s3_mini variant.

How can I include these files in proper order to make the #ifndef works?

### Sketch

```cpp
Any sketch
```

### Debug Message

```plain
No debug, compile warning
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-01-08:
I think that @Jason2866 may help with it.

Jason2866, 2025-01-08:
@BCsabaEngine  Please post your Platformio.ini

Jason2866, 2025-01-08:
@SuGlider I think the reason for is here https://github.com/espressif/arduino-esp32/blob/c23c78674995079a1c49ceae3b1e6ab9fdafb454/variants/lolin_s3_mini/pins_arduino.h#L17

BCsabaEngine, 2025-01-08:
Here is my platformio:

```
[env:default]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
board = lolin_s3_mini
framework = arduino
board_build.partitions = min_spiffs.csv
upload_speed = 921600
monitor_speed = 115200
custom_master_version = 1.5.0
build_flags = 
	-DCORE_DEBUG_LEVEL=0
	; -DEMULATE_ALLPANELS
	-DMASTER_VERSION=${this.custom_master_version}
	-DARDUINO_USB_MODE=1
	-DARDUINO_USB_CDC_ON_BOOT=1
	-DARDUINO_USB_MSC_ON_BOOT=0
	-DARDUINO_USB_DFU_ON_BOOT=0
	-DNDEBUG
	-Wall
	-Werror
	-Wextra
	-Wno-error=deprecated-enum-enum-conversion
	-Wlogical-op
	-Wnonnull
	-Wunused-parameter
	-Wuninitialized
	-Wno-free-nonheap-object
extra_scripts = pre:pre_build.py
lib_deps = 
	SPI
	Wire
	adafruit/Adafruit BusIO
	adafruit/Adafruit GFX Library
	adafruit/Adafruit SSD1306
	adafruit/Adafruit INA219
	bblanchon/ArduinoJson
	https://github.com/BCsabaEngine/TerminalWindow
	https://github.com/me-no-dev/ESPAsyncWebServer
check_tool = clangtidy
check_skip_packages = yes
check_flags =
  clangtidy: --checks=*,-llvmlibc-*,-modernize-macro-to-enum,-*braces*,-*casting*
```

BCsabaEngine, 2025-01-08:
> @SuGlider I think the reason for is here
> 
> https://github.com/espressif/arduino-esp32/blob/c23c78674995079a1c49ceae3b1e6ab9fdafb454/variants/lolin_s3_mini/pins_arduino.h#L17

Yes I think an #ifndef missing here

SuGlider, 2025-01-10:
@BCsabaEngine - Where is redefined?
Arduino Core doesn't define it. It is defined only within lolin_s3_mini board `pins_arduino.h` file.

Is PaltformIO redefining it in the configuration?

SuGlider, 2025-01-10:
It already has the `#ifndef` guard in the code:

https://github.com/espressif/arduino-esp32/blob/99e68a0518f454d0a9769d1da4f95900700228b2/cores/esp32/esp32-hal-rgb-led.h#L14-L16

SuGlider, 2025-01-10:
It has been added in https://github.com/espressif/arduino-esp32/pull/10128

Jason2866, 2025-01-10:
I guess that the core file is first compiled and after that the variant header file. So the redefine happens

SuGlider, 2025-01-10:
> I guess that the core file is first compiled and after that the variant header file. So the redefine happens

But *.h is never compiled. It must be included by some C/CPP file.

SuGlider, 2025-01-10:
When there is a `-DRGB_BUILTIN_LED_COLOR_ORDER=XXX` in the gcc command line, it may cause such error.

SuGlider, 2025-01-10:
I hve searched Arduino code. There are just 3 places when `RGB_BUILTIN_LED_COLOR_ORDER` appears:

in `variants/lolin_s3_min/pins_arduino.h` as a direct `#define`
in `cores/esp32/esp32-hal-rgb-led.h`  as a guarded `#define` - therefore, it should not cause the issue
in `cores/esp32/esp32-hal-rgb-led.c` in the code by `rgbLedWrite()`

SuGlider, 2025-01-10:
> I guess that the core file is first compiled and after that the variant header file. So the redefine happens

The only way is when `cores/esp32/esp32-hal-rgb-led.h` is included before `variants/lolin_s3_min/pins_arduino.h`

SuGlider, 2025-01-10:
Ok... found. `Arduino.h` includes `esp32-hal.h` that includes `esp32-hal-rgb-led.h` and latter it includes `pins_arduino.h`.
This sequence generates the issue.

The order is wrong. It must be in the reverse order.

SuGlider, 2025-01-10:
@BCsabaEngine @Jason2866 - Please test PR #10841 and let me know. Thanks.

BCsabaEngine, 2025-01-11:
Tnak you for detect problem properly. I have tested the PR, I adopted to my codebase (not a oneline example app), and **I confirm, this solves the situation**

Thank you again!

PR #10841 (2025-01-13): Bugfix/include order
## Description of Change
The PR #10128 has added `RGB_BUILTIN_LED_COLOR_ORDER` as a possible parameter for RGB LED writing function.
There is an issue with the include order.

`cores/esp32/esp32-hal-rgb-led.h` shall receive `RGB_BUILTIN_LED_COLOR_ORDER` from `variants/lolin_s3_min/pins_arduino.h` and do not modify it in case it is already defined.

But there are a few places where the include order is `esp-hal.h` first and latter `pins_arduino.h`. 
Given that `esp-hal.h` includes `esp32-hal-rgb-led.h` this cases a warning as error:
`pins_arduino.h:17: error: "RGB_BUILTIN_L [... cut]
closed 2025-01-13 by me-no-dev