espressif/arduino-esp32 #10825: BasicHttpsClient example rootCACertificate is outdated

blueskyson, 2025-01-08:
### Board

ESP32 Dev Kit V1

### Device Description

ESP32 Dev Kit V1

### Hardware Configuration

UART0, UART2

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Ubuntu 24.04

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

The  rootCACertificate in [BasicHttpsClient example](https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/examples/BasicHttpsClient/BasicHttpsClient.ino) is outdated. After checking its SSL by

```
openssl s_client -showcerts -connect jigsaw.w3.org:443
```

It can be renewed to:
```cpp
const char *rootCACertificate = R"string_literal(
-----BEGIN CERTIFICATE-----
MIIDejCCAmKgAwIBAgIQf+UwvzMTQ77dghYQST2KGzANBgkqhkiG9w0BAQsFADBX
MQswCQYDVQQGEwJCRTEZMBcGA1UEChMQR2xvYmFsU2lnbiBudi1zYTEQMA4GA1UE
CxMHUm9vdCBDQTEbMBkGA1UEAxMSR2xvYmFsU2lnbiBSb290IENBMB4XDTIzMTEx
NTAzNDMyMVoXDTI4MDEyODAwMDA0MlowRzELMAkGA1UEBhMCVVMxIjAgBgNVBAoT
GUdvb2dsZSBUcnVzdCBTZXJ2aWNlcyBMTEMxFDASBgNVBAMTC0dUUyBSb290IFI0
MHYwEAYHKoZIzj0CAQYFK4EEACIDYgAE83Rzp2iLYK5DuDXFgTB7S0md+8Fhzube
Rr1r1WEYNa5A3XP3iZEwWus87oV8okB2O6nGuEfYKueSkWpz6bFyOZ8pn6KY019e
WIZlD6GEZQbR3IvJx3PIjGov5cSr0R2Ko4H/MIH8MA4GA1UdDwEB/wQEAwIBhjAd
BgNVHSUEFjAUBggrBgEFBQcDAQYIKwYBBQUHAwIwDwYDVR0TAQH/BAUwAwEB/zAd
BgNVHQ4EFgQUgEzW63T/STaj1dj8tT7FavCUHYwwHwYDVR0jBBgwFoAUYHtmGkUN
l8qJUC99BM00qP/8/UswNgYIKwYBBQUHAQEEKjAoMCYGCCsGAQUFBzAChhpodHRw
Oi8vaS5wa2kuZ29vZy9nc3IxLmNydDAtBgNVHR8EJjAkMCKgIKAehhxodHRwOi8v
Yy5wa2kuZ29vZy9yL2dzcjEuY3JsMBMGA1UdIAQMMAowCAYGZ4EMAQIBMA0GCSqG
SIb3DQEBCwUAA4IBAQAYQrsPBtYDh5bjP2OBDwmkoWhIDDkic574y04tfzHpn+cJ
odI2D4SseesQ6bDrarZ7C30ddLibZatoKiws3UL9xnELz4ct92vID24FfVbiI1hY
+SW6FoVHkNeWIP0GCbaM4C6uVdF5dTUsMVs/ZbzNnIdCp5Gxmx5ejvEau8otR/Cs
kGN+hr/W5GvT1tMBjgWKZ1i4//emhA1JG1BbPzoLJQvyEotc03lXjTaCzv8mEbep
8RqZ7a2CPsgRbuvTPBwcOMBBmuFeU88+FSBX6+7iP0il8b4Z0QFqIwwMHfs/L6K1
vepuoxtGzi4CZ68zJpiq1UvSqTbFJjtbD4seiMHl
-----END CERTIFICATE-----
)string_literal";
```
[... 249 lines cut ...]
     1 : UART_TX[0]
     3 : UART_RX[0]
============ After Setup End =============
[HTTPS] begin...
[  4827][D][HTTPClient.cpp:293] beginInternal(): protocol: https, host: jigsaw.w3.org port: 443 url: /HTTP/connection.html
[HTTPS] GET...
[  4839][D][HTTPClient.cpp:574] sendRequest(): request type: 'GET' redirCount: 0

[  4847][D][NetworkManager.cpp:83] hostByName(): Clearing DNS cache
[  4857][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 104.18.22.19
[  5945][E][ssl_client.cpp:36] _handle_error(): [ssl_starttls_handshake():313]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[  5960][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -9984
[  5969][D][HTTPClient.cpp:1105] connect(): failed connect to jigsaw.w3.org:443
[  5977][W][HTTPClient.cpp:1421] returnError(): error(-1): connection refused
[  5984][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[HTTPS] GET... failed, error: connection refused
[  5992][E][NetworkClient.cpp:319] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[  6004][D][HTTPClient.cpp:393] disconnect(): tcp is closed

Waiting 10s before the next round...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-01-08:
Hi @blueskyson ! Do you want to create a PR and update the certificate in the example?

blueskyson, 2025-01-09:
@me-no-dev Sure.

PR #10911 (2025-01-30): fix: Update the root certificate for the BasicHttpsClient example
The certificate is outdated. The current certificate can be found by `openssl s_client -showcerts -connect jigsaw.w3.org:443`.

## Tests scenarios
I've tested on an ESP32 DEVKIT V1

Close #10825.
closed 2025-01-30 by me-no-dev