espressif/arduino-esp32 #10826: OTA by websocket

tmwckora, 2025-01-08:
### Related area

websocket, OTA

### Hardware specification

ESP32 Dev Module

### Is your feature request related to a problem?

I modified "Update.cpp" as below. I just add printf() for output key var.

size_t UpdateClass::write(uint8_t *data, size_t len) {
  if (hasError() || !isRunning()) {
   Serial.printf("[Update]:hasError\n");
    return 0;
  }

  if (len > remaining()) {
    _abort(UPDATE_ERROR_SPACE);
    Serial.printf("[Update]:SPACE\n");
    return 0;
  }

  size_t left = len;
  Serial.printf("[Update]: _bufferLen=%d;left=%d;SEC=%d;\n",_bufferLen,len,SPI_FLASH_SEC_SIZE);

  while ((_bufferLen + left) > SPI_FLASH_SEC_SIZE) {
    size_t toBuff = SPI_FLASH_SEC_SIZE - _bufferLen;
    memcpy(_buffer + _bufferLen, data + (len - left), toBuff);
    _bufferLen += toBuff;
    if (!_writeBuffer()) {
     Serial.printf("[Update1]: _bufferLen=%d; toBuff=%d; left=%d\n",_bufferLen,toBuff,left);
      return len - left;
    }
    left -= toBuff;
  }
  memcpy(_buffer + _bufferLen, data + (len - left), left);
  _bufferLen += left;
  if (_bufferLen == remaining()) {
    if (!_writeBuffer()) {
     Serial.printf("[Update2]: _bufferLen=%d; left=%d\n",_bufferLen,left);
      return len - left;
    }
  }
  Serial.printf("[Update3]: _bufferLen=%d; left=%d\n",_bufferLen,left);
  return len;
}

Serial Monitor:
11:07:08.062 -> [WSc] get text: {"fileName": "G3-V1.0.0.bin"}
11:07:08.128 -> Update Start!
11:07:08.128 -> RX =0, 0
11:07:08.128 -> RX =0, 0
11:07:08.128 -> [Update]: _bufferLen=0;left=4096;SEC=4096;
11:07:08.128 -> [Update3]: _bufferLen=4096; left=4096
11:07:08.128 -> [readCb] update 4096 bytes
11:07:08.128 -> RX =10, 0
11:07:08.128 -> RX =0, 0
11:07:08.128 -> [Update]: _bufferLen=4096;left=4096;SEC=4096;
11:07:08.128 -> [Update1]: _bufferLen=0; toBuff=0; left=4096
11:07:08.128 -> [readCb] Error updating:S=4096, W=0

Question:
Update.write can write 1st packet(4KB), but when write 2nd packet occur error as upon.

### Describe the solution you'd like

How solve this issue?
Need extra initial?
Below is my Update.begin code. And I execute Update.write when receive BIN code.

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {

	switch(type) {
		case WStype_DISCONNECTED:
			Serial.printf("[WSc] Disconnected!\n");
			break;
		case WStype_CONNECTED:
			Serial.printf("[WSc] Connected to url: %s\n", payload);

			// send message to server when Connected
			//webSocket.sendTXT("Connected");
			break;
		case WStype_TEXT:
			Serial.printf("[WSc] get text: %s\n", payload);
      if (StrCmpr(payload , ActCheck)) {
        if (Update.begin(UPDATE_SIZE_UNKNOWN)) {
          UpdateStart=1;
          SendDataTic.detach();  
          Serial.println("Update Start!");
        }
        else {
          Serial.println("Error updating firmware");
        }
      }

			// send message to server
			// webSocket.sendTXT("message here");
			break;
		case WStype_BIN:
			Serial.printf("[WSc] get binary length: %u\n", length);
			break;
		case WStype_ERROR:			
		case WStype_FRAGMENT_TEXT_START:
		case WStype_FRAGMENT_BIN_START:
		case WStype_FRAGMENT:
		case WStype_FRAGMENT_FIN:
			break;
	}

}

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2025-01-08:
Hi @tmwckora . What version of the core is this? Can you please post a more complete example code that would allow us to reproduce? From what you have given, there are many parts missing and is not very clear what is going on

tmwckora, 2025-01-13:
> Hi @tmwckora . What version of the core is this? Can you please post a more complete example code that would allow us to reproduce? From what you have given, there are many parts missing and is not very clear what is going on

#

closed 2025-01-13 by tmwckora