espressif/arduino-esp32 #10828: playWAV [esp32-hal-periman.c:122] perimanSetPinBus(): Invalid pin: 255

lmabbe-lm, 2025-01-08:
### Board

esp32s3-wroom-1u-n16r8

### Device Description

I put the .wav file in flash, then read it using LittleFS, and played the file using i2s's playWAV. 
It prompted esp32-hal-periman.c:122] perimanSetPinBus(): Invalid pin: 255, using max98357A.

音频文件放在了压缩包里面
[归档.zip](https://github.com/user-attachments/files/18344087/default.zip)

### Hardware Configuration

#define SPEACKER_LRC_NUM 1
#define SPEACKER_CLK_NUM 2
#define SPEACKER_DATA_NUM 42

### Version

v3.1.0

### IDE Name

vscode+pio

### Operating System

mac

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

[D][ESP_I2S.cpp:1011] playWAV(): Play WAV: rate:16000, bits:16, channels:1, size:51840
[ 14499][E][esp32-hal-periman.c:122] perimanSetPinBus(): Invalid pin: 255

### Sketch

```cpp
#include <Arduino.h>
#include "FS.h"
#include <LittleFS.h>
#include <ESP_I2S.h>

// hardware hw;

#define SAMPLE_RATE 16000

#define SPEACKER_LRC_NUM 1
#define SPEACKER_CLK_NUM 2
#define SPEACKER_DATA_NUM 42

I2SClass speaker1;
void setup()
{
    Serial.begin(115200);
    Serial.println("Starting...");

    // if (!hw.init())
    // {
    //     Serial.println("Hardware init failed!");
    //     while (1)
    //     {
    //         delay(1000);
    //     }
    // }

    if (!LittleFS.begin(true, "/littlefs", 10, "storage"))
    {
        Serial.println("LittleFS Mount Failed");
        return;
    }

    File file = LittleFS.open("/hijeson.wav");
    if (!file)
    {
        Serial.println("Failed to open file for reading");
        return;
    }

    // hw.ledOn();
    int32_t size = file.size(); // 使用 uint32_t 存储文件大小
    // 分配内存
    uint8_t *buffer = (uint8_t *)malloc(size);
    if (buffer == nullptr)
    {
        Serial.println("Failed to allocate memory");
        // 处理内存分配失败
        return;
    }
    // 读取文件
    if (file.readBytes((char *)buffer, size) != size)
    {
        // 处理文件读取失败
        free(buffer); // 释放内存
        return;
    }

    file.close(); // 关闭文件

    speaker1.setPins(SPEACKER_CLK_NUM, SPEACKER_LRC_NUM, SPEACKER_DATA_NUM);
    bool res = speaker1.begin(I2S_MODE_STD, SAMPLE_RATE, I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_MONO, I2S_STD_SLOT_RIGHT);
    if (!res)
    {
        log_e("hardware speaker init failed");
    }
    Serial.println("hardware speaker init success");
    // 打印 buffer 内容
    Serial.println("Buffer content:");
    for (int i = 0; i < size; i++)
    {
        Serial.print(buffer[i], HEX); // 以十六进制打印每个字节
        Serial.print(" ");
        if ((i + 1) % 16 == 0) // 每16个字节换行
        {
            Serial.println();
        }
    }
    // 播放音频
    speaker1.playWAV(buffer, size);

    // speaker1.playMP3(buffer, size);
    speaker1.end();
}

void loop()
{
    delay(1000);
}

--- pio config---
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
upload_speed = 921600
monitor_speed = 115200
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
build_flags = 
	"-DBOARD_HAS_PSRAM"
	"-DCORE_DEBUG_LEVEL=4"
	"-DEI_CLASSIFIER_ALLOCATION_STATIC"
board_build.arduino.memory_type = qio_opi
board_build.flash_mode = qio
board_upload.flash_size = 16MB
board_build.filesystem = littlefs
board_build.partitions = my_partition.csv
lib_deps = 
	fastled/FastLED@^3.9.9

--- my partition.csv----
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     ,        0x6000,
phy_init, data, phy,     ,        0x1000,
factory,  app,  factory, ,        2M,
storage,  data, littlefs, ,       8M,
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1188
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbf0
load:0x403cb700,len:0x30e4
entry 0x403c88ac
E (195) esp_core_dump_flash: No core dump partition found!
E (195) esp_core_dump_flash: No core dump partition found!
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 0.02
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000012
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   360268 B ( 351.8 KB)
  Free Bytes        :   324108 B ( 316.5 KB)
  Allocated Bytes   :    31192 B (  30.5 KB)
  Minimum Free Bytes:   319380 B ( 311.9 KB)
  Largest Free Block:   258036 B ( 252.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  8388608 B (8192.0 KB)
  Free Bytes        :  8385672 B (8189.1 KB)
  Allocated Bytes   :      576 B (   0.6 KB)
  Minimum Free Bytes:  8385672 B (8189.1 KB)
  Largest Free Block:  8257524 B (8064.0 KB)
  Bus Mode          : OPI
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 16777216 B (16 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    24.0 KB, type: DATA, subtype: NVS
           phy_init : addr: 0x0000F000, size:     4.0 KB, type: DATA, subtype: PHY
            factory : addr: 0x00010000, size:  2048.0 KB, type:  APP, subtype: FACTORY
            storage : addr: 0x00210000, size:  8192.0 KB, type: DATA, subtype: LITTLEFS
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Jan  8 2025 13:59:29
  ESP-IDF Version   : v5.3.2-174-g083aad99cf-dirty
  Arduino Version   : 3.1.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
  Arduino Variant   : esp32s3
  Core Debug Level  : 4
  Arduino Runs Core : 1
  Arduino Events on : 1
  Arduino USB Mode  : 1
  CDC On Boot       : 0
============ Before Setup End ============
Starting...
hardware speaker init success
[   369][D][ESP_I2S.cpp:1011] playWAV(): Play WAV: rate:16000, bits:16, channels:1, size:51840
[  1989][E][esp32-hal-periman.c:122] perimanSetPinBus(): Invalid pin: 255
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   360268 B ( 351.8 KB)
  Free Bytes        :   321856 B ( 314.3 KB)
  Allocated Bytes   :    33172 B (  32.4 KB)
  Minimum Free Bytes:   316824 B ( 309.4 KB)
  Largest Free Block:   258036 B ( 252.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  8388608 B (8192.0 KB)
  Free Bytes        :  8331536 B (8136.3 KB)
  Allocated Bytes   :    54488 B (  53.2 KB)
  Minimum Free Bytes:  8330644 B (8135.4 KB)
  Largest Free Block:  8257524 B (8064.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
    43 : UART_TX[0]
    44 : UART_RX[0]
============ After Setup End =============
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-01-08:
Hi @lmabbe-lm, is the WAV or MP3 played? Can you try to comment the `speaker1.end();` in your code and post the debug log again? Thank you

lmabbe-lm, 2025-01-09:
> 你好@lmabbe-lm，播放的是 WAV 还是 MP3？你能尝试在代码中注释掉`speaker1.end();`并再次发布调试日志吗？谢谢

After commenting, it can work normally now. Thank you for your answer. May I ask about the timing of using end()? The role of end().
By the way, what happens when the sound is too low and how can I adjust it. thank you

P-R-O-C-H-Y, 2025-01-09:
You should call the `end()` if you no longer want to play anything and free resources + the i2s peripheral.
I have opened a fix for the error that you were getting. 

About the volume, we have no software control implemented in the I2S library.
You may need to control the volume on HW layer by using some amplifier.

lmabbe-lm, 2025-01-10:
> `end()`如果您不想再播放任何内容，请拨打电话并释放资源 + i2s 外围设备。 我已经为您遇到的错误打开了修复程序。
> 
> 关于音量，我们没有在 I2S 库中实现软件控制。 您可能需要使用一些放大器在硬件层上控制音量。

ok，thinks

PR #10833 (2025-01-10): fix(i2s): Check if pin is used before clearing bus
## Description of Change
This PR fixes error `[esp32-hal-periman.c:122] perimanSetPinBus(): Invalid pin: 255`, when some of the pins are not used (-1) but the bus clear was called anyway, so it was marked as error Invalid pin.

## Tests scenarios

## Related links

Closes #10828
closed 2025-01-10 by me-no-dev