espressif/arduino-esp32 #10844: Could not find the package with 'espressif/toolchain-riscv32-esp' requirements for your system 'windows_arm64'

TobiKr, 2025-01-11:
### Board

Lilygo Touch S3

### Device Description

Default Board

### Hardware Configuration

Standard Config

### Version

latest master (checkout manually)

### IDE Name

VSCode

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Can't intialize Project on Windows 11 ARM64 (Snapdragon):
````Resolving lilygo-t-display-s3 dependencies...
Configuring toolchain packages from a remote source...
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 13.2.0+20240530
UnknownPackageError: Could not find the package with 'espressif/toolchain-riscv32-esp @ 13.2.0+20240530' requirements for your system 'windows_arm64'

### Sketch

```cpp
platformio.ini:

[env:lilygo-t-display-s3]
platform = https://github.com/sgryphon/platform-espressif32.git#sgryphon/add-esp32-arduino-libs
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master
    platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
board = lilygo-t-display-s3
framework = arduino
build_unflags = -std=gnu++11
build_flags = -std=gnu++2b ;https://community.platformio.org/t/which-c-standard-am-i-using/24597/4
              -D LV_CONF_INCLUDE_SIMPLE
              -D LV_CONF_SKIP
              -D LV_COLOR_DEPTH=16
              -D LV_USE_ST7789
              -D LV_USE_LOG
              -D LV_LOG_LEVEL=LV_LOG_LEVEL_INFO
              -D LV_USE_ASSERT_NULL
lib_deps = lvgl = https://github.com/lvgl/lvgl.git@^9.0.0
monitor_filters = esp32_exception_decoder, colorize, time ; debug
build_type = debug
```

### Debug Message

```plain
Resolving lilygo-t-display-s3 dependencies...
Configuring toolchain packages from a remote source...
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 13.2.0+20240530
UnknownPackageError: Could not find the package with 'espressif/toolchain-riscv32-esp @ 13.2.0+20240530' requirements for your system 'windows_arm64'
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2025-01-12:
This is not working anymore
```
platform = https://github.com/sgryphon/platform-espressif32.git#sgryphon/add-esp32-arduino-libs
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master
    platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
```
Use pioarduino, a community compatible fork of Platformio supporting latest espressif Arduino core
```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
```

Jason2866, 2025-01-12:
Btw. using Windows ARM as build platform is the worst possible choice.
It will be slow as hell since there are no native Windows ARM tools & toolchains.
The standard Windows AMD64 toolchains will be used together with the emulator.

lbernstone, 2025-01-12:
As an alternative, I believe you can install WSL and compile with native linux toolchains.

TobiKr, 2025-01-12:
Thank you all, found a solution with pioarduino and created a basic example [T-Display-S3-ModernBasicTemplate](https://github.com/TobiKr/T-Display-S3-ModernBasicTemplate)

closed 2025-01-12 by TobiKr