espressif/arduino-esp32 #10846: Serial stop working after upgrade to version 3.1.1

rezder, 2025-01-11:
### Board

Waveshare-esp32-S3-zero

### Device Description

No attachment

### Hardware Configuration

no attachement

### Version

latest master (checkout manually)

### IDE Name

Arduino ide

### Operating System

linux arch

### Flash frequency

80 mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Serial println and write stop working after upgrade to version 3.1.1 worked on 3.1.0
upload works with the arduino ide

### Sketch

```cpp
const long SERIAL_BPS = 115200;

void setup() {
  Serial.begin(SERIAL_BPS);
  delay(3000);
  Serial.println(USB_PRODUCT);

}

void loop() {
  Serial.println("Red");
  Serial.println("Green");
  Serial.println("Blue");

}
```

### Debug Message

```plain
NO debug messages.
```

### Other Steps to Reproduce

The variant esp32-s3-zero have  USB_SERIAL defined as empty #define USB_SERIAL       ""  

The waveshare esp32-c3-zero still works. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-01-11:
Please go into Arduino preferences and set verbose output during compilation.  Post the very first line of your log (with -dump-prefs) here.

rezder, 2025-01-12:
First line of compilation
FQBN: esp32:esp32:waveshare_esp32_s3_zero
Using board 'waveshare_esp32_s3_zero' from platform in folder: /home/rho/.arduino15/packages/esp32/hardware/esp32/3.1.1
Using core 'esp32' from platform in folder: /home/rho/.arduino15/packages/esp32/hardware/esp32/3.1.1

rezder, 2025-01-12:
Yes when using dev the serial works:

FQBN: esp32:esp32:esp32s3:CDCOnBoot=cdc
Using board 'esp32s3' from platform in folder: /home/rho/.arduino15/packages/esp32/hardware/esp32/3.1.1
Using core 'esp32' from platform in folder: /home/rho/.arduino15/packages/esp32/hardware/esp32/3.1.1

rezder, 2025-01-12:
If you are going to make any change to the variant file the rgb order is also wrong it is red green blue.
RGB_BUILTIN_LED_COLOR_ORDER

rezder, 2025-01-12:
I have several zero boards non of the works so believe it is a software problem.

lbernstone, 2025-01-12:
RGB color order is fixed in https://github.com/espressif/arduino-esp32/issues/10822
You do not have CDC on boot set when using the waveshare_esp32_s3_zero, or is it not registering properly?
The variant file needs to be fixed by someone that actually owns the board, so I will not be making any changes to it.

rezder, 2025-01-12:
Thanks when turning on cdc it works again. I have some problems with the arduino ide when more than one is open board settings flips around.

SuGlider, 2025-01-13:
My suggestion would be to:
- Select `ESP32S3 Dev Module` board, 
- Enable CDC on Boot 
- Add `build_opt.h` file to your project with just a line `-DRGB_BUILTIN_LED_COLOR_ORDER=LED_COLOR_ORDER_RGB` 

 `build_opt.h` file will conatain options that will be included in the gcc command line.
 In this case, it will  define the LED RGB order for your board.