espressif/arduino-esp32 #10847: i2c error

xkjack, 2025-01-12:
### Board

ESP32 Dev Module

### Device Description

ESP-WROOM-32

### Hardware Configuration

Pin 21 & 22 connect to LM75 by i2c
Pin 34 connect to LDR

### Version

v3.1.0

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Cannot connect to LM75 when use in code
Output:
```log
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
Scanning...
Unknow error at address 0x01
Unknow error at address 0x02
Unknow error at address 0x03
Unknow error at address 0x04
Unknow error at address 0x05
Unknow error at address 0x06
...
Unknow error at address 0x77
Unknow error at address 0x78
Unknow error at address 0x79
Unknow error at address 0x7A
Unknow error at address 0x7B
Unknow error at address 0x7C
Unknow error at address 0x7D
Unknow error at address 0x7E
No I2C devices found

```
LM75 is `0x4C` when I check with
```cpp
/*********
  Rui Santos
  Complete project details at https://randomnerdtutorials.com  
*********/

#include <Wire.h>
 
void setup() {
  Wire.begin();
  Serial.begin(115200);
  Serial.println("\nI2C Scanner");
}
 
void loop() {
  byte error, address;
  int nDevices;
  Serial.println("Scanning...");
  nDevices = 0;
  for(address = 1; address < 127; address++ ) {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address<16) {
        Serial.print("0");
      }
      Serial.println(address,HEX);
      nDevices++;
    }
    else if (error==4) {
      Serial.print("Unknow error at address 0x");
      if (address<16) {
        Serial.print("0");
      }
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0) {
    Serial.println("No I2C devices found\n");
  }
  else {
    Serial.println("done\n");
  }
  delay(5000);          
}
```
but it not found in my code. Why? How to fix it?
Thank you.

### Sketch

```cpp
#include <math.h>
#include "BluetoothSerial.h"
#include <Wire.h>

const int LDR_PIN = 34;
const float Vcc = 3.3;
const float R_fixed = 10000;
uint8_t temp_msb, temp_lsb;
const int temp_address = 0x4C;
uint16_t temp_reg;
float temperatureC;

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  pinMode(2, OUTPUT);
  SerialBT.begin("CDHTVL11L114N7B7");
  searchI2C();
}

void loop() {
  Wire.beginTransmission(temp_address);
  Wire.write(0x00);
  Wire.endTransmission();
  Wire.requestFrom(temp_address, 2);
  temp_msb = Wire.read();
  temp_lsb = Wire.read();
  temp_reg = (temp_msb << 8) | temp_lsb;
  temperatureC = temp_reg/256.0f;
  int adcValue = analogRead(LDR_PIN);
  float lux = calculateLux(adcValue);
  // Serial.printf("Lux: %.2f\n", lux);
  // Serial.printf("Temp: %.2f\n", temperatureC);
  if (lux < 100) {
    digitalWrite(2, HIGH);
  } else {
    digitalWrite(2, LOW);
  }
  SerialBT.println(lux);
  SerialBT.println(temperatureC,3);
  delay(100);
}

float calculateLux(int analogValue) {
    const float R_fixed = 10000.0;
    const float Vcc = 3.3;
    const float R0 = 45000.0; // 32500.0;
    const float luxAtR0 = 10.0;
    float Vout = (analogValue * Vcc) / 4095.0;
    float RLDR = (R_fixed * (Vcc - Vout)) / Vout;
    float lux = luxAtR0 * pow((R0 / RLDR), 1.0);
    return lux * 10;
}

void searchI2C() {
  byte error, address;
  int nDevices;
  Serial.println("Scanning...");
  nDevices = 0;
  for(address = 1; address < 127; address++ ) {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address<16) {
        Serial.print("0");
      }
      Serial.println(address,HEX);
      nDevices++;
    }
    else if (error==4) {
      Serial.print("Unknow error at address 0x");
      if (address<16) {
        Serial.print("0");
      }
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0) {
    Serial.println("No I2C devices found\n");
  }
  else {
    Serial.println("done\n");
  }
}
```

### Debug Message

```plain
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
Scanning...
Unknow error at address 0x01
Unknow error at address 0x02
Unknow error at address 0x03
Unknow error at address 0x04
Unknow error at address 0x05
Unknow error at address 0x06
...
Unknow error at address 0x77
Unknow error at address 0x78
Unknow error at address 0x79
Unknow error at address 0x7A
Unknow error at address 0x7B
Unknow error at address 0x7C
Unknow error at address 0x7D
Unknow error at address 0x7E
No I2C devices found
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-01-12:
Missing `Wire.begin(SDA, SCL);`?

dellancelod, 2025-02-05:
After trial and error with the same issue, I discovered that they updated wire.cpp so that now i2c needs Level Converter from 3.3v to 5v

[image]