espressif/arduino-esp32 #10850: Use of Ethernet and WiFi AP & STA simulateneously

enriquedbelec, 2025-01-13:
### Board

Custom board

### Device Description

Custom board with ESP32 Wroom 16 MB of flash with LAN8710A

### Hardware Configuration

ETH_CLK_MODE    -> ETH_CLOCK_GPIO17_OUT
ETH_POWER_PIN  -> 5
ETH_TYPE  -> ETH_PHY_LAN8720
ETH_ADDR ->  0
ETH_MDC_PIN  -> 23
ETH_MDIO_PIN  -> 18
ETH_CLOCK_PIN  -> 17

### Version

v2.0.17

### IDE Name

VSCode

### Operating System

Windows 10

### Flash frequency

40 Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Hi,
I've checked on Arduino 2.0.17 and 3.1.1 that if you begin Ethernet only it can perform an HTTP connection ok. But if you enable the AP, then the ETH connection drops (can't establish connection).

Is there any way to keep working the ethernet connections while the AP is up? Ídem question with ETH + AP/STA ?

Thanks.

### Sketch

```cpp
// Init Ethernet driver
ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
```

### Debug Message

```plain
No
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-01-13:
In 2.0.17 you can not. For 3.x you can decide which is the default outgoing interface by doing 
```cpp
Network.setDefaultInterface(ETH); //outgoing connections go through ETH by default
Network.setDefaultInterface(WiFi.STA); //outgoing connections go through STA by default
```

enriquedbelec, 2025-01-14:
Thank you @me-no-dev. If ETH is connected and sending data to HTTP(s) and WiFi is put in AP mode it also stops connecting to internet on ETH. Will it work in this situation the `Network.setDefaultInterface(ETH);` ?

me-no-dev, 2025-01-14:
AP should not interfere, unless they are both set in the same subnet.

enriquedbelec, 2025-01-14:
The AP give to my PC the IP 192.168.4.1, and ethernet obtains the IP 192.168.0.58 from the LAN, so they are not in the same subnet. But when the AP is started and/or when I connect with my PC to the AP and an IP is assigned, the HTTPS connections with WiFiClientSecure stop working. Any ideas?

me-no-dev, 2025-01-14:
@enriquedbelec please provide a test sketch to reproduce the issue

enriquedbelec, 2025-01-15:
Hi,

Here is a test sketch to reproduce the issue. The sketch starts ETH and waits for link up and get IP. After that it starts performing GET requests to URL. 30 seconds after start sketch, ir starts AP.

Here you can note, that the GET request is performed with 200 OK status code, until the AP is started. After the AP is started and also if you connect to AP, the requests can't be performed (log verbose shows a timeout).

```
#include <ETH.h>
#include <WiFi.h>
#include <HTTPClient.h>

#define ETH_CLK_MODE    ETH_CLOCK_GPIO17_OUT

// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_POWER_PIN   5

// Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_TYPE        ETH_PHY_LAN8720

// I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR        0

// Pin# of the I²C clock signal for the Ethernet PHY
#define ETH_MDC_PIN     23

// Pin# of the I²C IO signal for the Ethernet PHY
#define ETH_MDIO_PIN    18

// ETH clock pin
#define ETH_CLOCK_PIN   17

// NRST pin
#define ETH_NRST_PIN    12

// AP config
const char *apSSID = "ESP32_AP";
const char *apPassword = "12345678";

// Time until AP is started
const unsigned long apStartDelay = 30000; // 30 segundos

// Start ap flag
bool startAP = false;

// Already started AP flag
bool apAlreadyStarted = false;

// Task to get data from URL
void httpTask(void *parameter) {
  while (true) {
    if (ETH.linkUp()) {
      HTTPClient http;
      http.begin("http://jsonplaceholder.typicode.com/posts"); // URL de prueba
      int httpResponseCode = http.GET();
      if (httpResponseCode > 0) {
        Serial.printf("HTTP Response code: %d\n", httpResponseCode);
      } else {
        Serial.printf("HTTP POST failed, error: %s\n", http.errorToString(httpResponseCode).c_str());
      }
      http.end();
    }
    vTaskDelay(10000 / portTICK_PERIOD_MS); // Espera 10 segundos
  }
}

void setup() {
    Serial.begin(115200);
    delay(1000);
    // Init ETH
    // Power off/on PHY
    pinMode(ETH_POWER_PIN, INPUT);
    digitalWrite(ETH_POWER_PIN, 0);
    pinMode(ETH_POWER_PIN, OUTPUT);
    delay(99);
    // Reset LAN8710A
    pinMode(ETH_NRST_PIN, OUTPUT);
    digitalWrite(ETH_NRST_PIN, 0);
    delay(200);
    digitalWrite(ETH_NRST_PIN, 1);
    delay(500);
    // Init Ethernet driver
    ETH.begin(ETH_TYPE, ETH_ADDR, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_POWER_PIN, ETH_CLK_MODE);

    // Set hostname
    ETH.setHostname("ESP32_TEST");

    // Wait for linkUp Ethernet connection
    Serial.println("Waiting for ETH linkUp...");
    while (!ETH.linkUp()) {
        delay(500);
        Serial.print(".");
    }
    Serial.println("\nETH link UP!");

    // Wait for having an IP on Ethernet connection
    Serial.println("Waiting for IP on ETH...");
    while (!ETH.hasIP()) {
        delay(500);
        Serial.print(".");
    }
    Serial.println("\nETH now has IP!");

    // Create task to send http posts
    xTaskCreate(httpTask, "HTTP Task", 4096, NULL, 1, NULL);
}

void loop() {
    if(!apAlreadyStarted){
        if(millis() > apStartDelay){
            startAP = true;
        }
        if (startAP) {
            Serial.println("Starting AP...");
            WiFi.mode(WIFI_AP);
            WiFi.softAP(apSSID, apPassword, 1, 0, 1);

            IPAddress IP = WiFi.softAPIP();
            Serial.printf("AP started. IP: %s\n", IP.toString().c_str());

            startAP = false;
            apAlreadyStarted = true;
        }
    }
}
```

enriquedbelec, 2025-02-27:
Hi @me-no-dev do you have any news about this? 

I've done tests on my ESP32 and using ethernet (LAN8710) and WiFi AP simulateneously makes that if I ping from a LAN PC to ethernet IP does not always answer the ping. It loses 50-70% of packets.
If I start the board with only ethernet connected and I ping it from LAN, it always anwer, no packet loss.

Also, I've tried to get the default interface with Network.getDefaultInterface() method, and it always is at ETH_DEF. Also tried to do Network.setDefaultInterface(ETH) after AP is started and after AP gives IP to a client to force the iface priority to ethernet, but the issue is the same. Also tried to Network.setDefaultInterface(ETH) in a loop to ensure the priority is always overwritten.

I suspect that this is not a priority issue, because the is the AP what is interfering with Ethernet addressing.

I'm using Arduino Core 3.1.3 for the tests.

If you need I can do more tests. Thank you.

LynxEmbedded, 2025-03-05:
Hi there,
I am facing a similar issue however I solved this by giving a static configuration if STA is failed to connect as follows:

// Ethernet Init Codes
ETH.begin(....);

// PPP Init ....Codes
PPP.begin(...)

// WiFi start codes....
WiFi.begin(....)

// After waiting a while for connection to SSID & Then
if ( WiFi.status() != WL_CONNECTED )
{
  // Set a static configuration if STA failed to connect ssid
  IPAddress local_IP(192, 168, 1, 150);
  IPAddress gateway(192, 168, 1, 1);
  IPAddress subnet(255, 255, 255, 0);
  IPAddress primaryDNS(8, 8, 8, 8);    //optional
  IPAddress secondaryDNS(8, 8, 4, 4);  //optional
  
  if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS))
  {
    utilUartPrintf("STA Failed to configure");
  }
}

// After This Ethernet & and others (PPP) continues to connect internet....
...

enriquedbelec, 2025-03-12:
I've discovered the cause of this issue. And it's because the CLOCK_SOURCE of the LAN8720 is the GPIO_17. And Espressif has reported it in forums and on known issues:
https://www.esp32.com/viewtopic.php?t=30141&start=10
http://esp32.io/viewtopic.php?t=37757

Errata report from Espressif: https://docs.espressif.com/projects/esp-chip-errata/en/latest/esp32/03-errata-description/index.html#clock-esp32-cannot-be-used-as-the-phy-clock-source-if-wi-fi-and-ethernet-are-used-at-the-same-time

Hope it helps anyone with the same issue. Thank you so much for your help @LynxEmbedded and @me-no-dev