espressif/arduino-esp32 #10851: DMA driven DAC

tomcombriat, 2025-01-13:
### Board

ESP Dev Module WROOM-32D

### Device Description

Plain development board

### Hardware Configuration

GPIO25 connected to scope.

### Version

v3.1.0

### IDE Name

Arduino IDE

### Operating System

Debian

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I'm trying to port Mozzi to the latest version of the ESP32 library. I2S and timer establishment works well following https://docs.espressif.com/projects/esp-idf/en/v5.4/esp32/index.html but DMA driven DAC as per https://docs.espressif.com/projects/esp-idf/en/v5.4/esp32/api-reference/peripherals/dac.html does not.

### Sketch

```cpp
#include <driver/dac_continuous.h>

dac_continuous_handle_t dac_handle;
void setup() {
  dac_continuous_config_t dac_config = {
    .chan_mask = DAC_CHANNEL_MASK_ALL,
    .desc_num = 2,
    .buf_size = 4,
    .freq_hz = 30000,
    .offset = 120,
    .clk_src = DAC_DIGI_CLK_SRC_DEFAULT,
    .chan_mode = DAC_CHANNEL_MODE_SIMUL,
  };
  dac_continuous_new_channels(&dac_config, &dac_handle);
  dac_continuous_enable(dac_handle);
}

void loop() {
  size_t bytes_written;
  uint8_t tt = 120;
  dac_continuous_write(dac_handle, &tt, sizeof(uint8_t), &bytes_written, 0);
}
```

### Debug Message

```plain
22:11:09.084 -> E (4590) dac_continuous: dac_continuous_write(618): Load data failed
22:11:09.084 -> E (4596) dac_continuous: s_dac_wait_to_load_dma_data(551): Get available descriptor timeout
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-01-13:
@P-R-O-C-H-Y, could you please take a look? Thanks!

lucasssvaz, 2025-01-13:
Just a heads up that you are using documentation for IDF v5.4 while Arduino v3.1.0 uses IDF v5.3. Idk if there are significant changes but it might help.

tomcombriat, 2025-01-13:
@lucasssvaz Thanks for this! If I am correct, there is no significant differences for the DAC API. 

Also, allow myself to add that I also tried to get this DAC to work using I2S (the old API allowed access of the native DAC using I2S), but also did not manage, which is why I turned to the dac_conti API.

me-no-dev, 2025-01-13:
Have you tried with more than one byte? Documentation states `It is usually used to transport a long signal like an audio.`

lbernstone, 2025-01-14:
The IDF example uses a much larger buffer (2048). Try capturing the results from new_channels and enable.  They probably will indicate an invalid parameter.

tomcombriat, 2025-01-14:
@lbernstone Increasing the buffer does not solve it. Also both `new_channels` and `enable` return `ESP_OK`.

@me-no-dev I tried sending a 1024 buffer but got exactly the same errors and symptoms.

Thanks for your help!

yaqwsx, 2025-02-01:
I can confirm that on IDF 5.4, I observed the same problem with the continuous DAC driver. Very often the `dac_continuous_write` gets stuck forever. Changing the size of the buffer has no impact whatsoever.