espressif/arduino-esp32 #10857: upload firmware via USB after USB.begin()

savejeff, 2025-01-14:
### Board

ESP32-S3

### Device Description

Here is what I'm trying to do, but am not able to accomplish: 
I want to upload a Firmware File after calling USB.begin() without a complete power cycle restart

I need this, as I have a battery-powered device where I can't access the battery connection to disconnect and reconnect it (due to waterproofness requirement)

The Problem I'm facing is: as soon as I call USB.begin() to allow SerialUSB communication there is nothing I can do to bring the Processor into a state that allows uploading firmware files. 
What I mean is, when I plug in the ESB32 on the USB Data Port it opened a COM port I can upload sketches to. As soon as I call USB.begin this COM port goes away and another one comes up what I cant use to upload sketches anymore. I tired doing a Software restart with ESP.restart(); but this does not reset the USB. i have to completely unplug the ESP32 removing power and replug it to get this upload port back.

What can I do to bring the USB Data Serial connection back into the state that lets me upload Sketches? 
Im open to any suggestions like making it so that with also I can upload to the second comport that opened after USB.begin or a way to complelty reset the ESP through a powercycle. 
I'm

### Hardware Configuration

ESP32-S3 DevKit C1

The easiest setup is to have both USB port (UART and USB) into a PC. This way you can unplug the USB Data port without a powercycle. 
It is accomplished what I want when the ESP is brought back into a state where I can upload a sketch via the USB Data port without unplugging the UART USB Port 

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

.

### Sketch

```cpp
// based on the USBSerial.ino example (see https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/USBSerial/USBSerial.ino)

void setup() {

	Serial.begin(115200);
	while(!Serial.available()) {
		Serial.print(".");
		delay(1000);
	}

	Serial.setDebugOutput(true);

	USB.onEvent(usbEventCallback);
	USBSerial.onEvent(usbEventCallback);

	USBSerial.begin();
	USB.begin();
}

void loop() {

	static uint32_t last_time_print = millis();
	if(millis() - last_time_print > 1000) {
		last_time_print = millis();
		Serial.println("loop");
	}

	while (Serial.available()) {
		char c = Serial.read();
		if(c == '#')
			ESP.restart();

		USBSerial.write(c);
	}

}
```

### Debug Message

```plain
.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-09-28:
@Parsaabasi - This issue shall be fixed by ESP32 Arduino Core 3.3.1

Tested with ESP32-S3 Devkit + Arduino IDE 2.3.6 and the sketch below:
``` cpp
// ESP32-S3 | ESP32-S2 
// Arduino IDE Setting: CDC on Boot = ENABLED (both S2|S3) 
// Arduino IDE Setting for S3: USB Mode = USB OTG (TinyUSB) ||  Upload Mode =  USB OTG CDC (TinyUSB)
// First time sketch upload may require to make the ESP32-S2/S3 to enter in Download Mode
// In order to achieve it: Hold BOOT Button pressed, press and release RESET button then release BOOT button.
#include "USB.h"
#include "USBHIDGamepad.h"
USBHIDGamepad Gamepad;

#define RGB_LED_PIN 48 // some ESP32-S3 may use pin 38 or something else. S2 Devkit uses pin 8

void setup() {
  Serial.begin();  // TinyUSB CDC Port
  while(!Serial) {
    // blinks RGB LED to tell user that the Serial Terminal has not been open
    rgbLedWrite(RGB_LED_PIN, 64, 0, 0);  // RED
    delay(250);
    rgbLedWrite(RGB_LED_PIN, 0, 0, 0);  // OFF
    delay(250);
  }
  USB.begin();
  Gamepad.begin();
  Serial.println("TinyUSB Sketch will send CDC and HID reports in loop()");
}

uint32_t c = 0;
void loop() {
  Serial.println(String(c++) + " ==> Hello ESP32S3");
  delay(500);
  if (c & 1) {
    Gamepad.pressButton(0);
  } else {
    Gamepad.releaseButton(0);
  }
  delay(500);
}
```

SuGlider, 2025-10-07:
@savejeff - Could you please confirm that the `esptool.py` version that is installed with the Arduino Core 3.3.1 solves this issue? Thanks!

Parsaabasi, 2026-09-15:
Dear @savejeff 

Since you were using an older version of Arduino, I close this case for now. Please refer to the comments above. In case you needed further assistance, please free free to reopen or create a new issue. 

Kind regards,

closed 2026-09-15 by Parsaabasi