espressif/arduino-esp32 #10871: ESP32-C6 Zigbee and wifi AP and STA error

sonirohit3, 2025-01-16:
### Board

ESP32-C6

### Device Description

Esp32-c6 custom board

### Hardware Configuration

ESP32-C6 custom board

### Version

latest master (checkout manually)

### IDE Name

Arduino

### Operating System

Macos

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

Wifi doesn't work in STA or AP more when Zigbee is initialized. Also I don't see a way to de-init zigbee.
Idea is to use Wifi OTA activated with a switch.  
-> When using AP mode, the client can find the AP but cannot connect. 
-> When using STA mode, the WiFi won't connect and return error:

### Sketch

```cpp
STA code:

  WiFi.mode(WIFI_MODE_AP);
  if (!WiFi.softAP(ssid, password)) 
  {
    log_e("Soft AP creation failed.");
  }
  IPAddress IP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(IP);

STA code:

WiFi.mode(WIFI_OFF);
  delay(100);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  delay(5000);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    WiFi.begin(ssid, password);
    log_i("WiFi failed, retrying.");
  }
```

### Debug Message

```plain
Debug logs for STA mode

[ 20549][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 110 - STA_START
[ 22960][V][STA.cpp:216] _onStaEvent(): STA Disconnected: SSID: Soni_wifi2G_IOT, BSSID: 00:00:00:00:00:00, Reason: 201
[ 22961][V][NetworkEvents.cpp:113] _checkForEvent(): Network Event: 113 - STA_DISCONNECTED
[ 22962][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 113 - STA_DISCONNECTED
[ 22962][W][STA.cpp:137] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[ 22963][D][STA.cpp:155] _onStaArduinoEvent(): WiFi Reconnect Running
[ 22963][W][STA.cpp:541] disconnect(): STA already disconnected.
[ 25370][V][STA.cpp:216] _onStaEvent(): STA Disconnected: SSID: Soni_wifi2G_IOT, BSSID: 00:00:00:00:00:00, Reason: 201
[ 25371][V][NetworkEvents.cpp:113] _checkForEvent(): Network Event: 113 - STA_DISCONNECTED
[ 25371][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 113 - STA_DISCONNECTED
[ 25372][W][STA.cpp:137] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[ 25372][D][STA.cpp:158] _onStaArduinoEvent(): WiFi AutoReconnect Running
[ 25373][W][STA.cpp:541] disconnect(): STA already disconnected.
E (29754) wifi:sta is connecting, return error
[ 25566][E][STA.cpp:417] connect(): STA connect failed! 0x3007: ESP_ERR_WIFI_CONN
[ 25567][I][Handle_OTA.ino:37] init_wifiAP(): WiFi failed, retrying.
E (29766) wifi:sta is connecting, return error
```

### Other Steps to Reproduce

Init Zigbee and then init the wifi.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-01-16:
please provide full sketch code to reproduce the issue

lbernstone, 2025-01-16:
Try adding this to your code:
```
#include <esp_coexist.h>
void setup() {
    esp_coex_wifi_i154_enable(void)
```

sonirohit3, 2025-01-17:
Hi @lbernstone , Your suggestion worked for STA mode. Still the problem persists in AP more. The AP is created but the client such as mobile is not able to connect. Any insight on that?

P-R-O-C-H-Y, 2025-01-17:
Just a quick note: Since there is only one RF path in ESP32-C6, which means Wi-Fi and Zigbee can't receive simultaneously, it has a significant impact on performance. Hence the two SoCs solution is recommended.

sonirohit3, 2025-01-18:
Hi @lbernstone & @P-R-O-C-H-Y , The SSID is visible in mobile but the connection is always getting failed. Unfortunately there are no logs when client tries to connect. I do understand that RF path is single. If I can disable Zigbee ( like Zigbee.disable()) and init a WiFi AP mode, this will solve my purpose.
Here is the code snippet:

setup:

void setup()
{
    Serial.begin(115200);       
    log_i("Booting up...");
    esp_coex_wifi_i154_enable();
    init_pheripherals();
    init_Zigbee();
    init_flash();
    digitalWrite(PCA_OE, LOW);

}

Init_ota function which is called with longpress of switch:

void OTA_init()
{
  is_ota_en=1;
  WiFi.mode(WIFI_MODE_AP);
  if (!WiFi.softAP(ssid, password)) 
  {
    log_e("Soft AP creation failed.");
  }
  IPAddress IP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(IP);

  OTA_server_init(); // simple AP based webserver
}

Below are logs for same:

[ 48363][V][NetworkEvents.cpp:113] _checkForEvent(): Network Event: 101 - WIFI_READY
[ 48372][V][AP.cpp:108] _onApEvent(): AP Started
[ 48373][V][NetworkEvents.cpp:113] _checkForEvent(): Network Event: 130 - AP_START
[ 48373][V][AP.cpp:88] _onApArduinoEvent(): Arduino AP Event: 130 - AP_START
[ 48376][V][AP.cpp:112] _onApEvent(): AP Stopped
[ 48377][V][NetworkEvents.cpp:113] _checkForEvent(): Network Event: 131 - AP_STOP
[ 48377][V][AP.cpp:88] _onApArduinoEvent(): Arduino AP Event: 131 - AP_STOP
[ 48669][V][AP.cpp:108] _onApEvent(): AP Started
[ 48670][V][NetworkEvents.cpp:113] _checkForEvent(): Network Event: 130 - AP_START
[ 49184][I][Handle_Zigbee.ino:59] Update_devstate_server(): Updating Endpoint : 11 with value : 0
[ 49185][I][Handle_Zigbee.ino:59] Update_devstate_server(): Updating Endpoint : 12 with value : 0
[ 49186][I][Handle_Zigbee.ino:59] Update_devstate_server(): Updating Endpoint : 13 with value : 1
[ 49187][I][Handle_Zigbee.ino:59] Update_devstate_server(): Updating Endpoint : 14 with value : 0
[ 49188][I][Handle_Zigbee.ino:59] Update_devstate_server(): Updating Endpoint : 15 with value : 1
[ 49189][I][Handle_Zigbee.ino:69] Update_devstate_server(): Updating Temperature Endpoint : 16 with value : 0
[ 49190][V][ZigbeeTempSensor.cpp:58] setTemperature(): Updating temperature sensor value...
[ 49190][D][ZigbeeTempSensor.cpp:60] setTemperature(): Setting temperature to 0
[ 57066][V][ZigbeeCore.cpp:352] esp_zb_app_signal_handler(): ZDO signal: NLME Status Indication (0x32), status: ESP_OK
[ 59217][I][Handle_Zigbee.ino:59] Update_devstate_server(): Updating Endpoint : 10 with value : 0
[ 59218][I][Handle_Zigbee.ino:59] Update_devstate_server(): Updating Endpoint : 11 with value : 0
[ 59219][I][Handle_Zigbee.ino:59] Update_devstate_server(): Updating Endpoint : 12 with value : 0
[ 59220][I][Handle_Zigbee.ino:59] Update_devstate_server(): Updating Endpoint : 13 with value : 1
[ 59221][I][Handle_Zigbee.ino:59] Update_devstate_server(): Updating Endpoint : 14 with value : 0
[ 59222][I][Handle_Zigbee.ino:59] Update_devstate_server(): Updating Endpoint : 15 with value : 1
[ 59223][I][Handle_Zigbee.ino:69] Update_devstate_server(): Updating Temperature Endpoint : 16 with value : 0
[ 59224][V][ZigbeeTempSensor.cpp:58] setTemperature(): Updating temperature sensor value...
[ 59224][D][ZigbeeTempSensor.cpp:60] setTemperature(): Setting temperature to 0

TD-er, 2025-01-18:
I don't know enough about Zigbee, but I wonder what channels it needs.
Is there some overlap in frequencies?
If so, maybe you could try to force the AP to use a channel which is overlapping with the Zigbee frequency?

For STA mode, you do have 'idle' time inbetween beacons.
However when using AP mode, you have not.
That's also why you absolutely need to use the same channel for STA and AP in AP+STA mode, or else your connection is as stable as a drunk on rollerskates.

So I guess the problem here is the constant frequency hopping.

I found this online:

[image]

So maybe you could use WiFi channel 1 along with Zigbee channel 12 or 13?
And perhaps you can limit your WiFi AP mode to 802.11g as this is smaller than the 20 MHz width as depicted by the blue area in the image.

If Zigbee does use frequency hopping, like Bluetooth does, then it sounds to me like a lost cause as it is technically hardly usable.

sonirohit3, 2025-01-20:
@TD-er  , So in this case only way to tackle this issue is to disable Zigbee when AP for firmware update has to be init. Better to have Zigbee.disable function.

@P-R-O-C-H-Y , Can you suggest if we have this functionality?

P-R-O-C-H-Y, 2025-01-20:
@sonirohit3 I was searching through all available APIs of esp-zigbee-sdk and I didn't see any option to disable/deinit Zigbee.

sonirohit3, 2025-01-24:
@P-R-O-C-H-Y , I agree. I also looked on this and found this missing. Any workaround you can suggest?

TD-er, 2025-01-27:
@P-R-O-C-H-Y 
Does the C5 allow the 2.4 and 5 GHz WiFi to be active at the same time?
I guess if it does, then it will draw significant amounts of energy, especially when both are in AP mode.
Just based on the [video Andreas Spiess made about the C5](https://www.youtube.com/watch?v=glEvHRHdboA&ab_channel=AndreasSpiess), it looks like both bands share the same antenna feed, so I doubt both can be active at the same time.
[image]

If you can do 5 GHz WiFi + some 2.4 GHz activity (Zigbee/BT/WiFi), then I guess it will only be in STA mode for WiFi and not in AP mode. (so also not STA+AP)
In AP mode the antenna is continuously occupied, so it cannot be used for something else.

A simple test for this hypothesis is to use Bluetooth and WiFi AP mode at the same time.
I suspect that if it is even supported, it will be extremely unstable.

P-R-O-C-H-Y, 2025-01-27:
@TD-er You can check some coexistence info here in the IDF docs for C6 https://docs.espressif.com/projects/esp-idf/en/stable/esp32c6/api-guides/coexist.html

I asked for Zigbee/WiFi to be added, as I saw many users asking about this in other repositories also.

About the C5, I just got it for few days now, did not test many things, was just curious about the AP on 5GHz + Zigbee. 
Also the C5 support is still in a preview in IDF master, so there might be updates to this also.

sonirohit3, 2025-02-04:
@P-R-O-C-H-Y , any update from zigbee team?

P-R-O-C-H-Y, 2025-02-04:
@sonirohit3 I will implement the Zigbee OTA. Will start on it tomorrow.

P-R-O-C-H-Y, 2025-02-05:
Meanwhile I got a reply on the disable/deinit feature: 

> The zigbee sdk still has not provide the feature. If you intend to disable the zigbee, I think you only can pend the zigbee task now.

So there might be the way with the Zigbee task, we can try that also. But for sure I will add the OTA cluster.

P-R-O-C-H-Y, 2025-02-06:
@sonirohit3 Can you please give the PR a try?? I can see the OTA, but I am not sure and did not figure out yet how to run the update. If you know please let me know so I can add some tutorial for that. Thanks

[image]

sonirohit3, 2025-02-08:
@P-R-O-C-H-Y , I got some reference here. https://community.home-assistant.io/t/zha-ota-firmware-update-upgrade/488960 .
I will give it a try soon.

P-R-O-C-H-Y, 2025-02-10:
@sonirohit3 I will give a new test tomorrow, I may get it to work finally and will post a tutorial also :)

P-R-O-C-H-Y, 2025-02-12:
@sonirohit3 I was able to make it work, check the Pull Request description. There are links for tutorial for HA and how to create the OTA file. At the end I got an error in HA, that update failed, but the update succeed in the device and the version was incremented properly, the "new" feature I added to the code was working fine. Also HA was showing the new version, so update was no longer available showing "up-to-date" again :)

sonirohit3, 2025-02-16:
@P-R-O-C-H-Y , Great! this looks like sorted. I was not able to give it a complete try since I am traveling. Will give it a try once back to India.

ilker-aktuna, 2025-02-27:
I am using Hubitat and I don't think I can handle Zigbee OTA on Hubitat.
So my best way is to enable wifi & disable zigbee. Then update through wifi and restart.
How can I disable zigbee ?
pending zigbee task is mentioned above.
how can I do that ?

P-R-O-C-H-Y, 2025-02-27:
There is currently no "safe way" or API to disable Zigbee, the feature request for stopping Zigbee is in the esp-zigbee-sdk repository (https://github.com/espressif/esp-zigbee-sdk/issues/561). I did not test to stop the Zigbee by stopping the task and retuning it again. If you want to go this way take a look to the [docs for FreeRTOS Tasks](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/freertos_idf.html#tasks).

What I can suggest you to use a C6 to provide a radio to any other SoC (ESP32, S2, S3, C3) so you can use both Wifi and Zigbee. Take a look at the [Zigbee gateway example](https://github.com/espressif/arduino-esp32/tree/master/libraries/Zigbee/examples/Zigbee_Gateway), but you can use any example with Zigbee mode set to ROUTER or COORDINATOR, with setting the configuration as shown in the Gateway example.

ilker-aktuna, 2025-02-27:
ok. I took another way;
added a zigbee endpoint (switch) 
when this is called , I write a wifistart variable to eeprom and restart esp.
Then when it boots , if this variable is set, I start wifi (not initing zigbee). After the update , I set the eeprom variable to another value so when esp boots it does not go into the wifi loop , instead starts wifi.
this resolves my scenario. But I believe there should be an easy way to turn off zigbee radio on demand.

P-R-O-C-H-Y, 2025-02-28:
Yeah this is also a way on how to do it. When the API for stopping Zigbee will be available, I will also implement that so it can be used.

closed 2025-09-04 by me-no-dev