espressif/arduino-esp32 #10875: Zigbee HVAC endpoint support

P-R-O-C-H-Y, 2025-01-17:
### Related area

Zigbee

### Hardware specification

C6/H2

### Is your feature request related to a problem?

Add support of HVAC endpoint to the Zigbee library.

### Describe the solution you'd like

Add support of HVAC endpoint to the Zigbee library.

### Additional context

Discussed in #10869 

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

P-R-O-C-H-Y, 2025-01-17:
cc @mvermand22

MatAtBread, 2025-02-07:
I'd be interested in this too. I'm half way through building one that kind of works, but struggling to understand how to implement the `setReporting(...)` functionality for things like `runningMode`.

As far as I can see, the HVAC spec doesn't specify a `valve_position` attribute. I've implemented this as a manufacturer-specific attribute on my HVAC cluster, but this is very much a work in progress and having a working example would be great 😃

Danixu, 2025-02-07:
Just what I need. I want to convert my current ESP32 device which is controlling my A/C using ESPHome to Zigbee. I've located the used library and the only I need is to have support for HVAC.

mvermand22, 2025-02-10:
This is what I've got so far:

https://gist.github.com/mvermand22/18778bf1601e6c03f50a9cf72a3fd606
https://gist.github.com/mvermand22/4a81239b6037dcd73e3ed93d185feb63

sendReport does send ESP_ZB_ZCL_ATTR_THERMOSTAT_LOCAL_TEMPERATURE_ID, but ESP_ZB_ZCL_ATTR_THERMOSTAT_OCCUPIED_HEATING_SETPOINT_ID doesn't seem to be sent... strange.
I also need to use address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT, else if I use ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT nothing is sent too even for ESP_ZB_ZCL_ATTR_THERMOSTAT_LOCAL_TEMPERATURE_ID.

It's really still some trail & error code. I'm not a Zigbee / Arduino / C++ expert :-)
I'm trying to get it working together with Samsung SmartThings hub.

On/Off works and changing temperature setpoint works... sometimes.
Sometimes I get INVALID_VALUE and then suddenly it works for the same value.
Haven't figured that one out yet...

stevevanlare, 2026-03-31:
+1, trying to build out an ESP32-H2 Thermostat and wanted to use the Climate type in HA on the other end.