espressif/arduino-esp32 #10877: Add repeater and bridge for Zigbee

DaTTcz, 2025-01-18:
### Related area

Zigbee

### Hardware specification

ESP32-C6

### Is your feature request related to a problem?

I miss using ESP32-C6 as a repeater in Zigbee. 
And much greater using ESP32-C6 as a bridge in Zigbee. For example between Zigbee and MQTT(via LAN module) or Zigbee and RS485 MySensors.

### Describe the solution you'd like

As Integration in an existing system.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

Greginkansas, 2025-01-19:
Add these lines for a repeater

#ifndef ZIGBEE_MODE_ZCZR                        /////////////// Router here

  esp_zb_cfg_t zigbeeConfig = ZIGBEE_DEFAULT_ROUTER_CONFIG();      //////////////// Router here

P-R-O-C-H-Y, 2025-01-20:
Hi @DaTTcz,

to use the device as a router (repeater), all you need is to select this role in the call `Zigbee.begin()`:

Roles defined in the ZigbeeCore.h:
```
// enum of Zigbee Roles
typedef enum {
  ZIGBEE_COORDINATOR = 0,
  ZIGBEE_ROUTER = 1,
  ZIGBEE_END_DEVICE = 2
} zigbee_role_t;
```

Zigbee.begin call in your sketch:
`Zigbee.begin(ZIGBEE_ROUTER);`

About the second part, the bridge (gateway) is a more complex stuff. For this I currently recommend to move using `esp-zigbee-sdk` directly. This is not currently on the roadmap, but it may be added in future.

DaTTcz, 2025-01-20:
> Hi [@DaTTcz](https://github.com/DaTTcz),
>
> to use the device as a router (repeater), all you need is to select this role in the call `Zigbee.begin()`:
>
> Roles defined in the ZigbeeCore.h:
>
> ```
> // enum [...]

Thank you.

MikeBP13, 2025-02-12:
@P-R-O-C-H-Y : can you add a full example for a zigbee router? Thank you.

P-R-O-C-H-Y, 2025-02-13:
If you don't add any endpoint you cannot specify manufacturer/model and no callback for identify. 
If you need it that way, you can create your own Endpoint Class with just the basic and identify clusters.

To answer the question above about max. number of clients - do you mean children?

For that case you can use your custom config in` Zigbee.begin()` function:

```cpp
  esp_zb_cfg_t zigbeeConfig = ZIGBEE_DEFAULT_ROUTER_CONFIG();
  zigbeeConfig.nwk_cfg.zczr_cfg.max_children = 10; // 10 is default, specify it here.

  // Pass the configuration to the Zigbee.begin
  Zigbee.begin(&zigbeeConfig, false));
```

MikeBP13, 2025-02-13:
> If you don't add any endpoint you cannot specify manufacturer/model and no callback for identify. If you need it that way, you can create your own Endpoint Class with just the basic and identify clusters.
> 
> To answer the question above about max. number of clients - do you mean children?
> 
> For that case you can use your custom config in` Zigbee.begin()` function:
> 
>   esp_zb_cfg_t zigbeeConfig = ZIGBEE_DEFAULT_ROUTER_CONFIG();
>   zigbeeConfig.nwk_cfg.zczr_cfg.max_children = 10; // 10 is default, specify it here.
> 
>   // Pass the configuration to the Zigbee.begin
>   Zigbee.begin(&zigbeeConfig, false));

Thanks for the hints.
Unfortunately,  I don't have the knowledge at this point to define such a class. I'm really a beginner in Zigbee stuff... any chance you can give some pointers?

P-R-O-C-H-Y, 2025-02-13:
@MikeBP13  I have quickly created this code for the class + the sketch. Give it a try.
I did not compiled that so there might be some errors/warnings, please keep that in mind, was writing it in the GIST.

https://gist.github.com/P-R-O-C-H-Y/153c88cd7159c934c030ceccf7d4b379

Let me know if its fine or you have any issues :)
I can add this class to the Zigbee library also if that worked for you.

MikeBP13, 2025-02-13:
> [@MikeBP13](https://github.com/MikeBP13) I have quickly created this code for the class + the sketch. Give it a try. I did not compiled that so there might be some errors/warnings, please keep that [...]

Many thanks for your prompt response and help.

The header file is missing an endif at the end and the ino file is missing an include for the new class (which will not be needed when you publish the new endpoint). 

Other than that, it compiles fine and the device is recognized in HA as a router. I will leave the test device powered over night to see if it acts as a router.

Great work!

P-R-O-C-H-Y, 2025-02-14:
@MikeBP13 I am glad it worked, It was done like in 5 minutes :D 

Let me know if everything worked out so I will add this EP to the Zigbee Library :)

MikeBP13, 2025-02-14:
> [@MikeBP13](https://github.com/MikeBP13) I am glad it worked, It was done like in 5 minutes :D
> 
> Let me know if everything worked out so I will add this EP to the Zigbee Library :)

From my preliminary tests, it seems to be working fine. You can add this endpoint to the library.

Thanks!