espressif/arduino-esp32 #10881: Library for face detection Missing on Arduino core for the ESP32 V3.1

Sig2018, 2025-01-18:
### Board

ESP32s3 cam Dev Board

### Device Description

ESP32s3 cam Dev Board

### Hardware Configuration

Camera OV2640

### Version

v3.1.0

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

On arduino IDE when switch to board ESP32 version 3.1.x the files for face recognition are missing.
Running compilation will report error: 
#include "face_recognition_112_v1_s8.hpp" Not Found.
It works right on ESP32 board version 3.0.7

### Sketch

```cpp
/*
This code is based on:
1. CameraWebServer example
2. Offline version of it with face detection only (without recognition) - https://github.com/espressif/arduino-esp32/issues/9671#issuecomment-2126825897
3. Adafruit's example - https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/main/MEMENTO/Memento_Face_Detect_Recognize

This code demonstrate face detection and recognition on Xiao ESP32S3 Sense. Short blink for face detection, long blink for face reconition.
For enrolling new face you need to short D10 to GND and release. Then, you need to direct the camera to the face till long blink or Serial print indicate the face enrolled.

Notes:
1. You can use CameraWebServer example to enroll faces with web monitor
2. You can use "Erase all flash before sketch upload -> enabled" to remove enrolled faces.
3. It's recommended to enroll few different ID's from different angles for every face to make it easier to recognized.

Tools configurations for uploading: 
  a. Partition scheme: Maximum App
  b. PSRAM: OPI PSRAM

Written by Omri David
*/
#include "esp_camera.h"
#include "face_recognition_112_v1_s8.hpp"
#include "face_recognition_tool.hpp"
#include "fb_gfx.h"
[... 378 lines cut ...]
            Serial.println("Reconociendo");
            face_id = run_face_recognition(&rfb, &results);
          } else {
            face_id = 0;
            Serial.println("no enrolled faces -> not running face recognition");
            updateLED(COLOR_WAITING);  // turn the LED to wait color
            delay(500);
          }
        }
      }
  }
  // Release the framebuffer
  esp_camera_fb_return(fb);
}
```

### Debug Message

```plain
Not a debug but compilation error #include "face_recognition_112_v1_s8.hpp" Not Found.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2025-01-18:
Face recognition has been removed in Arduino Core 3.1.x
Last version where it is available is Arduino core 3.0.7

Sig2018, 2025-01-18:
Ohh really?
Any replacement or workaround?
I didn't found any reference of the removal on documentation.
@Jason2866 thank you for your reply.

me-no-dev, 2025-01-19:
The face detections and recognition library was changed and no longer supports the chips before ESP32-P4 and the old versions that do support them can not be compiled with the new versions of ESP-IDF, so we had to remove support.

Gfisk25, 2025-01-19:
So is there any way of still using it??

CongducPham, 2025-01-23:
I've read that ESP-DL v3.0 has been released and that normally it supports ESP32S3. Any plan to have back ESP-DL v3.0 in the Arduino ESP core in the future?

closed 2025-03-12 by Jason2866