espressif/arduino-esp32 #10882: Please correct the order in which the compiler flags are compiled?

mikrocoder, 2025-01-18:
### Board

Arduino Nano ESP32

### Device Description

N/A

### Hardware Configuration

N/A

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.4

### Operating System

Windows 11

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hello,

can you please correct the order in which the compiler flags are compiled?

It has always been common that you can change the compilation for yourself using platform.local.txt. 
For this you normally use **compiler.cpp.extra_flags** Unfortunately, this is not possible with the ESP32 Core Package
because you have reversed the order. Why is that?

> Normally I would recommend using the compiler.cpp.extra_flags property to inject the additional flag into the compilation command instead of compiler.cpp.flags, as compiler.cpp.extra_flags was added to the platform for exactly this sort of use case. But unfortunately the developers of the "esp32" boards platform made the poor decision to place the {compiler.cpp.extra_flags} reference before the reference to compiler.cpp.flags in the compilation command template, which makes it impossible to use compiler.cpp.extra_flags to override flags from compiler.cpp.flags (since when a flag is used multiple times, the later occurrence overrides the prior one). So compiler.cpp.extra_flags can't be used.

Also, you must have changed something in the default settings? Since Arduino ESP32 Core Package 3.1.1, C++20 is no longer standard. With Core Package 3.0.7 I did not have to change anything in the compiler flags.

Would you please change the order for the compiler flags? **Please.**

### Sketch

```cpp
N/A
```

### Debug Message

```plain
warning: use of 'auto' in parameter declaration only available with '-std=c++20' or '-fconcepts'

warning: range-based 'for' loops with initializer only available with '-std=c++20' or '-std=gnu++20' [-Wc++20-extensions]
```

### Other Steps to Reproduce

N/A

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mikrocoder, 2025-01-18:
My current workaround in the platform.local.txt is as follows.

`compiler.cpp.flags=-MMD -c “@{compiler.sdk.path}/flags/cpp_flags” -std=gnu++20 -Wno-volatile {compiler.warning_flags} {compiler.optimization_flags} {compiler.common_werror_flags}`

However, this is not the normal way and should not be done.

me-no-dev, 2025-01-19:
Since you have a use case, why not open a PR with the changes you need? We are open source and accept pull requests from the community