espressif/arduino-esp32 #10883: stdbool.h in Arduino.h missing when compiling Blink example for Nano ESP32

ednieuw, 2025-01-20:
### Board

Nano ESP32

### Device Description

Nano ESP32 connected to W11 PC compiling with IDE1.8.19 with core 3.1.1

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

IDE1.8.19 and 2.3.4

### Operating System

W11

### Flash frequency

NA

### PSRAM enabled

yes

### Upload speed

115200

### Description

Compilation was OK with core 3.0.7. 

C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/Arduino.h:23:10: fatal error: stdbool.h:
 No such file or directory
   23 | #include <stdbool.h>

### Sketch

```cpp
example Blink does not compile
```

### Debug Message

```plain
`
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\ednie\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\ednie\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\ednie\Documents\Files\Arduino\libraries -fqbn=esp32:esp32:nano_nora:PartitionScheme=default,PinNumbers=byGPIONumber,USBMode=default -vid-pid=2341_0070 -ide-version=10819 -build-path C:\Users\ednie\AppData\Local\Temp\arduino_build_678197 -warnings=all -build-cache C:\Users\ednie\AppData\Local\Temp\arduino_cache_294429 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.dfu-util.path=C:\Users\ednie\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=C:\Users\ednie\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.openocd-esp32.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20241016 -prefs=runtime.tools.openocd-esp32-v0.12.0-esp32-20241016.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20241016 -prefs=runtime.tools.xtensa-esp-elf-gdb.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\14.2_20240403 -prefs=runtime.tools.xtensa-esp-elf-gdb-14.2_20240403.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\14.2_20240403 -prefs=runtime.tools.esp32-arduino-libs.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1 -prefs=runtime.tools.esp32-arduino-libs-idf-release_v5.3-cfea4f7c-v1.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1 -prefs=runtime.tools.esp-x32.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2405 -prefs=runtime.tools.esp-x32-2405.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2405 -prefs=runtime.tools.esptool_py.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.9.dev3 -prefs=runtime.tools.esptool_py-4.9.dev3.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.9.dev3 -prefs=runtime.tools.mkspiffs.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mklittlefs.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.esp-rv32.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2405 -prefs=runtime.tools.esp-rv32-2405.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2405 -prefs=runtime.tools.riscv32-esp-elf-gdb.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\14.2_20240403 -prefs=runtime.tools.riscv32-esp-elf-gdb-14.2_20240403.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\14.2_20240403 -verbose C:\Users\ednie\Documents\Files\Arduino\Arduino_Nano-ESP32-WordClock\ESP32Arduino_WordClockV070\ESP32Arduino_WordClockV070.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\ednie\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\ednie\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\ednie\Documents\Files\Arduino\libraries -fqbn=esp32:esp32:nano_nora:PartitionScheme=default,PinNumbers=byGPIONumber,USBMode=default -vid-pid=2341_0070 -ide-version=10819 -build-path C:\Users\ednie\AppData\Local\Temp\arduino_build_678197 -warnings=all -build-cache C:\Users\ednie\AppData\Local\Temp\arduino_cache_294429 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.dfu-util.path=C:\Users\ednie\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=C:\Users\ednie\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.openocd-esp32.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20241016 -prefs=runtime.tools.openocd-esp32-v0.12.0-esp32-20241016.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20241016 -prefs=runtime.tools.xtensa-esp-elf-gdb.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\14.2_20240403 -prefs=runtime.tools.xtensa-esp-elf-gdb-14.2_20240403.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\14.2_20240403 -prefs=runtime.tools.esp32-arduino-libs.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1 -prefs=runtime.tools.esp32-arduino-libs-idf-release_v5.3-cfea4f7c-v1.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1 -prefs=runtime.tools.esp-x32.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2405 -prefs=runtime.tools.esp-x32-2405.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2405 -prefs=runtime.tools.esptool_py.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.9.dev3 -prefs=runtime.tools.esptool_py-4.9.dev3.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.9.dev3 -prefs=runtime.tools.mkspiffs.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mklittlefs.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.esp-rv32.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2405 -prefs=runtime.tools.esp-rv32-2405.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2405 -prefs=runtime.tools.riscv32-esp-elf-gdb.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\14.2_20240403 -prefs=runtime.tools.riscv32-esp-elf-gdb-14.2_20240403.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\14.2_20240403 -verbose C:\Users\ednie\Documents\Files\Arduino\Arduino_Nano-ESP32-WordClock\ESP32Arduino_WordClockV070\ESP32Arduino_WordClockV070.ino
Using board 'nano_nora' from platform in folder: C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1
Using core 'esp32' from platform in folder: C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1
cmd /c if exist "C:\\Users\\ednie\\Documents\\Files\\Arduino\\Arduino_Nano-ESP32-WordClock\\ESP32Arduino_WordClockV070\\partitions.csv" COPY /y "C:\\Users\\ednie\\Documents\\Files\\Arduino\\Arduino_Nano-ESP32-WordClock\\ESP32Arduino_WordClockV070\\partitions.csv" "C:\\Users\\ednie\\AppData\\Local\\Temp\\arduino_build_678197\\partitions.csv"
[... cut ...]
In file included from C:\Users\ednie\AppData\Local\Temp\arduino_build_678197\sketch\ESP32Arduino_WordClockV070.ino.cpp:1:
C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/Arduino.h:23:10: fatal error: stdbool.h: No such file or directory
   23 | #include <stdbool.h>
      |          ^~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Nano ESP32.

`
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-01-20:
Something must have gone wrong on your end. Blink compiles fine with 3.1.1 here

ednieuw, 2025-01-20:
Thanks for the quick reply. Will delete the core and reinstall it.

closed 2025-01-21 by ednieuw