espressif/arduino-esp32 #10888: Zigbee - error when combining multiple sensors

mwxp06, 2025-01-21:
### Board

ESP32C6 Dev Module

### Device Description

ESP32-C6-WROOM-1

### Hardware Configuration

16m flash

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS 15.2

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hallo, 
I tried to combination sensors for Temperature and humidity with Pressure from your examples

[Zigbee_Temp_Hum_Sensor_Sleepy]
[Zigbee_Pressure_Flow_Sensor]

It look like pressure sensor try to reasign temperature sensor cluster.
Can you check it, or helo i how to combinate multiple sensors to show it in Zigbee2Mqtt like 1 device with multiple values?

Thank you

### Sketch

```cpp
++
#include "Zigbee.h"

/* Zigbee temperature + humidity sensor configuration */
#define TEMP_SENSOR_ENDPOINT_NUMBER 10
#define PRESSURE_SENSOR_ENDPOINT_NUMBER 11

ZigbeeTempSensor zbTempSensor = ZigbeeTempSensor(TEMP_SENSOR_ENDPOINT_NUMBER);
ZigbeePressureSensor zbPressureSensor = ZigbeePressureSensor(PRESSURE_SENSOR_ENDPOINT_NUMBER);
...

void loop() {
...
  // Update temperature and humidity values in Temperature sensor EP
  zbTempSensor.setTemperature(temperature);
  zbTempSensor.setHumidity(humidity);

  // Report temperature and humidity values
  zbTempSensor.report();

  zbPressureSensor.setPressure(pressure_value);
  zbPressureSensor.report();
...
}

void setup() {
  ...
 // Add endpoint to Zigbee Core
  Zigbee.addEndpoint(&zbTempSensor);
  Zigbee.addEndpoint(&zbPressureSensor);

  // Create a custom Zigbee configuration for End Device with keep alive 10s to avoid interference with reporting data
  esp_zb_cfg_t zigbeeConfig = ZIGBEE_DEFAULT_ED_CONFIG();
}
```

### Debug Message

```plain
[0;31mE (261) ESP_ZIGBEE_CLUSTER: The requested add cluster ID:0x3 is already existed[0m
[0;31mE (278) ESP_ZIGBEE_ATTRIBUTE: Wrong cluster(0x0403) to add attribute, expect 0x0402[0m
[0;31mE (279) ZB OSIF: Zigbee lock is not ready![0m
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-01-22:
Hi @mwxp06, can you please post a complete reproducible sketch? Thanks

mwxp06, 2025-01-22:
Of course @P-R-O-C-H-Y :
`
#ifndef ZIGBEE_MODE_ED
#error "Zigbee end device mode is not selected in Tools->Zigbee mode"
#endif

// Core
#include "Zigbee.h"

/* Zigbee temperature + humidity sensor configuration */
#define TEMP_SENSOR_ENDPOINT_NUMBER 10
#define PRESSURE_SENSOR_ENDPOINT_NUMBER 11

#define uS_TO_S_FACTOR 1000000ULL /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP 55          /* Sleep for 55s will + 5s delay for establishing connection => data reported every 1 minute */

uint8_t button = BOOT_PIN;

ZigbeeTempSensor zbTempSensor = ZigbeeTempSensor(TEMP_SENSOR_ENDPOINT_NUMBER);
ZigbeePressureSensor zbPressureSensor = ZigbeePressureSensor(PRESSURE_SENSOR_ENDPOINT_NUMBER);

/************************ Temp sensor *****************************/
void meausureAndSleep() {
  // Measure temperature sensor value
  float temperature = 20;  //temperatureRead();

  // Use temperature value as humidity value to demonstrate both temperature and humidity
  float humidity = 80;

  // Update temperature and humidity values in Temperature sensor EP
  zbTempSensor.setTemperature(temperature);
  zbTempSensor.setHumidity(humidity);

  // Report temperature and humidity values
  zbTempSensor.report();
  Serial.printf("Reported temperature: %.2f°C, Humidity: %.2f%%\r\n", temperature, humidity);

  //////////////
  uint16_t pressure_value = (uint16_t)2345;  //*100 for demonstration so the value is in 1000-3000hPa

  Serial.printf("Updating pressure sensor value to %d hPa\r\n", pressure_value);
  zbPressureSensor.setPressure(pressure_value);
  zbPressureSensor.report();

  // Add small delay to allow the data to be sent before going to sleep
  delay(100);

  // Put device to deep sleep
  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
}

/********************* Arduino functions **************************/
void setup() {
  Serial.begin(115200);
  //Zigbee.factoryReset();

  // Init button switch
  pinMode(button, INPUT_PULLUP);

  // Configure the wake up source and set to wake up every 5 seconds
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);

  // Optional: set Zigbee device name and model
  zbTempSensor.setManufacturerAndModel("Espressif", "SleepyZigbeeTempSensorTest");

  // Set minimum and maximum temperature measurement value (10-50°C is default range for chip temperature measurement)
  zbTempSensor.setMinMaxValue(10, 50);

  // Set tolerance for temperature measurement in °C (lowest possible value is 0.01°C)
  zbTempSensor.setTolerance(1);

  // Set power source to battery and set battery percentage to measured value (now 100% for demonstration)
  // The value can be also updated by calling zbTempSensor.setBatteryPercentage(percentage) anytime
  zbTempSensor.setPowerSource(ZB_POWER_SOURCE_BATTERY, 100);

  // Add humidity cluster to the temperature sensor device with min, max and tolerance values
  zbTempSensor.addHumiditySensor(0, 100, 1);

  /////////////////////////
  // Optional: set Zigbee device name and model
  zbPressureSensor.setManufacturerAndModel("Espressif", "ZigbeePressureSensor");

  // Set minimum and maximum pressure measurement value in hPa
  zbPressureSensor.setMinMaxValue(0, 10000);

  // Optional: Set tolerance for pressure measurement in hPa
  zbPressureSensor.setTolerance(1);

  zbPressureSensor.setReporting(0, 30, 1);

  // Add endpoint to Zigbee Core
  Zigbee.addEndpoint(&zbTempSensor);
  Zigbee.addEndpoint(&zbPressureSensor);

  // Create a custom Zigbee configuration for End Device with keep alive 10s to avoid interference with reporting data
  esp_zb_cfg_t zigbeeConfig = ZIGBEE_DEFAULT_ED_CONFIG();
  zigbeeConfig.nwk_cfg.zed_cfg.keep_alive = 10000;

  // When all EPs are registered, start Zigbee in End Device mode
  if (!Zigbee.begin(&zigbeeConfig, false)) {
    Serial.println("Zigbee failed to start!");
    Serial.println("Rebooting...");
    ESP.restart();
  }
  Serial.println("Connecting to network");
  while (!Zigbee.connected()) {
    Serial.print(".");
    delay(100);
  }
  Serial.println();
  Serial.println("Successfully connected to Zigbee network");

  // Delay approx 1s (may be adjusted) to allow establishing proper connection with coordinator, needed for sleepy devices
  delay(1000);
}

void loop() {
  // Checking button for factory reset
  if (digitalRead(button) == LOW) {  // Push button pressed
    // Key debounce handling
    delay(100);
    int startTime = millis();
    while (digitalRead(button) == LOW) {
      delay(50);
      if ((millis() - startTime) > 3000) {
        // If key pressed for more than 3secs, factory reset Zigbee and reboot
        Serial.println("Resetting Zigbee to factory and rebooting in 1s.");
        delay(1000);
        Zigbee.factoryReset();
      }
    }
  }

  // Call the function to measure temperature and put the device to sleep
  meausureAndSleep();
}
`

The main reason is to create a weather station with pressure, temperature, humidity and light intensity.

rickbhs, 2025-01-27:
i had al most the same issues my solution:
Flash with Erase All Flash: Enabeld
Call Zigbee.factoryReset();
i was using Zigbee2MQTT for the network, i had to remove the device, restart Z2M,
restart the Board

P-R-O-C-H-Y, 2025-01-27:
Hi @mwxp06, thank you for reporting this. The Fix is already pending in https://github.com/espressif/arduino-esp32/pull/10863.

closed 2025-01-30 by me-no-dev

mwxp06, 2025-02-18:
Hi @P-R-O-C-H-Y its seems that problem still persist in release v 3.1.2. Can you check it pls?

D4VID0x2, 2025-03-07:
Seem to be running into the same issue on ESP32-H2-MINI-1.
Just bought it and I can't get the provided Zigbee Occupancy Sensor example to run.
It ends up in an endless restart loop with
```
[0;31mE (270) ESP_ZIGBEE_CLUSTER: The requested add cluster ID:0x3 is already existed[0m
```