espressif/arduino-esp32 #10895: Unable to send data with esp now

Gabriele-Cano, 2025-01-23:
### Board

ESP32-C3 Custom PCB

### Device Description

It's a custom PCB that I made, it's connected via I2C to MC3419 (accelerometer) and MAX17055 (battery fuel gauge). Then with "PWM" it's connected to AL1783 (led driver), and with a mosfet for running a buzzer. 

### Hardware Configuration

18 and 19 are for USB data

#define LED_R_PIN 0
#define LED_G_PIN 1
#define LED_B_PIN 4

#define SDA_PIN 5
#define SCL_PIN 6

#define ACCEL_INT_PIN 9

#define BUZZER 2
#define BATTERY_PIN 3 (to check if it's charging) 

### Version

latest master (checkout manually)

### IDE Name

VSCode, PlatformIO

### Operating System

Ubuntu 24.04.1 LTS

### Flash frequency

Crystal is 40MHz (not sure about flash freq)

### PSRAM enabled

no

### Upload speed

Changing baud rate to 460800 (upload log)

### Description

Not sure on PSRAM, and I don't know how to check framework version

Upload log that i noticed only right now:
WARNING: This chip doesn't appear to be a ESP32-C3 (chip magic value 0x4361606f). Probably it is unsupported by this version of esptool.
Chip is ESP32-C3 (revision v1.1)

I'm having a problem with ESP Now, it seems that my ESPs are able to receive data but not the send. I know that because I tried to use another ESP32 (from Waveshare) and they both received the data sent by the last one.

Both ESPs have the same code i wrote here. When i run the code (I checked that the other esp was turned on) the output is 
Data sent
Send Status: Fail

and so on for ever.

I tried to change code but nothing seems to be working 

### Sketch

```cpp
#include <esp_now.h>
#include <WiFi.h>

uint8_t peerAddress[] = {0xD8, 0x3B, 0xDA, 0xE3, 0x1C, 0xCC}; // Replace with your peer's MAC

void onDataSent(const uint8_t *mac_addr, esp_now_send_status_t status) {
  Serial.print("Send Status: ");
  Serial.println(status == ESP_NOW_SEND_SUCCESS ? "Success" : "Fail");
}

void onDataRecv(const uint8_t *mac, const uint8_t *data, int len) {
  Serial.print("Data received: ");
  Serial.write(data, len);
  Serial.println();
}

void setup() {
  Serial.begin(115200);
  
  WiFi.mode(WIFI_STA);
  Serial.println(WiFi.macAddress());
  if (esp_now_init() != ESP_OK) {
    Serial.println("Error initializing ESP-NOW");
    return;
  }
  esp_now_register_send_cb(onDataSent);
  esp_now_register_recv_cb(onDataRecv);

  esp_now_peer_info_t peerInfo = {};
  memcpy(peerInfo.peer_addr, peerAddress, 6);
  peerInfo.channel = 0;
  peerInfo.encrypt = false;
  if (esp_now_add_peer(&peerInfo) != ESP_OK) {
    Serial.println("Failed to add peer");
    return;
  }
}

void loop() {
  const char *message = "Hello ESP-NOW";
  esp_err_t result = esp_now_send(peerAddress, (uint8_t *)message, strlen(message));
  if (result == ESP_OK) {
    Serial.println("Data sent");
  } else {
    Serial.println("Send Error");
  }
  delay(2000);
}
```

### Debug Message

```plain
Not sure on how to do that with PlatformIO (The program doesn't crash)
```

### Other Steps to Reproduce

I tried another OS with macOS Ventura I think, but the problem remains the same. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-01-23:
The version of this core that PlatformIO supports is no longer supported. That is also true for the underlying ESP-IDF. Please update to using `pioarduino` instead and also look into using our own ESP_NOW library. The code above is ESP-IDF

Gabriele-Cano, 2025-01-23:
I installed Arduino IDE, with the latest version of the framework (3.1.1)

I uploaded the example code for ESP NOW that is called ESP_NOW_Broadcast_Slave and the Master on the other ESP. 

This is the output from the Master using Verbose debug level 

`[    69][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Dein=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-C3
  Package           : 0
  Revision          : 1.01
  Cores             : 1
  CPU Frequency     : 160 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000012
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   298100 B ( 291.1 KB)
  Free Bytes        :   263924 B ( 257.7 KB)
  Allocated Bytes   :    30272 B (  29.6 KB)
  Minimum Free Bytes:   259112 B ( 253.0 KB)
  Largest Free Block:   122868 B ( 120.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : DIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Jan 23 2025 15:49:06
  Compile Host OS   : linux
  ESP-IDF Version   : v5.3.2-282-gcfea4f7c98-dirty
  Arduino Version   : 3.1.1
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32C3_DEV
  Arduino Variant   : esp32c3
  Arduino FQBN      : esp32:esp32:esp32c3:UploadSpeed=921600,CDCOnBoot=cdc,CPUFreq=160,FlashFreq=80,FlashMode=dio,FlashSize=4M,PartitionScheme=default,DebugLevel=verbose,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default
============ Before Setup End ============
[   746][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 18 already has type USB_DM (38) with bus 0x3fc92fac
[   746][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 19 already has type USB_DP (39) with bus 0x3fc92fac
[   753][V][NetworkEvents.cpp:113] _checkForEvent(): Network Event: 101 - WIFI_READY
[   793][V][STA.cpp:186] _onStaEvent(): STA Started
[   794][V][NetworkEvents.cpp:113] _checkForEvent(): Network Event: 110 - STA_START
[   794][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 110 - STA_START
ESP-NOW Example - Broadcast Master
Wi-Fi parameters:
[... 103 lines cut ...]
ESP-NOW Example - Broadcast Slave
Wi-Fi parameters:
  Mode: STA
  MAC Address: D8:3B:DA:E2:64:EC
  Channel: 6
Setup complete. Waiting for a master to broadcast a message...
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   298116 B ( 291.1 KB)
  Free Bytes        :   206188 B ( 201.4 KB)
  Allocated Bytes   :    85512 B (  83.5 KB)
  Minimum Free Bytes:   206040 B ( 201.2 KB)
  Largest Free Block:   114676 B ( 112.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
    18 : USB_DM
    19 : USB_DP
    20 : UART_RX[0]
    21 : UART_TX[0]
============ After Setup End =============
`

me-no-dev, 2025-01-23:
@P-R-O-C-H-Y PTAL

Gabriele-Cano, 2025-01-23:
I don't know if it can be useful but I tried with an older custom PCB very similar to the one it's not working.

(the previous revision is actually sending data, the latest one doesn't send but still can receive)

I noticed that the chip of the older one is rev 0.4 (ESP32-C3FH4) while the new one is rev 1.1 (ESP32-C3FH4X). 

I know it shouldn't matter but this is the only hint that I have. 

(I can send schematic if needed, but I don't know how it could be a hardware problem)