espressif/arduino-esp32 #10898: Support for ZigbeeButton

mvermand22, 2025-01-24:
### Related area

Zigbee 

### Hardware specification

ESP32-H2, ESP32-C6

### Is your feature request related to a problem?

I would like a ZigbeeButton endpoint that allows to get notified of button state change requests from the hub and is also able to report back any change in state that is imposed locally (physical button pressed).

I tried to build it myself, starting from the ZigbeeDimmableLight endpoint implementation, but I can't get the code to report a local change to the Zigbee hub. It simply does not send anything back.

### Describe the solution you'd like

I created this so far:

**ZigbeeButton.h**

```
/* Class of Zigbee On/Off Button endpoint inherited from common EP class */

#pragma once

#include "soc/soc_caps.h"
#include "sdkconfig.h"
#if SOC_IEEE802154_SUPPORTED && CONFIG_ZB_ENABLED

#include "ZigbeeEP.h"
#include "ha/esp_zigbee_ha_standard.h"

/**
 * @brief Zigbee HA standard button device clusters.
 *        Added here as not supported by ESP Zigbee library.
 *
 *
 */
typedef struct zigbee_button_cfg_s {
  esp_zb_basic_cluster_cfg_t basic_cfg;       /*!<  Basic cluster configuration, @ref esp_zb_basic_cluster_cfg_s */
  esp_zb_identify_cluster_cfg_t identify_cfg; /*!<  Identify cluster configuration, @ref esp_zb_identify_cluster_cfg_s */
  esp_zb_groups_cluster_cfg_t groups_cfg;     /*!<  Groups cluster configuration, @ref esp_zb_groups_cluster_cfg_s */
  esp_zb_scenes_cluster_cfg_t scenes_cfg;     /*!<  Scenes cluster configuration, @ref esp_zb_scenes_cluster_cfg_s */
  esp_zb_on_off_cluster_cfg_t on_off_cfg;     /*!<  On off cluster configuration, @ref esp_zb_on_off_cluster_cfg_s */
} zigbee_button_cfg_t;

/**
 * @brief Zigbee HA standard button device default config value.
 *        Added here as not supported by ESP Zigbee library.
 *
 */
// clang-format off
#define ZIGBEE_DEFAULT_BUTTON_CONFIG()                                  \
  {                                                                             \
    .basic_cfg =                                                                \
      {                                                                         \
        .zcl_version = ESP_ZB_ZCL_BASIC_ZCL_VERSION_DEFAULT_VALUE,              \
        .power_source = ESP_ZB_ZCL_BASIC_POWER_SOURCE_DEFAULT_VALUE,            \
      },                                                                        \
    .identify_cfg =                                                             \
      {                                                                         \
        .identify_time = ESP_ZB_ZCL_IDENTIFY_IDENTIFY_TIME_DEFAULT_VALUE,       \
      },                                                                        \
    .groups_cfg =                                                               \
      {                                                                         \
        .groups_name_support_id = ESP_ZB_ZCL_GROUPS_NAME_SUPPORT_DEFAULT_VALUE, \
      },                                                                        \
    .scenes_cfg =                                                               \
      {                                                                         \
        .scenes_count = ESP_ZB_ZCL_SCENES_SCENE_COUNT_DEFAULT_VALUE,            \
[... 255 lines cut ...]
    }
    
    Serial.println("Reporting button state...");
    zbButton.setButtonState(false);
    delay(500);
    zbButton.reportButton();
    delay(2000);
    zbButton.setButtonState(true);
    delay(500);
    zbButton.reportButton();
  }
  delay(100);
}

```

It compiles and the ESP32-H2 module gets notified from messages sent from the hub (handled in "setButton"), but when I call zbButton.setButtonState(true); and zbButton.reportButton(); then nothing is sent back to the hub.

Any idea what I am doing wrong?

I also added a ZigbeeTempSensor and that nicely reports back the temperature to the hub

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

mvermand22, 2025-01-24:
@P-R-O-C-H-Y could you please have a quick look?
I created this ZigbeeButton based on your other Endpoints.
I just don't understand why ZigbeeTempSensor sends measurement data to the Hub and this ZigbeeButton does not.
I think I just must have made a little config mistake somewhere?

mvermand22, 2025-01-24:
I changed the reportButton to the following, and it now works, though I'm not sure this is the optimal way:

```
void ZigbeeButton::reportButton() {
  /* Send report attributes command */
  /*
  esp_zb_zcl_report_attr_cmd_t report_attr_cmd;
  report_attr_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
  report_attr_cmd.attributeID = ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID;
  report_attr_cmd.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI;
  report_attr_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
  report_attr_cmd.zcl_basic_cmd.src_endpoint = _endpoint;

  esp_zb_lock_acquire(portMAX_DELAY);
  esp_zb_zcl_report_attr_cmd_req(&report_attr_cmd);
  esp_zb_lock_release();
*/

  esp_zb_zcl_on_off_cmd_t cmd_req;
  cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
  // cmd_req.dst_address.addr_short = /* Hub's short address */;
  cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = 0;
  cmd_req.zcl_basic_cmd.dst_endpoint = 1;
  cmd_req.zcl_basic_cmd.src_endpoint = _endpoint;
  if (_current_state) {
    cmd_req.on_off_cmd_id = ESP_ZB_ZCL_CMD_ON_OFF_ON_ID;
  } else {
    cmd_req.on_off_cmd_id = ESP_ZB_ZCL_CMD_ON_OFF_OFF_ID;
  }
  // Send the command
  esp_zb_zcl_on_off_cmd_req(&cmd_req);
  log_v("Button state report sent");
}
```

Especially the hardcoded dst adress and endpoint...

P-R-O-C-H-Y, 2025-01-27:
Hi @mvermand22 I will take a look. For sure the hardcoded Endpoint+Address cannot exist in the Button code. You can add multiple funtion call of reportButton() so you can hardcode the endpoint/address in the sketch if its needed:
```
void reportButton();
void reportButton(uint16_t group_addr);
void reportButton(uint8_t endpoint, uint16_t short_addr);
void reportButton(uint8_t endpoint, esp_zb_ieee_addr_t ieee_addr);
```
Get inspired by the On/Off switch example.
But if you use any system like HomeAssistant I believe that sending the report without specifying the destination endpoint and address will work.

mvermand22, 2025-01-28:
Hi @P-R-O-C-H-Y thanks for your feedback. I started from the ZigbeeDimmableLight code, as the ZigbeeSwitch has a lot of things going on to bind light bulbs, which I didn't need.
Further more, the ZigbeeSwitch sends a "esp_zb_zcl_on_off_cmd_req" which reaches the SmartThings hub, but not as an attribute change. I might have a look at how to handle these command messages on the SmartThings hub.

ZigbeeDimmableLight does not send feedback to the Zigbee hub, ZigbeeSwitch and ZigbeeTempSensor do.
So I glued some code together and it works, but still I have this one issue:

ZigbeeTempSensor has a reportTemperature method which uses ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT:
It works on my esp32-h2 and sends attribute messages to my hub.

```
void ZigbeeTempSensor::reportTemperature() {
  /* Send report attributes command */
  esp_zb_zcl_report_attr_cmd_t report_attr_cmd;
  report_attr_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
  report_attr_cmd.attributeID = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID;
  report_attr_cmd.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI;
  report_attr_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT;
  report_attr_cmd.zcl_basic_cmd.src_endpoint = _endpoint;

  esp_zb_lock_acquire(portMAX_DELAY);
  esp_zb_zcl_report_attr_cmd_req(&report_attr_cmd);
  esp_zb_lock_release();
  log_v("Temperature report sent");
}
```
But when I create a similar method to send buttonState, the attribute messages are not sent.

```
void ZigbeeButton::reportButton() {
  /* Send report attributes command */
  esp_zb_zcl_report_attr_cmd_t report_attr_cmd;
  report_attr_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
  report_attr_cmd.attributeID = ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID;
  report_attr_cmd.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI;
  report_attr_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
  report_attr_cmd.zcl_basic_cmd.src_endpoint = _endpoint;
  
  esp_zb_lock_acquire(portMAX_DELAY);
  esp_zb_zcl_report_attr_cmd_req(&report_attr_cmd);
  esp_zb_lock_release();
  log_v("Button state report sent");
}
```
I need to set address_mode to ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT:

```
  report_attr_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT; 
  report_attr_cmd.attributeID = ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID;
  report_attr_cmd.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI;
  report_attr_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
  report_attr_cmd.zcl_basic_cmd.dst_addr_u.addr_short = 0;
  report_attr_cmd.zcl_basic_cmd.dst_endpoint = 1;
  report_attr_cmd.zcl_basic_cmd.src_endpoint = _endpoint;
```
Then it works and sends the attribute message.
So strange.

Full ZigbeeButton code:

```
/* Class of Zigbee On/Off Button endpoint inherited from common EP class */

#pragma once

#include "soc/soc_caps.h"
#include "sdkconfig.h"
#if SOC_IEEE802154_SUPPORTED && CONFIG_ZB_ENABLED

#include "ZigbeeEP.h"
#include "ha/esp_zigbee_ha_standard.h"
[... 182 lines cut ...]
  reporting_info.manuf_code = ESP_ZB_ZCL_ATTR_NON_MANUFACTURER_SPECIFIC;
  esp_zb_lock_acquire(portMAX_DELAY);
  esp_zb_zcl_update_reporting_info(&reporting_info);
  esp_zb_lock_release();
}

esp_zb_cluster_list_t *ZigbeeButton::zigbee_button_clusters_create(zigbee_button_cfg_t *button_cfg) {
  esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_basic_cluster_create(&button_cfg->basic_cfg);
  esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_identify_cluster_create(&button_cfg->identify_cfg);
  esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&button_cfg->on_off_cfg);

  // ------------------------------ Create cluster list ------------------------------
  esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
  esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
  esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
  esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

  return esp_zb_cluster_list;
}

#endif  // SOC_IEEE802154_SUPPORTED
```
It works, but with the requirement of address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT, which is strange.
Any ideas?
Thanks a lot for your time!