espressif/arduino-esp32 #10900: ESP NOW stops receiving data after BLE initialization

Gabriele-Cano, 2025-01-24:
### Board

ESP32-C3 custom PCB

### Device Description

It's a custom PCB that I made, it's connected via I2C to MC3419 (accelerometer). Then with "PWM" it's connected to AL1783 (led driver), and with a mosfet for running a buzzer.

### Hardware Configuration

18 and 19 are for USB data

#define LED_R_PIN 0
#define LED_G_PIN 1
#define LED_B_PIN 4

#define SDA_PIN 5
#define SCL_PIN 6

#define ACCEL_INT_PIN 9

#define BUZZER 2
#define BATTERY_PIN 3 (to check voltage)

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Ubuntu 24

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Hi, 
I'm working on a project that requires both the use of ESPNOW with other ESP32-C3 and BLE for connecting to a smartphone. 
The problem is that when BLE is initialised the device doesn't receive any data from other ESPs via ESPNOW. 

I need to use both protocols at the "same" time (I mean that I won't be sending data continuously with BLE nor with ESPNOW, just occasionally, like a few times every 2 seconds or so for ESPNOW, while it will be more rarely with BLE)

In the debug message you can see that at the end the devices tries to broadcast a message and it works since I can see it with another esp, but when this one replies back it says that there was an error and indeed the message is not received by the first one.

### Sketch

```cpp
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>
#include "ESP32_NOW.h"
#include "WiFi.h"
#include "structs.cpp"

#include <esp_mac.h>  // For the MAC2STR and MACSTR macros
[... 316 lines cut ...]
  Largest Free Block:   114676 B ( 112.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
    18 : USB_DM
    19 : USB_DP
    20 : UART_RX[0]
    21 : UART_TX[0]
============ After Setup End =============
[  3717][V][ESP32_NOW.cpp:384] send(): ff:ff:ff:ff:ff:ff, data length 1
[  3718][V][ESP32_NOW.cpp:133] _esp_now_tx_cb(): ff:ff:ff:ff:ff:ff : SUCCESS
[  3719][I][ESP32_NOW.h:78] onSent(): Message transmission to peer ff:ff:ff:ff:ff:ff successful
[  4722][V][ESP32_NOW.cpp:384] send(): ff:ff:ff:ff:ff:ff, data length 1
[  4738][V][ESP32_NOW.cpp:133] _esp_now_tx_cb(): ff:ff:ff:ff:ff:ff : SUCCESS
[  4739][I][ESP32_NOW.h:78] onSent(): Message transmission to peer ff:ff:ff:ff:ff:ff successful
[  5823][V][ESP32_NOW.cpp:384] send(): ff:ff:ff:ff:ff:ff, data length 1
[  5838][V][ESP32_NOW.cpp:133] _esp_now_tx_cb(): ff:ff:ff:ff:ff:ff : SUCCESS
[  5839][I][ESP32_NOW.h:78] onSent(): Message transmission to peer ff:ff:ff:ff:ff:ff successful
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

JoergSH, 2025-02-15:
i have the same problem on a esp32 devkit

Jason2866, 2025-03-12:
Closing since solution for the needed code change has been provided

closed 2025-03-12 by Jason2866