espressif/arduino-esp32 #10902: Upgrade from 3.0.7 to 3.1.0 and 3.1.1 causes severe crash with SPNtpClient library

ednieuw, 2025-01-25:
### Board

Arduino Nano ESP2, ESP-C3-13, ESP32-S3-Zero

### Device Description

None

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 1 and IDE 2

### Operating System

Windows 11

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

After an upgrade from core 3.0.7 to 3.1.0 and also 3.1.1 something changed that crashes the MCU of the Arduino Nano ESP32 after the NTP.begin (); . 
On the Arduino Nano there is no crash dump visible, it just loosing its COM port and the onboard LED is flasing red-green-blue. 
With the ESP-C3-13 I can get a memory dump in the serial monitor. (see below)

The NTP library is from here: [https://github.com/gmag11/ESPNtpClient](https://github.com/gmag11/ESPNtpClient)

When NTP.begin (); is removed from the code the C3 MCU also crashed with a shorter dump

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1144
load:0x403cc710,len:0xad8
load:0x403ce710,len:0x2f80
entry 0x403cc710
E (186) esp_core_dump_flash: Core dump data check failed:
Calculated checksum='7a015a5f'
Image checksum='ffffffff'
```

after commenting these lines 
```
  //  NTP.setTimeZone (TZ_Etc_UTC);
  //  NTP.begin ();

```
The ESP-C3-13 overheated and died. Maybe this last upload did not finish at all. I cannot remember. 
There was nothing attached to the ESP-C3-13.
The Espressif metal case became very hot. 

I have no idea where to look further without knowing what is changed between 3.0.7 and 3.1.0
The overheating can be a coincidence but for now I stop experimenting with this core version. 

### Sketch

```cpp
#include <Arduino.h>
#include <ESPNtpClient.h>
#include <WiFi.h>

#define SHOW_TIME_PERIOD 1000

void setup() {
    Serial.begin (115200);
    Serial.println ();
    WiFi.begin ("FRITZ!BoxEd", "eddie2506");
    NTP.setTimeZone (TZ_Etc_UTC);
    NTP.begin ();
}

void loop() {
    static int last = 0;

    if ((millis () - last) >= SHOW_TIME_PERIOD) {
        last = millis ();
        Serial.println (NTP.getTimeDateStringUs ());
    }
}
```

### Debug Message

```plain
Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403818f0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1144
load:0x403cc710,len:0xad8
load:0x403ce710,len:0x2f80
entry 0x403cc710

assert failed: udp_new /IDF/components/lwip/lwip/src/core/udp.c:1239 (Required to lock TCPIP core functionality!)
Core  0 register dump:
MEPC    : 0x4038139e  RA      : 0x4038737a  SP      : 0x3fca1420  GP      : 0x3fc8fe00  
TP      : 0x3fca1610  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000001  S1      : 0x3fca15c1  A0      : 0x3fca1484  A1      : 0x3fc922f9  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc98000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000056  S3      : 0x3fca1478  
S4      : 0x3fca1478  S5      : 0x3c0c9f60  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fca1420: 0x00000000 0x00000000 0x3c0d4ea8 0x4038de66 0x00000000 0x00000000 0x00000000 0x3fc922f8
3fca1440: 0x00000000 0x00000000 0x00000000 0x39333231 0x00000000 0x3fc92660 0x3c0d4ea8 0x3fc92b24
3fca1460: 0x3c0c9f60 0x3fc92670 0x3fca144c 0x3fc92674 0x3c0c8728 0x3fc922f8 0x00000000 0x00000000
3fca1480: 0x00000000 0x65737361 0x66207472 0x656c6961 0x75203a64 0x6e5f7064 0x2f207765 0x2f464449
3fca14a0: 0x706d6f63 0x6e656e6f 0x6c2f7374 0x2f706977 0x7069776c 0x6372732f 0x726f632f 0x64752f65
3fca14c0: 0x3a632e70 0x39333231 0x65522820 0x72697571 0x74206465 0x6f6c206f 0x54206b63 0x50495043
3fca14e0: 0x726f6320 0x75662065 0x6974636e 0x6c616e6f 0x21797469 0x00000029 0x00000005 0x00000004
3fca1500: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3c0d1000 0x420a2180
3fca1520: 0x00001800 0x00000005 0x3fc982d0 0x3fca1568 0x00000e10 0x00000000 0xffffff9f 0x00000001
3fca1540: 0x00000001 0x00000000 0x3fc9425c 0x0e354d7a 0x00000000 0x00000000 0x00000000 0x00000000
3fca1560: 0x3fc98000 0x00000000 0x3fc92ef8 0x42027a12 0x3c0c0140 0x00000000 0x3fc92ef8 0x420003f0
3fca1580: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc93000 0x4038279a
3fca15a0: 0x00000000 0x00000000 0x3fc93000 0x3fc9425c 0x00000000 0x00000000 0x3fc93000 0x0e354d7a
3fca15c0: 0x00000000 0x00000000 0x00000000 0x4200763c 0x00000000 0x00000000 0x00000000 0x40388636
3fca15e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca1600: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678
3fca1620: 0x00000168 0xabba1234 0x0000015c 0x3fca1320 0x00000009 0x3fc936c0 0x3fc936c0 0x3fca162c
3fca1640: 0x3fc936b8 0x00000018 0x00000000 0x00000000 0x3fca162c 0x00000000 0x00000001 0x3fc9f61c
3fca1660: 0x706f6f6c 0x6b736154 0x00000000 0x00000000 0x3fca1610 0x00000005 0x00000000 0x00000001
3fca1680: 0x00000000 0x3fca6944 0x42010e88 0x00000b14 0x00000000 0x3fc99064 0x3fc990cc 0x3fc99134
3fca16a0: 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x42010574 0x00000000
3fca16c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca16e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca1700: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca1720: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca1740: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca1760: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca1780: 0x00000000 0x00000000 0xbaad5678 0x00000068 0xabba1234 0x0000005c 0x00000000 0x3fca1798
3fca17a0: 0x00000000 0x00000000 0x00000000 0x3fca17b0 0xffffffff 0x3fca17b0 0x3fca17b0 0x00000000
3fca17c0: 0x3fca17c4 0xffffffff 0x3fca17c4 0x3fca17c4 0x00000001 0x00000001 0x00000000 0x9100ffff
3fca17e0: 0x00000000 0x50708bab 0x7ecb0c01 0xc413fdf9 0x5d0f14ce 0xbaad5678 0x00000168 0xabba1234
3fca1800: 0x0000015c 0x3fca1860 0x3fca1860 0x3fca1960 0x3fca195f 0x00000000 0x3fca181c 0xffffffff

ELF file SHA256: 3d88f2235
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2025-01-25:
The network part in core 3.1.x has been refactored. The NTP lib needs to be changed to work with these changes.

lbernstone, 2025-01-25:
FYI, the sntp method used in the built-in sntp client is microsecond accurate, but does not calculate round trip.  Does your app really need such precision?

ednieuw, 2025-01-25:
No, it does not need this accuracy. I will rebuilt my sketch and use the sntp method.

closed 2025-01-25 by ednieuw

Sujjiiiya, 2025-05-22:
Hello, did the above problem get solved ?
I am facing a similar error and I am using Arduino IDE 2.3.6 , the error is like this 

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x29 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x118c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc20
load:0x403cb700,len:0x30e0
entry 0x403c88b8
E (215) esp_core_dump_flash: Core dump data check failed:
Calculated checksum='48e0ac35'
Image checksum='ffffffff'

and these are my settings for the camera webserver code

[image]