espressif/arduino-esp32 #10903: ESP32-C6 - high light sleep current

michapr, 2025-01-26:
### Board

ESP32-C6FH4

### Device Description

Dev-board

### Hardware Configuration

no hardware attached in this time

### Version

v3.1.0

### IDE Name

Arduino IDE

### Operating System

Win 11

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Cannot get the device in correct light sleep mode, will get current >250uA instead of the published 35uA

In deep sleep all is working as expected (<10uA). How the deep sleep sample must be modified?

I've tried all sorts of things, but nothing has helped...

### Sketch

```cpp
/*
        _          _ ______ _____ _____
       (_)        | |  ____|  __ \_   _|
  _ __  ___  _____| | |__  | |  | || |
 | '_ \| \ \/ / _ \ |  __| | |  | || |
 | |_) | |>  <  __/ | |____| |__| || |_
 | .__/|_/_/\_\___|_|______|_____/_____|
 | |
 |_|

www.pixeledi.eu | https://links.pixeledi.eu
ESP32S3 XIAO DeepSleep | V1.0 | 03.2024

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html
*/
#include <Arduino.h>

unsigned long previousMillis = millis();
int cnt = 0;

void setDeepSleep()
{
//  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  Serial.println("Going to sleep zzZZZzz");
  Serial.end();

  //ESP.deepSleep(20 * 1e6);
	esp_sleep_enable_timer_wakeup(1000*(uint64_t)4500);
	esp_light_sleep_start();    // replaced deep sleep
}

void setup()
{
  Serial.begin(115200);
}
void loop()
{

  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= (1000 * 1))
  {
    previousMillis=currentMillis;
    cnt++;
    Serial.println(cnt);
  }

  if (cnt > 10)
  {
    cnt = 0;
    setDeepSleep();
  }
}
```

### Debug Message

```plain
no debug
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2025-03-13:
@michapr Can you please try latest actual Arduino core and if the behaviour is still there Arduino  core 3.2.0-rc2?

michapr, 2025-07-28:
Yes, it is the same for me with 3.3.0

lbernstone, 2025-07-28:
You may need to add [esp_sleep_pd_config](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#power-down-of-flash)(ESP_PD_DOMAIN_VDDSDIO, ESP_PD_OPTION_OFF). That may require a longer sleep time. If you can't guarantee that you will be in sleep for seconds (so that you can turn off the flashmem), then you won't be able to get to the lowest consumption levels.

Parsaabasi, 2025-10-29:
Hi @michapr 

is this issue still valid considering the comments above?

michapr, 2025-10-29:
Yes, it's still the same. I cannot use [esp_sleep_pd_config](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#power-down-of-flash)(ESP_PD_DOMAIN_VDDSDIO, ESP_PD_OPTION_OFF)  because need shorter sleep times.

lbernstone, 2025-10-29:
The code can't change how the power domains work. Can you please explain what would be the solution you are looking for from this issue?