espressif/arduino-esp32 #10904: ESP32S3 ULP and ADC reads

astrogene1000, 2025-01-26:
### Board

ESP32S3-Zero

### Device Description

ESP32S3-Zero
Board Setting: ESP32S3 Dev Module

### Hardware Configuration

GPIO 7 = ULP ADC 6
Voltage divider to GPIO 7
5V->10k->GPIO7->3.7k->GND

### Version

v3.0.7

### IDE Name

Arduino 1.8.14

### Operating System

W11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

If I do not attempt reading the ADC using the ULP but rather use Arduino analogRead I get expected results
In attached sketch, see
#define DO_ARDUINO_ADC

To switch from Arduino analogRead to ULP read of ADC

Results Arduino analogRead

Not ULP wakeup
ADC Value in Setup = 1624
ADC Value in Loop = 1627
ADC Value in Loop = 1620
ADC Value in Loop = 1623
...

When running the sketch with the ULP doing the read's of the ADC
Results ULP reading ADC
Not ULP wakeup
I= 0 V= 11
I= 1 V= 10
I= 2 V= 13
I= 3 V= 11
I= 4 V= 12
ULP Wakeup
Val= 18
Val= 19
Val= 19
Val= 18
Val= 19

### Sketch

```cpp
Please see attached sketch
```

### Debug Message

```plain
Output results from attached test sketch

Please see description above
....
```

### Other Steps to Reproduce

Sketch:

[ulp1-2-test.zip](https://github.com/user-attachments/files/18551037/ulp1-2-test.zip)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2025-01-27:
ULP is not supported when using Arduino.

me-no-dev, 2025-01-27:
@Jason2866 the FSM 4bit one can be used

@astrogene1000 please ask in the ESP-IDF repo. They know more about the FSM ULP and can help better

Jason2866, 2025-01-27:
Thx, for the info ULP FSM! Always forget S3 has the old FSM ULP too.

astrogene1000, 2025-01-27:
@me-no-dev I have checked the ESP-IDF repo, the attached is a direct take converted to ESP ULP Macro for loading the ULP.
All it does is read the ADC and store in slow memory then after a number of samples wakes the INO code up from deep sleep which then prints the value from slow memory.

Basically this example simplified to just wake periodically versus on thresholds
[https://github.com/espressif/esp-idf/tree/f00c78b020b129b6a5371997be9fd1ab4cf36f70/examples/system/ulp_fsm/ulp_adc](url)

It is working, just the values out of the ADC are way out of what is expected.

The ULP Macro language is shown here
[https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/system/ulp_macros.html](url)

My 'workaround' in the actual application is to just read in the INO code in order to BLE advertise to BThome the battery voltage and relegate the ULP to just pulse counting's for the rest of the features that get advertised.

At first thought it may be the power domain's being shut down as BLE enabled/disabled programatically as it goes into/out of deep sleep for power consumption so then just pared app down to the attached which basically does not much of anything.

Gene.

astrogene1000, 2025-01-28:
Seems the doc's do not match up with the MACRO definitions for ULP
From S3 macro doc:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/system/ulp_macros.html
/////////////////////////
I_ADC(reg_dest, adc_idx, pad_idx)
Perform ADC measurement and store result in reg_dest.

adc_idx selects ADC (0 or 1). pad_idx selects ADC pad (0 - 7)
/////////////////////////

But looking at the macro itself, it auto adds + 1 to the pad
#define I_ADC(reg_dest, adc_idx, pad_idx) { .adc = {\
        .dreg = reg_dest, \
        .mux = pad_idx + 1, \
        .sar_sel = adc_idx, \
        .unused1 = 0, \
        .cycles = 0, \
        .unused2 = 0, \
        .opcode = OPCODE_ADC } }

Compared to IDF assembler for the ULP, From IDF example:
#define EXAMPLE_ADC_CHANNEL     6 // ADC_CHANNEL_6, GPIO34 on ESP32, GPIO7 on ESP32-S3

adc r1, 0, adc_channel + 1

closed 2025-01-28 by astrogene1000