espressif/arduino-esp32 #10905: Some example code for Network interface

CelliesProjects, 2025-01-26:
### Related area

Network

### Hardware specification

All esp32

### Is your feature request related to a problem?

I would like to see some examples for https://github.com/espressif/arduino-esp32/tree/master/libraries/Network

### Describe the solution you'd like

Looking at the code I can figure out some stuff, but a couple of examples would be nice.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2025-01-27:
`WiFi.STA`, `WiFi.AP`, `ETH` and `PPP` all implement the `NetworkInterface` class. It's the glue between Arduino and LwIP. It's not really usable on it's own. One thing that can be called is the `NetworkManager` to let's say attach an event listener (because it implements the `NetworkEvents`) or set the default outgoing interface (STA, ETH or PPP)

CelliesProjects, 2025-01-28:
@me-no-dev Thanks for the explanation! 

A more concrete question is about this code that does not compile since 3.0.x

```c++
inline __attribute__((always_inline)) bool
ESP32_VS1053_Stream::_networkIsActive()
{
    for (int i = TCPIP_ADAPTER_IF_STA; i < TCPIP_ADAPTER_IF_MAX; i++)
        if (tcpip_adapter_is_netif_up((tcpip_adapter_if_t)i))
            return true;
    return false;
}
```

How would I do this with 3.x.x?
 
For now I changed this to something like `return WiFi.isConnected()` but I would like to be able to check all the interfaces.

closed 2025-01-28 by CelliesProjects

me-no-dev, 2025-01-28:
Yes please use our new API. The reason why your code does not work is that `tcpip_adapter_*` API is no longer present in ESP-IDF. It has been replaced by `esp_netif` APIs and our new network layer is based on that.  You can use `bool WiFi.STA.connected();`, `bool WiFi.AP.connected();`, `bool ETH.connected();` and `bool PPP.connected();`

me-no-dev, 2025-01-28:
Instead of `.connected()` you can use `.linkUp()` if you just want to see if the interface is UP (not necessarily connected)