espressif/arduino-esp32 #10907: internal pulldown during deep sleep on esp32c6 broken since version 3.1.0 (works fine on 3.0.7)

illusionmanager, 2025-01-26:
### Board

esp32c6

### Device Description

its the xiao seeed studio esp32c6 board

### Hardware Configuration

nothing connected but want to use GPIO_7 (same problem with GPIO_5)

### Version

latest master (checkout manually)

### IDE Name

arduino ide

### Operating System

windows 10

### Flash frequency

?

### PSRAM enabled

yes

### Upload speed

115200

### Description

I wanted to use the internal pull down on GPIO_7 during deep sleep and wake up when it becomes high. It doesn't work and wakes up automatically (probably from the 50hz mains noise) I tried lots of different options, none worked. Strangely enough it worked for someone else in the forum. After a long time it was discovered that it works fine in version 3.0.7 while I was using version 3.1.1. I switch to 3.0.7 and it works. I switched to 3.1.0 and it fails. switched back to 3.0.7 and it works.

### Sketch

```cpp
#include "driver/rtc_io.h"

#define BUTTON_PIN_BITMASK(GPIO) (1ULL << GPIO) 
#define WAKEUP_GPIO              GPIO_NUM_7
RTC_DATA_ATTR int bootCount = 0;

void setup() {
  Serial.begin(115200);
  delay(1000);

  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));
/*
  // wake up in HIGH
  esp_sleep_enable_ext1_wakeup_io(BUTTON_PIN_BITMASK(WAKEUP_GPIO), ESP_EXT1_WAKEUP_ANY_HIGH);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  rtc_gpio_pulldown_en(WAKEUP_GPIO);
  rtc_gpio_pullup_dis(WAKEUP_GPIO);
*/

  // wake up in LOW
  esp_sleep_enable_ext1_wakeup_io(BUTTON_PIN_BITMASK(WAKEUP_GPIO), ESP_EXT1_WAKEUP_ANY_LOW);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  rtc_gpio_pulldown_dis(WAKEUP_GPIO);
  rtc_gpio_pullup_en(WAKEUP_GPIO);

  delay(200);
  Serial.printf("level %d (should be 1)\n",rtc_gpio_get_level(WAKEUP_GPIO));
  delay(1000);

  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop() {
}
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-01-27:
Can you try adding `rtc_gpio_hold_en(WAKEUP_GPIO);` after configuring the GPIO and see if that will work?

illusionmanager, 2025-01-27:
no, that didn't help. I also tried adding gpio_reset_pin() before I configured, and that didn't help.

P-R-O-C-H-Y, 2025-01-27:
@illusionmanager Working fine for me with the devkit I have.
```cpp
#include "driver/rtc_io.h"

#define BUTTON_PIN_BITMASK(GPIO) (1ULL << GPIO)  // 2 ^ GPIO_NUMBER in hex
#define WAKEUP_GPIO              GPIO_NUM_7      // Only RTC IO are allowed - ESP32 Pin example
RTC_DATA_ATTR int bootCount = 0;

void print_wakeup_reason() {
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch (wakeup_reason) {
    case ESP_SLEEP_WAKEUP_EXT0:     Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1:     Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER:    Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD: Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP:      Serial.println("Wakeup caused by ULP program"); break;
    default:                        Serial.printf("Wakeup was not caused by deep sleep: %d\n", wakeup_reason); break;
  }
}

void setup() {
  Serial.begin(115200);
  delay(1000);  //Take some time to open up the Serial Monitor

  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  //Print the wakeup reason for ESP32
  print_wakeup_reason();

  //If you were to use ext1, you would use it like
  esp_sleep_enable_ext1_wakeup_io(BUTTON_PIN_BITMASK(WAKEUP_GPIO), ESP_EXT1_WAKEUP_ANY_LOW);
  rtc_gpio_pullup_en(WAKEUP_GPIO);   // Disable PULL_UP in order to allow it to wakeup on HIGH
  rtc_gpio_pulldown_dis(WAKEUP_GPIO);  // GPIO33 is tie to GND in order to wake up in HIGH

  //Go to sleep now
  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop() {
  //This is not going to be called
}
```

P-R-O-C-H-Y, 2025-01-27:
Can you set debug level to verbose and post a complete log? Thanks

illusionmanager, 2025-01-27:
i never used the debugger, if you can quickly tell me who to create this
log for you, I'm glad to assist.

P-R-O-C-H-Y, 2025-01-27:
Just in the Arduino IDE go to Tools menu and set "Debug level" -> Verbose. Then flash/upload again the sketch to the esp32c6 as usual. In the Serial monitor you will see a huge log with all informations. So please send that (log after reboot to before going to sleep)

illusionmanager, 2025-01-27:
16:47:22.387 -> Chip Info:
16:47:22.387 -> ------------------------------------------
16:47:22.387 ->   Model             : ESP32-C6
16:47:22.387 ->   Package           : 1
16:47:22.387 ->   Revision          : 0.01
16:47:22.387 ->   Cores             : 1
16:47:22.387 ->   CPU Frequency     : 160 MHz
16:47:22.387 ->   XTAL Frequency    : 40 MHz
16:47:22.387 ->   Features Bitfield : 0x00000052
16:47:22.387 ->   Embedded Flash    : No
16:47:22.387 ->   Embedded PSRAM    : No
16:47:22.387 ->   2.4GHz WiFi       : Yes
16:47:22.387 ->   Classic BT        : No
16:47:22.387 ->   BT Low Energy     : Yes
16:47:22.387 ->   IEEE 802.15.4     : Yes
16:47:22.387 -> ------------------------------------------
16:47:22.387 -> INTERNAL Memory Info:
16:47:22.387 -> ------------------------------------------
16:47:22.387 ->   Total Size        :   470768 B ( 459.7 KB)
16:47:22.387 ->   Free Bytes        :   436776 B ( 426.5 KB)
16:47:22.387 ->   Allocated Bytes   :    27600 B (  27.0 KB)
16:47:22.387 ->   Minimum Free Bytes:   431964 B ( 421.8 KB)
16:47:22.387 ->   Largest Free Block:   401396 B ( 392.0 KB)
16:47:22.387 -> ------------------------------------------
16:47:22.387 -> Flash Info:
16:47:22.387 -> ------------------------------------------
16:47:22.387 ->   Chip Size         :  4194304 B (4 MB)
16:47:22.387 ->   Block Size        :    65536 B (  64.0 KB)
16:47:22.387 ->   Sector Size       :     4096 B (   4.0 KB)
16:47:22.387 ->   Page Size         :      256 B (   0.2 KB)
16:47:22.387 ->   Bus Speed         : 40 MHz
16:47:22.387 ->   Bus Mode          : QIO
16:47:22.387 -> ------------------------------------------
16:47:22.387 -> Partitions Info:
16:47:22.387 -> ------------------------------------------
16:47:22.387 ->                 nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
16:47:22.387 ->             otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
16:47:22.387 ->                app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
16:47:22.387 ->                app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
16:47:22.387 ->              spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
16:47:22.387 ->            coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
16:47:22.387 -> ------------------------------------------
16:47:22.387 -> Software Info:
16:47:22.387 -> ------------------------------------------
16:47:22.387 ->   Compile Date/Time : Jan 27 2025 16:45:08
16:47:22.387 ->   Compile Host OS   : windows
16:47:22.387 ->   ESP-IDF Version   : v5.3.2-282-gcfea4f7c98-dirty
16:47:22.387 ->   Arduino Version   : 3.1.1
16:47:22.387 -> ------------------------------------------
16:47:22.387 -> Board Info:
16:47:22.387 -> ------------------------------------------
16:47:22.387 ->   Arduino Board     : XIAO_ESP32C6
16:47:22.387 ->   Arduino Variant   : XIAO_ESP32C6
16:47:22.387 ->   Arduino FQBN      : esp32:esp32:XIAO_ESP32C6:UploadSpeed=921600,CDCOnBoot=cdc,CPUFreq=160,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=verbose,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default
16:47:22.387 -> ============ Before Setup End ============
16:47:22.467 -> [   549][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 12 already has type USB_DM (38) with bus 0x4080ece0
16:47:22.467 -> [   549][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 13 already has type USB_DP (39) with bus 0x4080ece0
16:47:23.467 -> Boot number: 1
16:47:23.467 -> Wakeup was not caused by deep sleep: 0
16:47:38.467 -> Going to sleep now
16:47:39.195 -> =========== Before Setup Start ===========
16:47:39.195 -> Chip Info:
16:47:39.195 -> ------------------------------------------
16:47:39.195 ->   Model             : ESP32-C6
16:47:39.195 ->   Package           : 1
16:47:39.195 ->   Revision          : 0.01
16:47:39.195 ->   Cores             : 1
16:47:39.195 ->   CPU Frequency     : 160 MHz
16:47:39.195 ->   XTAL Frequency    : 40 MHz
16:47:39.195 ->   Features Bitfield : 0x00000052
[... 94 lines cut ...]
17:00:37.743 ->                 nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
17:00:37.743 ->             otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
17:00:37.743 ->                app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
17:00:37.743 ->                app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
17:00:37.743 ->              spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
17:00:37.743 ->            coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
17:00:37.743 -> ------------------------------------------
17:00:37.743 -> Software Info:
17:00:37.743 -> ------------------------------------------
17:00:37.743 ->   Compile Date/Time : Jan 27 2025 16:59:50
17:00:37.743 ->   Compile Host OS   : windows
17:00:37.743 ->   ESP-IDF Version   : v5.1.4-972-g632e0c2a9f-dirty
17:00:37.743 ->   Arduino Version   : 3.0.7
17:00:37.743 -> ------------------------------------------
17:00:37.743 -> Board Info:
17:00:37.743 -> ------------------------------------------
17:00:37.743 ->   Arduino Board     : XIAO_ESP32C6
17:00:37.743 ->   Arduino Variant   : XIAO_ESP32C6
[... cut ...] verbose,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default
17:00:37.743 -> ============ Before Setup End ============
17:00:37.815 -> [   739][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 12 already has type USB_DM (38) with bus 0x4080dc80
17:00:37.815 -> [   739][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 13 already has type USB_DP (39) with bus 0x4080dc80
17:00:38.858 -> Boot number: 1
17:00:38.858 -> Wakeup was not caused by deep sleep: 0
17:00:53.815 -> Going to sleep now

P-R-O-C-H-Y, 2025-01-28:
Comparing your log with mine I have everything the same (chip revision, idf version). I have no clue what's wrong there, as it works for me fine with the example I provided on my C6 board. There was no change to this part of the Arduino core between 3.0.X and 3.1.X.

illusionmanager, 2025-01-28:
Well, by now I have repeatedly going between 3.1.1 (and once 3.1.0) and
3.0.7 and the behaviour is consistent. Something definitely did change
between those versions. I also tried another esp32c6 board and it shows
exactly the same behavior. Is very repeatable on my end. Of course I'm
happy that finally I discovered it works fine on 3.0.7, but it would be bad
if in all versions after that it would stop working.

P-R-O-C-H-Y, 2025-01-28:
I am trying to make it not work, but in every scenario it works for me. Can you try this code?
I have switched the order of pullup/pulldown. Give it a try and if that I still not working for you, can you try to uncomment the section above the `esp_sleep_enable_ext1_wakeup_io` and try again. Please do both tests. Thanks
```cpp
#include "driver/rtc_io.h"

#define BUTTON_PIN_BITMASK(GPIO) (1ULL << GPIO)  // 2 ^ GPIO_NUMBER in hex
#define WAKEUP_GPIO              GPIO_NUM_7      // Only RTC IO are allowed - ESP32 Pin example
RTC_DATA_ATTR int bootCount = 0;

void print_wakeup_reason() {
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch (wakeup_reason) {
    case ESP_SLEEP_WAKEUP_EXT0:     Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1:     Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER:    Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD: Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP:      Serial.println("Wakeup caused by ULP program"); break;
    default:                        Serial.printf("Wakeup was not caused by deep sleep: %d\n", wakeup_reason); break;
  }
}

void setup() {
  Serial.begin(115200);
  delay(1000);  //Take some time to open up the Serial Monitor

  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  //Print the wakeup reason for ESP32
  print_wakeup_reason();

  //rtc_gpio_init(WAKEUP_GPIO);
  //rtc_gpio_set_direction(WAKEUP_GPIO, RTC_GPIO_MODE_INPUT_OUTPUT);
  //rtc_gpio_set_level(WAKEUP_GPIO, 1);
    
  //If you were to use ext1, you would use it like
  esp_sleep_enable_ext1_wakeup_io(BUTTON_PIN_BITMASK(WAKEUP_GPIO), ESP_EXT1_WAKEUP_ANY_LOW);
  rtc_gpio_pulldown_dis(WAKEUP_GPIO); 
  rtc_gpio_pullup_en(WAKEUP_GPIO); 

  //Go to sleep now
  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop() {
  //This is not going to be called
}
```

illusionmanager, 2025-01-28:
I will in a moment. Just to make sure it is not the problem that it won't
wake up, it will, it will however do so automatically in 3.1.0 and above.

illusionmanager, 2025-01-28:
bingo, your downloaded version as is didn't work, but with uncommenting the part above it it works fine. No more automatic wakeups and the manual wake up works. Great.
Thank you.

P-R-O-C-H-Y, 2025-01-28:
Thank you for quick tests. I am closing this issue as solved.

closed 2025-01-28 by P-R-O-C-H-Y

illusionmanager, 2025-01-28:
could you explain to me why the new code works?

P-R-O-C-H-Y, 2025-01-28:
Thanks for the additional feedback and test. I am glad that using` rtc_gpio_init()` + `rtc_gpio_set_direction()` worked out.
And to answer the question, as the pin was set before to INPUT_OUTPUT, it should be fine to connect it to the GND.

illusionmanager, 2025-01-28:
> Thanks for the additional feedback and test. I am glad that using` rtc_gpio_init()` + `rtc_gpio_set_direction()` worked out. And to answer the question, as the pin was set before to INPUT_OUTPUT, it should be fine to connect it to the GND.

was it? you first defined the pin to both input and output and then you set the output level to 1. Doesn't that drive it high? connecting it directly to gnd for a wakeup, might cause a big current, a too big current perhaps.