espressif/arduino-esp32 #10910: ESP32S3 Cam

josef2600, 2025-01-28:
### Board

FREENOVE ESP32S3 CAM

### Device Description

CAMERA_MODEL_FREENOVE_ESP32S3_CAM
esp32-s3 with 16MByte flash and 8 MByte psram.

### Hardware Configuration

CAMERA_MODEL_FREENOVE_ESP32S3_CAM

### Version

latest master (checkout manually)

### IDE Name

Arduino

### Operating System

win 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

256000

### Description

hi, i am working with esp32-s3 and ov5640.
i have read a ton of info about the camera and esp32. i am going to divide my problems in separate parts:
1. the clock setting in `esp_camera_init(&config);`.
2. the unstable or unworkable camera with any clock except 20MHz.
3. the low frame rate issue.
4. the Camera is on fire! fire... fire... fire!!
5. this is a very long long novel that keeps bugging everyone for a very very long time.

first, i go to the last one! the unstable boards and cameras. sure, it is a big problem, but i didn't fined any real info about what frame can it produce. 
for you to be sure i did the code and see it and you check it if you want and i am doing it right, please check my almost simple test sketch.
**some explanation:**
- i use my SD card in 4-bit. it is irrelevant to this.
- for getting the real frame rate, i count `esp_camera_fb_get()` only if successful for 5 second and divide it by exact time. the print it.
- you can change the camera clock by `#define camera_xclk_freq_hz` . this thing only works with 20MHz clock, and if i put the clock over 20mhz, and do not change any LEDC for it, it generates higher clocks than given to it. for example, if i set it to 24MHz, the generated clock on the pin is 26.3 MHz. and that is a very wrong code. and most importantly, i do not have access to the source code in Arduino. i do have access to the original source code:
https://github.com/espressif/esp32-camera
but it is useless to me, since i cannot put it in Arduino. if any one can convert that to Arduino, it would be excellent. 
anyway, i can create any clock with LEDC in arduino up to 40MHz and accurate too. to usual work is to disable the LEDC and redefine it, then i get the exact 24MHz on the pin. but, the camera frame rate gets unstable. meaning, it constantly loosing FPS. it starts from 10, then after a minute, it gets to maybe 6 FPS. this is the clocks i tested:
```
// stable = 20Mhz - 
//works stable but slow =  10-6-8-9
// unstable all go over = 24-23-22-21
```
funny thing. i did test the `set_reg` to be sure they work, and they do work! i could read them, but to be sure i put the sensor to sleep:
`senn->set_reg(senn, 0x3008, 0xFF, 0x42); // Enter software power-down mode`
and it worked. if i put the line in it, the camera wont work `Failed to capture frame` , but without it, it works fine.
the point is, if i set the camera setting manually, either it wont work, or nothing changes. i mean i can not change the default setting in it.
but, in my tests, this is my FPS:  `@ 20mhz stable => FRAMESIZE_P_FHD = FPS: 6.74 - FRAMESIZE_HD FPS: 25.42`

so, i have a question, why the FPS is so low? what is blocking it?
and, why the PLL setting (base on the esp32-camera project for IDF) is so wrong for ov5640?
the clocking is crazy. i would imagine that is the main issue for most of the problems.
[... 256 lines cut ...]
    
    current_time = millis();
    float fps = frame_count / float (float (current_time - last_report_time)/1000);
    Serial.printf("FPS: %.2f\n", fps);
    last_report_time = current_time;
    frame_count = 0;
    fflush(file);
  }
  
}
```

### Debug Message

```plain
`Failed to capture frame`
```

### Other Steps to Reproduce

explained before.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

josef2600, 2025-01-28:
and why the same resolution but with different names, produce different FPS??? why?
the naming is wrong?

```
frame_size = FRAMESIZE_FHD  => FPS: 5.91
FRAMESIZE_P_HD => FPS: 6.74
but the same res with different name:
FRAMESIZE_HD FPS: 25.42
```
!!!!!!!!!!!!
what is going on ?
```
FRAMESIZE_P_HD,     //  720x1280
FRAMESIZE_FHD,      // 1920x1080
FRAMESIZE_HD,       // 1280x720
```

josef2600, 2025-01-28:
and as i understand, the camera driver also is mixed with LCD driver that put everything in slow motion. at least put some define to disable the LCD or make them separate. this is bad. very few people use or need the LCD. 99% don't use it. but you force it on esp32 and it slow it down.

josef2600, 2025-01-31:
so, i got the math proof that the driver is wrong.
lets say it is set to 10bit, and set the resolution of 1080p, and i am overclocking it for the peruse of test!. input clock is 15MHz, and then i read the settings from camera (calculated and given by your driver).
Active Pixels and Dummy Pixels:
For 1080p, the active resolution is 1920x1080 pixels.
Including dummy pixels, the total resolution extends to 1936x1088 pixels. per datasheet.
Frame Timing and Blanking Intervals:
Total frame timing includes both the active pixels and the blanking intervals (dummy pixels).
Horizontal Blanking Interval (HBI) and Vertical Blanking Interval (VBI) are critical for correct frame synchronization.
Horizontal Timing:
Active Pixels: 1920 pixels
Dummy Pixels: (1936 - 1920) = 16 pixels (additional horizontal pixels) per datasheet.
Total Horizontal Pixels (HTS): $$ HTS = Active\_Width + Horizontal\_Blanking $$
 the values from your driver: HTS from your settings: 2844 pixels (what??!!)
Vertical Timing:
Active Pixels: 1080 pixels
Dummy Pixels: (1088 - 1080) = 8 pixels (additional vertical pixels) per datasheet.
Total Vertical Pixels (VTS): $$ VTS = Active\_Height + Vertical\_Blanking $$
 the values from your driver: VTS from your settings: 1488 pixels (what??!!)

Pixel Clock (PCLK):
the PCLK is around 56 MHz, which aligns with the expected clock rate for 1080p at 30 FPS. its even more. 48MHz is typ. per datasheet.

Frame Timing:
Total Pixels Per Frame: $$ Total\_Pixels = HTS \times VTS $$
Example Calculation:
HTS: 2844 pixels
VTS: 1488 pixels

Frame Rate Calculation:
Frame Time: $$ Frame\_Time = \frac{Total\_Pixels}{PCLK} $$
PCLK: 56 MHz
This translates to approximately 13.22 FPS, which is too low for the requirement of 1080p. but in reality, i get exact that. 13.4 FPS. well, my PCLK is a little more.

and i have to mention this too. the clocking i am giving it is too crazy that ADC doesn't even work! the test is only to proof that the esp32-s3 speed has nothing to do with the FPS that is so low. nor the external clocking of 24MHz that you are sticking with. they are irrelevant to speed or FPS. 
you are setting the camera all wrong and the timing to read it is so long, in a nutshell, you are reading all the pixels in the 1080p, but you shouldn't. 
and nobody is here!

me-no-dev, 2025-01-31:
Please feel free to install ESP-IDF and the camera component in a project. Then you will have access to everything you are talking about. I don't mind updating the driver once you prove that what you are saying is true. You seem to have made your mind, so trying to explain why things are as they are will not help much. Maybe you have a better datasheet than the one we had. After all we are talking about sensors that are EOL from a very long time and do not get any support from the manufacturer. If you boot the camera with 16MHz XCLK, then the driver will switch into GDMA mode and that will yield different results. It's experimental and does not always work in all cases. It requires specific bit clock and so on.

josef2600, 2025-01-31:
thank you. i think i may gave you the wrong idea. i am sorry, my thoughts process are weird. mainly i get tired and mostly i get heated. but please, i have no bad intent. 
the worst thing, as you mentioned, is the datasheet. it is so bad in explanation. 
that said, i already gave you the test code. you can check it yourself. and the weird thing is `HTS: 2844 pixels` and` VTS: 1488 pixels` that driver is making for the 1080p resolution. that almost make the reading time double for the sensor. 
and about the clock, again, i have provided the code for it. it is easy to test. 
but anyway, first of, thank you for your provided code and driver. it is an amazing work. i am not trying to say otherwise. 
also, i know the IDF is way better than something called Arduino! but, Arduino is way easier to work and most codes are written for it. 
anyway, i am trying to do it with IDF. i am aware of that Arduino codes can be run in IDF, i just never worked with them and it is hard for me. 
but thank you and i will try to see if i can make it.

josef2600, 2025-02-02:
@me-no-dev , just for you! 
i have moved to espressif-ide, after 6 years of Arduino. thank you. wow! that plays another game completely! but it is very hard.
any way, base on my very very long and extensive research and tests, i got this. i know what clocks i am giving to sensor and what is its output pixel clock. using oscop. this doesn't exist on any datasheet i have seen so far. and it is the most important piece of the puzzle we all missing. i have not checked it extensively, but i am going to. i just wanted to share it before i get bored! just ignor the addresses, they might not be full or accurate! but they are in datasheet. 
```
                          +----------------+
                          |                           |
     XCLK (Input)  |  External Clock    |
   6~27 MHz      |                            |
                          +-------+--------+
                                  |
                                  |
                                  v
                          +-------+--------+
                          |                            |
                          |    Pre-Divider       |
                          |  (0x3035 [7:0])     |
                          +-------+--------+
                                  |
                                  |
                                  v
                          +-------+--------+
                          |                           |
                          |    Multiplier        |
                          |  (0x3036 [7:0])   |
                          +-------+--------+
                                  |
                                  |
                                  v
                          +-------+--------+
                          |                           |
                          |   PLL Root Div    |
                          |  (0x3037 [7:0])    |
                          +-------+--------+
                                  |
                                  |
                                  v
                          +-------+--------+
                          |                           |
                          |   PLL Control      |
                          |  (0x3108 [7:0])    |
                          +-------+--------+
                                  |
                                  |
                                  v
                          +-------+--------+
                          |                            |
                          |   System Clock     |
                          |  (48~96 MHz)     |
                          +-------+--------+
                                |    |
               +----------------+    +----------------+
               |                                        |
               |                                        |
               v                                       v
        +------+------+                       +-----+-------+
        |                        |                       |                       |
        |   SCCB Clock    |                       | Pixel Clock     |
        |                        |                       | (6~96 MHz)   |
        | (up to 400 KHz)                     +-------------+
        +-------------+                              |        
                                                              |        
                                                              v        
                                            +-------+--------+
                                            |                            |
                                            |   DVP Clock         |
                                            | (up to 96 MHz)    |
                                            +----------------+
                                                   |
                                                   |
                                                   v
                                            +-------+--------+
                                            |                            |
                                            |   ADC Clock         |
                                            | (0x303E [7:0])      |
                                            |   (6~27 MHz)      |
                                            +----------------+

```
System Clock (XCLK):
Frequency Range: 6~27 MHz
Description: The initial clock source provided to the sensor.

Pre-Divider:
Register: 0x3035
Description: Divides the input clock before feeding it to the PLL.

PLL Multiplier:
Register: 0x3036
Description: Multiplies the divided clock to achieve a higher frequency.

PLL Root Divider:
Register: 0x3037
Description: Further divides the PLL output.

PLL Control:
Register: 0x3108
Description: Final stage division of the PLL output.

System Clock:
Frequency Range: 48~96 MHz
Description: The main clock derived from the PLL settings.

ADC Clock:
Register: 0x303E
Frequency Range: 6~27 MHz
Description: Clock for the analog-to-digital converter, crucial for converting the analog signals from the sensor into digital data.

Pixel Clock:
Frequency Range: 6~96 MHz
Description: Used to clock the pixel data out of the sensor.

SCCB Clock:
Description: Clock derived from the System Clock, used for communication between the camera and the processor.
Maximum Clocking: Typically up to 400 KHz.

DVP Clock:
Description: Derived from the Pixel Clock, used for transferring video data.
Maximum Clocking: Typically up to 96 MHz.

josef2600, 2025-02-02:
and as dummy pixels, it is in datasheet, or you can count them by your fingers!

[image]

[image]

there is no 500 dummy pixel !

Jason2866, 2025-03-12:
Closing since answered. Feel free to ask to reopen if there is something left open

closed 2025-03-12 by Jason2866