espressif/arduino-esp32 #10922: Rain Maker example keeps rebooting after restart

Aditya-404, 2025-01-31:
### Board

esp32-wroom-32E 8Mb module

### Device Description

it is custom board made using esp32 wroom 32e 8MB module

### Hardware Configuration

gpio 18,21 are relay pins 
gpio 19 is rainmaker resert switch 

### Version

latest master (checkout manually)

### IDE Name

Arduino ide 2.3.4

### Operating System

ubuntu 24.04

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

the rainmaker arduino example code keeps rebooting after restarting the device. after programming everything worked fine until the device is powered off and powered on again to simulate a power failure. the device is keeps rebooting after it is powered on. partition scheme is rainmaker 4MB with on ota

### Sketch

```cpp
//This example demonstrates the ESP RainMaker with a standard Switch device.
#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"

#define DEFAULT_POWER_MODE false
const char *service_name = "PROV_SONOFF_DUALR3";
const char *pop = "123456";

// GPIO for push button
static uint8_t gpio_reset = 19;
// GPIO for switch
static uint8_t gpio_switch1 = 32;
static uint8_t gpio_switch2 = 33;
// GPIO for virtual device
static uint8_t gpio_relay1 = 18;
static uint8_t gpio_relay2 = 21;
/* Variable for reading pin status*/
bool switch_state_ch1 = true;
bool switch_state_ch2 = true;
// GPIO for link status LED
// static uint8_t gpio_led = 13;

struct LightSwitch {
  const uint8_t pin;
  bool pressed;
};

// Define the light switches for channel 1 and 2
LightSwitch switch_ch1 = {gpio_switch1, false};
LightSwitch switch_ch2 = {gpio_switch2, false};

//The framework provides some standard device types like switch, lightbulb, fan, temperature sensor.
static Switch *my_switch1 = NULL;
static Switch *my_switch2 = NULL;

// WARNING: sysProvEvent is called from a separate FreeRTOS task (thread)!
void sysProvEvent(arduino_event_t *sys_event) {
  switch (sys_event->event_id) {
    case ARDUINO_EVENT_PROV_START:
#if CONFIG_IDF_TARGET_ESP32
      Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on BLE\n", service_name, pop);
      WiFiProv.printQR(service_name, pop, "ble");
#else
      Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on SoftAP\n", service_name, pop);
      WiFiProv.printQR(service_name, pop, "softap");
#endif
      break;
    case ARDUINO_EVENT_WIFI_STA_CONNECTED:
      Serial.printf("\nConnected to Wi-Fi!\n");
      //digitalWrite(gpio_led, true);
      break;
    case ARDUINO_EVENT_PROV_INIT:         WiFiProv.disableAutoStop(10000); break;
    case ARDUINO_EVENT_PROV_CRED_SUCCESS: WiFiProv.endProvision(); break;
    default:                              ;
  }
}

void write_callback(Device *device, Param *param, const param_val_t val, void *priv_data, write_ctx_t *ctx) {
  const char *device_name = device->getDeviceName();
  const char *param_name = param->getParamName();

  if (strcmp(device_name, "Switch_ch1") == 0) {

    Serial.printf("Lightbulb = %s\n", val.val.b ? "true" : "false");

    if (strcmp(param_name, "Power") == 0) {
      Serial.printf("Received value = %s for %s - %s\n", val.val.b ? "true" : "false", device_name, param_name);
      switch_state_ch1 = val.val.b;
      (switch_state_ch1 == false) ? digitalWrite(gpio_relay1, LOW) : digitalWrite(gpio_relay1, HIGH);
      param->updateAndReport(val);
    }

  } else if (strcmp(device_name, "Switch_ch2") == 0) {

    Serial.printf("Switch value = %s\n", val.val.b ? "true" : "false");

    if (strcmp(param_name, "Power") == 0) {
      Serial.printf("Received value = %s for %s - %s\n", val.val.b ? "true" : "false", device_name, param_name);
      switch_state_ch2 = val.val.b;
      (switch_state_ch2 == false) ? digitalWrite(gpio_relay2, LOW) : digitalWrite(gpio_relay2, HIGH);
      param->updateAndReport(val);
    }
  }
}

// void ARDUINO_ISR_ATTR isr(void *arg) {
//   LightSwitch *s = static_cast<LightSwitch *>(arg);
//   s->pressed = true;
// }

void setup() {

  uint32_t chipId = 0;
  

  // Configure the input GPIOs
  pinMode(gpio_reset, INPUT_PULLUP);
  // pinMode(switch_ch1.pin, INPUT_PULLUP);
  // attachInterruptArg(switch_ch1.pin, isr, &switch_ch1, CHANGE);
  // pinMode(switch_ch2.pin, INPUT_PULLUP);
  // attachInterruptArg(switch_ch2.pin, isr, &switch_ch2, CHANGE);

  // Set the Relays GPIOs as output mode
  pinMode(gpio_relay1, OUTPUT);
  pinMode(gpio_relay2, OUTPUT);
  // pinMode(gpio_led, OUTPUT);
  // Write to the GPIOs the default state on booting
  digitalWrite(gpio_relay1, DEFAULT_POWER_MODE);
  digitalWrite(gpio_relay2, DEFAULT_POWER_MODE);
  //digitalWrite(gpio_led, false);

  Node my_node;
  my_node = RMaker.initNode("Sonoff Dual R3");

  //Initialize switch device
  my_switch1 = new Switch("Switch_ch1", &gpio_relay1);
  my_switch2 = new Switch("Switch_ch2", &gpio_relay2);

  if (!my_switch1 || !my_switch2) {
    return;
  }
  //Standard switch device
  my_switch1->addCb(write_callback);
  my_switch2->addCb(write_callback);

  //Add switch device to the node
  my_node.addDevice(*my_switch1);
  my_node.addDevice(*my_switch2);

  //This is optional
 // RMaker.enableOTA(OTA_USING_TOPICS);
  //If you want to enable scheduling, set time zone for your region using setTimeZone().
  //The list of available values are provided here https://rainmaker.espressif.com/docs/time-service.html
  RMaker.setTimeZone("Asia/Kolkata");
  // Alternatively, enable the Timezone service and let the phone apps set the appropriate timezone
  RMaker.enableTZService();
  RMaker.enableSchedule();
  RMaker.enableScenes();

  //Service Name
  for (int i = 0; i < 17; i = i + 8) {
    chipId |= ((ESP.getEfuseMac() >> (40 - i)) & 0xff) << i;
  }

  Serial.printf("\nChip ID:  %lu Service Name: %s\n", chipId, service_name);

  Serial.printf("\nStarting ESP-RainMaker\n");
  RMaker.start();

  WiFi.onEvent(sysProvEvent);  // Will call sysProvEvent() from another thread.
#if CONFIG_IDF_TARGET_ESP32
  WiFiProv.beginProvision(NETWORK_PROV_SCHEME_BLE, NETWORK_PROV_SCHEME_HANDLER_FREE_BTDM, NETWORK_PROV_SECURITY_1, pop, service_name);
#else
  WiFiProv.beginProvision(NETWORK_PROV_SCHEME_SOFTAP, NETWORK_PROV_SCHEME_HANDLER_NONE, NETWORK_PROV_SECURITY_1, pop, service_name);
#endif
}

void loop() {

  // if (switch_ch1.pressed) {
  //   Serial.printf("Switch 1 has been changed\n");
  //   switch_ch1.pressed = false;
  //   // Toggle switch 1 device state
  //   switch_state_ch1 = !switch_state_ch1;
  //   Serial.printf("Toggle State to %s.\n", switch_state_ch1 ? "true" : "false");
  //   if (my_switch1) {
  //     my_switch1->updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, switch_state_ch1);
  //   }
  //   (switch_state_ch1 == false) ? digitalWrite(gpio_relay1, LOW) : digitalWrite(gpio_relay1, HIGH);
  // } else if (switch_ch2.pressed) {
  //   Serial.printf("Switch 2 has been changed\n");
  //   switch_ch2.pressed = false;
  //   // Toggle switch 2 device state
  //   switch_state_ch2 = !switch_state_ch2;
  //   Serial.printf("Toggle State to %s.\n", switch_state_ch2 ? "true" : "false");
  //   if (my_switch2) {
  //     my_switch2->updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, switch_state_ch2);
  //   }
  //   (switch_state_ch2 == false) ? digitalWrite(gpio_relay2, LOW) : digitalWrite(gpio_relay2, HIGH);
  // }

  // Read GPIO0 (external button to reset device
  if (digitalRead(gpio_reset) == LOW) {  //Push button pressed
    Serial.printf("Reset Button Pressed!\n");
    // Key debounce handling
    delay(100);
    int startTime = millis();
    while (digitalRead(gpio_reset) == LOW) {
      delay(50);
    }
    int endTime = millis();

    if ((endTime - startTime) > 10000) {
      // If key pressed for more than 10secs, reset all
      Serial.printf("Reset to factory.\n");
      RMakerFactoryReset(2);
    } else if ((endTime - startTime) > 3000) {
      Serial.printf("Reset Wi-Fi.\n");
      // If key pressed for more than 3secs, but less than 10, reset Wi-Fi
      RMakerWiFiReset(2);
    }
  }
  delay(100);
}
```

### Debug Message

```plain
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4916
load:0x40078000,len:16436
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3524
entry 0x400805b8
[    81][I][RMaker.cpp:18] event_handler(): RainMaker Initialized.

assert failed: sntp_setoperatingmode /IDF/components/lwip/lwip/src/apps/sntp/sntp.c:730 (Operating mode must not be set while SNTP client is running)
Backtrace: 0x400825cd:0x3ffd2120 0x40096391:0x3ffd2140 0x4009c75e:0x3ffd2160 0x400ff1fc:0x3ffd2290 0x40102186:0x3ffd22b0 0x400ea8b9:0x3ffd22d0 0x4009700a:0x3ffd2300

ELF file SHA256: 955c0bc86
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-01-31:
We are aware of this issue, but it comes from the main RainMaker component and we are waiting for a fix from them.

Parsaabasi, 2026-09-15:
Dear @Aditya-404, 

Since at the moment we cannot resolve this issue, I close the case for now. 
Should you need further assistance, please reopen or create a new one. 

Kind regards,

closed 2026-09-15 by Parsaabasi