espressif/arduino-esp32 #10923: Support setting the Link Speed for Ethernet

Ing-Dom, 2025-01-31:
### Related area

(Wired) Ethernet driver

### Hardware specification

especially for LAN8720, but relevant for all PHYs with the capability of setting the LINK speed by software

### Is your feature request related to a problem?

While a lot of drivers for the LAN8720 PHY support setting the link speed manually, it is not possible within arduino-esp32.
It IS indeed possible when you are using plain IDF, but the Arduino Ethernet driver (ETH.h) hides that from the user.
One could argue why this is neccessary - in 10Base-T Ethernet, which is often sufficient for embedded devices like the ESP32, the PHY needs much less power (50mA @3,3V). While you can set you PHY to 10MBIT in general by using the boot strap pins on the phy, this can cause serious issues with newer switches that do not support 10Base-T anymore.
So the option should be in software.

### Describe the solution you'd like

I would like to have an interface in the ethernet driver to be usable like

```
ETH.setLinkSpeed(eth_speed_t ::ETH_SPEED_MAX);
ETH.setLinkSpeed(eth_speed_t ::ETH_SPEED_10M);
ETH.setLinkSpeed(eth_speed_t ::ETH_SPEED_100M);
```

and also a `ETH.getLinkSpeed();`

### Describe alternatives you've considered

I tried to bypass the Ethernetdriver like this:

 eth_speed_t speed = ETH_SPEED_10M;
    esp_eth_ioctl(ETH.handle(), ETH_CMD_S_SPEED, &speed);

but his does not work.
It is not possible when you do it before ETH.begin(), and als not afterwards.
It has to be done somewhere in the middle of ETH.begin()..

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

Avenitos, 2025-02-11:
I will support the addition of this functionality, I also encountered the fact that the standard autonegotiation mechanism does not work, it was necessary to disable Full Duplex so that the network would work normally, I added the code to the library manually and it worked, but I want a normal way of management.

I use rtl8201 by the way.

Need Add:
1. Auto-negotiation on/off
2. Speed setting
3. Duplex Mode Settings

```      // Set auto-negotiation status
      bool auto_nego_en = false; // Disable auto-negotiation      
      esp_eth_ioctl(ETH.handle(), ETH_CMD_S_AUTONEGO, &auto_nego_en);

      // Set speed
      eth_speed_t eth_speed = ETH_SPEED_10M; // Set speed to 10Mbps
      esp_eth_ioctl(ETH.handle(), ETH_CMD_S_SPEED, &eth_speed); 

      // Set duplex
      eth_duplex_t duplex = ETH_DUPLEX_HALF;
      esp_eth_ioctl(ETH.handle(), ETH_CMD_S_DUPLEX_MODE, &duplex);```

me-no-dev, 2025-03-07:
Fix is coming into 3.2.0-RC2 https://github.com/espressif/arduino-esp32/pull/11053

Ing-Dom, 2025-03-10:
thank you. thats great to hear!

closed 2025-03-10 by me-no-dev