espressif/arduino-esp32 #10924: AsyncUDP创建客户端连接ipv6地址前必须要listen固定端口，WiFiUDP也一样要先begin固定端口

cers000, 2025-01-31:
### Board

ESP32-C3 Dev Board

### Device Description

合宙ESP32C3-CORE开发板

### Hardware Configuration

Type-C 连接到电脑USB接口，其他没有连接设备

### Version

latest master (checkout manually)

### IDE Name

arduino ide

### Operating System

Win10

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

9612000

### Description

使用AsyncUDP作为客户端连接服务器时总是需要使用listen绑定端口，如果不绑定指定端口，就无法在ipv6协议栈发送数据，提示
/** Illegal value.           */
  ERR_VAL        = -6,

使用WiFiUDP（或者NetworkUDP）作为客户端连接服务器时总是需要使用begin绑定端口，如果不绑定指定端口，就无法在ipv6协议栈发送数据，提示
#define  EINVAL          22  /* Invalid argument */

当然，在ipv4环境下不会有任何错误提示，也能正常发送数据，似乎是ipv6的协议栈没有做判别处理导致调用了ipv4的接口来发送ipv6的数据导致参数错误。

### Sketch

```cpp
//AsyncUDP
IPAddress localAddress("::");
const uint16_t localUdpPort = 2333;

IPAddress sAddress("240e:368:FFFF:FFEE:6255:f9ff:fe72:d914");
const uint16_t sPort = 2333;

udp.listen(localAddress, localUdpPort);//如果不使用这一句来绑定ipv6地址和端口就会报错
udp.onPacket(onPacketCallBack);  //注册收到数据包事件

udp.connect(sAddress, sPort);
AsyncUDPMessage aUdpMsg(100);
aUdpMsg.write('c');
Serial.println(aUdpMsg.length());
int si = udp.send(aUdpMsg);
Serial.println(si);
Serial.println(udp.lastErr());

//WiFiUDP or NetworkUDP
udp.begin(localAddress, localUdpPort );//如果不使用这一句来绑定ipv6地址和端口就会报错
udp.beginPacket(sAddress, sPort);
udp.print("d");
udp.endPacket();
```

### Debug Message

```plain
/** Illegal value.           */
  ERR_VAL        = -6,

#define  EINVAL          22  /* Invalid argument */
```

### Other Steps to Reproduce

我仔细比较了一下connect和listen的实现，发现确实connect没有对ipv6做针对性处理
begin和beginPacket也一样，beginPacket没有对ipv6做针对性处理
[image]
[image]

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-01-31:
Please use English and open a new issue

closed 2025-01-31 by me-no-dev

cers000, 2025-02-01:
你可以使用翻译软件翻译我的描述，谁告诉你这是英文区的？