espressif/arduino-esp32 #10930: Lack of proper cleanup in ArduinoOTA class

cziter15, 2025-02-04:
### Board

N/A

### Device Description

N/A

### Hardware Configuration

N/A

### Version

latest master (checkout manually)

### IDE Name

pio

### Operating System

Windows 11

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hello,

I'm the author of ksIotFrameworkLib (iot library for arduino esp), working extensively with various ESP devices. Recently, I encountered a frustrating issue that led to crashes under specific conditions. The problem was challenging to pinpoint due to its random nature.

During the transition between the provisioning app and the regular operational app, I faced random exceptions:

`Guru Meditation Error: Core 0 panic'ed (LoadProhibited). Exception was unhandled.`

My code is written in modern C++, utilizing `unique_ptr` as the standard practice. After investigating, I discovered that the root cause of the crash was linked to the mDNS service (an IDF component). Since I use ArduinoOTA without any global instances enabled, this was another area I needed to explore.

And - bingo!

It turns out that the ArduinoOTAClass does not correctly clean up mDNS in its destructor. It should be calling the end() method, but it doesn't. This issue also impacts the ESP8266 framework.

https://github.com/espressif/arduino-esp32/blob/0302b4db4741ae51ff45a3b1b533175ee0c0d92c/libraries/ArduinoOTA/src/ArduinoOTA.cpp#L32

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-02-04:
Nice catch! I guess we never noticed it, because globals are enabled in Arduino by default. Fix is coming

closed 2025-02-04 by me-no-dev

cziter15, 2025-02-04:
It was really tricky (and my luck) to find the cause. Actually stacktrace was not pointing to the OTA at all.

Rather it was randomly crashing somewhere after device connected the network. The only thing I've noticed was the logline like "mdns service is already running" or something like this, but this unfortunatelly didn't get my attention earlier.