espressif/arduino-esp32 #10936: ESP32S3 Interface CDC invalid or already enabled

yunyizhi, 2025-02-05:
### Board

ESP32S3 DeV Module

### Device Description

esp32s3 DevKitc N16R8

### Hardware Configuration

just  board 
[image]

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Win11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I found in the library file USBCDC.cpp that it supports a [maximum of two CDC instances](https://github.com/espressif/arduino-esp32/blob/2f423afc4eb481bcc1e31c795643f009d184bf4f/cores/esp32/USBCDC.cpp#L28), but in reality, only one can be initialized.

### Sketch

```cpp
#include <Arduino.h>
#include "USB.h"
USBCDC cdc0(0);
USBCDC cdc1(1);

void setup()
{
    Serial.begin(115200);
    log_i("init uart");
    cdc0.begin(115200);
    log_i("init cdc0");
    cdc1.begin(115200);
    log_i("init cdc1");
    USB.begin();
    // write your initialization code here
}

void loop()
{
    cdc0.println("Hello World cdc0 ");
    cdc1.println("Hello World cdc1");
    Serial.println("Hello World uart ");
    delay(1000);
}
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1188
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbf0
load:0x403cb700,len:0x30e4
entry 0x403c88ac
[     1][D][esp32-hal-tinyusb.c:807] tinyusb_enable_interface2(): Interface CDC enabled
[     9][E][esp32-hal-tinyusb.c:788] tinyusb_enable_interface2(): Interface CDC invalid or already enabled
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 0.02
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000012
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   372244 B ( 363.5 KB)
  Free Bytes        :   332292 B ( 324.5 KB)
  Allocated Bytes   :    34728 B (  33.9 KB)
  Minimum Free Bytes:   327564 B ( 319.9 KB)
  Largest Free Block:   262132 B ( 256.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 16777216 B (16 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Feb  5 2025 18:08:27
  Compile Host OS   : windows
  ESP-IDF Version   : v5.3.2-282-gcfea4f7c98-dirty
  Arduino Version   : 3.1.1
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32S3_DEV
  Arduino Variant   : esp32s3
  Arduino FQBN      : esp32:esp32:esp32s3:UploadSpeed=921600,USBMode=default,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=debug,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default
============ Before Setup End ============
[   369][I][sketch_dec24a.ino:9] setup(): init uart
[   374][I][sketch_dec24a.ino:11] setup(): init cdc0
[   379][I][sketch_dec24a.ino:13] setup(): init cdc1
[   384][D][esp32-hal-tinyusb.c:685] tinyusb_load_enabled_interfaces(): Load Done: if_num: 2, descr_len: 75, if_mask: 0x10
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   372244 B ( 363.5 KB)
  Free Bytes        :   324264 B ( 316.7 KB)
  Allocated Bytes   :    42292 B (  41.3 KB)
  Minimum Free Bytes:   324264 B ( 316.7 KB)
  Largest Free Block:   262132 B ( 256.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
    43 : UART_TX[0]
    44 : UART_RX[0]
============ After Setup End =============
Hello World uart 
Hello World uart
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-02-05:
I think you would need to build out the USB as a composite if you are going to put two devices on it.

SuGlider, 2025-02-07:
@yunyizhi - I have looked into the code and it seems that USB CDC only supports a single interface.
We may need to change the code to `#define MAX_USB_CDC_DEVICES 1`

@me-no-dev  - Do you know anything else about it?

SuGlider, 2025-02-11:
@yunyizhi - The code actually defines a single set of CDC IN/OUT endpoints. It fixes the endpoint address and it will never work with a dual CDC port.

From https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/USBCDC.cpp#L31-L39
``` cpp
static uint16_t load_cdc_descriptor(uint8_t *dst, uint8_t *itf) {
  uint8_t str_index = tinyusb_add_string_descriptor("TinyUSB CDC");
  uint8_t descriptor[TUD_CDC_DESC_LEN] = {// Interface number, string index, EP notification address and size, EP data address (out, in) and size.
                                          TUD_CDC_DESCRIPTOR(*itf, str_index, 0x85, CFG_TUD_ENDOINT_SIZE, 0x03, 0x84, CFG_TUD_ENDOINT_SIZE)
  };
  *itf += 2;
  memcpy(dst, descriptor, TUD_CDC_DESC_LEN);
  return TUD_CDC_DESC_LEN;
}
```

It would be necessary to rewrite this part of the code to consider more than 1 CDC USB port.
It may be added to the backlog for future investigation.

SuGlider, 2025-02-11:
Based on https://esp32.com/viewtopic.php?t=24777, it is possible to enable up to 2 CDC ports and no other USB profile.
Each CDC port needs 3 endpoint from a total of 6 available HW endpoints in ESP32-S3.

Therefore, it is possible to achieve it using IDF, but not using Arduino at this time.
I'll add it as feature request for the next release.

me-no-dev, 2025-02-12:
I have delved into the usage of more than one CDC port and what I've found is that the maximum number is defined at lib compilation time and is not runtime adjustable. Currently Arduino supports only one CDC port. To resolve this:
- I'm preparing a PR to the core to add some fixes and properly support the defined max number of ports
- I'm also preparing a PR to the lib-builder in order to support two CDC ports at once. This will have a small memory penalty also, because TinyUSB pre-allocates it's CDC structures

SuGlider, 2025-02-12:
Excellent, @me-no-dev! Thanks!

closed 2025-02-13 by me-no-dev