espressif/arduino-esp32 #10951: Zigbee_On_Off_Light fails to connect immediately on ESP32C6

sierra-m, 2025-02-10:
### Board

ESP32C6

### Device Description

Seeed Studio XIAO ESP32C6

### Hardware Configuration

No

### Version

other

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

After flashing the example Zigbee_On_Off_Light.ino sketch to the device, it appears to initialize Zigbee, then almost immediately fails with `Network steering was not successful`. I expected it to connect to my open Sonoff Zigbee dongle, which I successfully detected (after like 5-10 seconds) with the Zigbee_Scan_Networks.ino sketch:
```
[  7994][V][ZigbeeCore.cpp:367] scanCompleteCallback(): Network 0: PAN ID: 0x2ac8, Permit Joining: Yes, Extended PAN ID: db:94:6e:a1:2d:85:41:56, Channel: 11, Router Capacity: Yes, End Device Capacity: Yes
```
Normally, Zigbee devices connect automatically to it, so I'm not sure why this one won't.

### Sketch

```cpp
// Copyright 2024 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at

//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/**
 * @brief This example demonstrates simple Zigbee light bulb.
 *
 * The example demonstrates how to use Zigbee library to create a end device light bulb.
 * The light bulb is a Zigbee end device, which is controlled by a Zigbee coordinator.
 *
 * Proper Zigbee mode must be selected in Tools->Zigbee mode
 * and also the correct partition scheme must be selected in Tools->Partition Scheme.
 *
[... 146 lines cut ...]
[  4499][D][ZigbeeCore.cpp:84] addEndpoint(): Endpoint: 10, Device ID: 0x0100
[  4499][D][ZigbeeCore.cpp:123] zigbeeInit(): Initialize Zigbee stack
[  4556][D][ZigbeeCore.cpp:130] zigbeeInit(): Register all Zigbee EPs in list
[  4557][I][ZigbeeCore.cpp:138] zigbeeInit(): List of registered Zigbee EPs:
[  4558][I][ZigbeeCore.cpp:140] zigbeeInit(): Device type: On/Off Light Device, Endpoint: 10, Device ID: 0x0100
[  4561][V][ZigbeeCore.cpp:352] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[  4562][I][ZigbeeCore.cpp:219] esp_zb_app_signal_handler(): Zigbee stack initialized
[  4564][I][ZigbeeCore.cpp:225] esp_zb_app_signal_handler(): Device started up in  factory-reset mode
[  4565][I][ZigbeeCore.cpp:232] esp_zb_app_signal_handler(): Start network steering
Connecting to network
.......................[  6792][I][ZigbeeCore.cpp:290] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
................................[ 10021][I][ZigbeeCore.cpp:290] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
................................[ 13249][I][ZigbeeCore.cpp:290] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
.................................[ 16477][I][ZigbeeCore.cpp:290] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
................................[ 19706][I][ZigbeeCore.cpp:290] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
......
```

### Other Steps to Reproduce

Running on board version 3.1.1, but it wasn't in the versions drop-down. The only modification I made to the sketch was changing `RBG_BUILTIN` to `LED_BUILTIN`.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-02-10:
Hi @alecthegeek, I am also using the SonOff Zigbee dongle without any issues. Can you please select to use the debug zigbee libs in the `Tools menu - Zigbee mode: Zigbee ED (end Device) - Debug ` and post a full log again?

From what I remember on other issues, the seedstudio may have a poor antenna. We will see from logs :)

ZhangKeLiang0627, 2025-02-21:
The weak antenna signal on the XIAO-ESP32C6 may be due to the antenna switch being turned off. This switch is connected to the GPIO3 pin of the ESP32C6. By default, GPIO3 is pulled high, which disables the antenna. To enable the antenna and improve signal quality, you need to set the GPIO3 pin low during program initialization.

If you are still not satisfied with the performance of the onboard antenna, the XIAO-ESP32C6 also features an external antenna interface for enhanced signal reception.

For more information, please visit: [SeeedStuido XIAO ESP32C6 Getting Started](https://wiki.seeedstudio.com/xiao_esp32c6_getting_started/).

Jason2866, 2025-03-12:
Closing since issue can not be reproduced.

closed 2025-03-12 by Jason2866