espressif/arduino-esp32 #10953: Endpoint settings

MikeBP13, 2025-02-10:
### Related area

Zigbee

### Hardware specification

ESP32C6

### Is your feature request related to a problem?

I'm missing an example / the possibility to expose endpoint settings to the controller. Example: a switch endpoint having a setting for power on behavior. This setting should be available on the controller side for user configuration.

### Describe the solution you'd like

See example above

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

P-R-O-C-H-Y, 2025-02-11:
Hi @MikeBP13, can you please send any specification or detailed informations about this? I may implements that if its something what is missing. Thanks

MikeBP13, 2025-02-11:
Please see the attached screenshot:

[image]

This is from Home Assistant. 

The first section shows some endpoints (in this case 2). This is covered by the examples posted.

The second section contains some configuration parameters which can be adjusted by the user. These are exposed by the sensor firmware to the controller. How can I add such a feature to my zigbee sensor using your Arduino port?

I'm not sure about the correct terminology, but I hope that the screenshot sheds some light on what I'm after...

P-R-O-C-H-Y, 2025-02-12:
@MikeBP13  Thank you for detailed info. I will take a look and let you know soon.

MikeBP13, 2025-04-04:
Yes, the screenshot is taken from my HA (Home Assistant) configuration. The device is a [Sonoff SNZB-06P](https://sonoff.tech/product/gateway-and-sensors/snzb-06p/) presence detection sensor.

The entities shown under the Sensors category are 2 endpoints for ambient illumination and occupancy detection.

The device permits the user to adjust several parameters (detection sensitivity, timeout, ...). These options are listed under the Configuration section.

Several other Zigbee devices offer the possibility to adjust some parameters (e.g.: turn on state for a switch - on, off or last state).

I can give you more examples / details if needed.

MikeBP13, 2025-04-04:
The device diagnostic data for the SNZB-06P:
[zha-0599336f21fabc2115916b60ef516cd6-SONOFF SNZB-06P-66f2ebdd4d537489bcb7f6f11824bd3f.json](https://github.com/user-attachments/files/19604927/zha-0599336f21fabc2115916b60ef516cd6-SONOFF.SNZB-06P-66f2ebdd4d537489bcb7f6f11824bd3f.json)

P-R-O-C-H-Y, 2025-04-04:
Thank you :) I will investigate how they are doing that so I can add the implementation if possible