espressif/arduino-esp32 #10955: esp32-c6 n8 can not work with "BluetoothSerial.h"

taoloo897, 2025-02-10:
### Board

esp32-c6

### Device Description

simple esp32-c6 wrrom1

### Hardware Configuration

simple esp32-c6 wrrom1 n8

### Version

latest master (checkout manually)

### IDE Name

arduino ide

### Operating System

windows 11

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

When i try to upload from esp32 example SerialToSerialBT 
I receive this error
C:\Users\taolo\AppData\Local\Temp\.arduinoIDE-unsaved2025110-15336-mdlic9.hcp6r\SerialToSerialBT\SerialToSerialBT.ino:19:2: error: #error Serial Port Profile for Bluetooth is not available or not enabled. It is only available for the ESP32 chip.
   19 | #error Serial Port Profile for Bluetooth is not available or not enabled. It is only available for the ESP32 chip.

With and esp32-wroom 32u it works ok!
thanks.

### Sketch

```cpp
// This example code is in the Public Domain (or CC0 licensed, at your option.)
// By Evandro Copercini - 2018
//
// This example creates a bridge between Serial and Classical Bluetooth (SPP)
// and also demonstrate that SerialBT have the same functionalities of a normal Serial
// Note: Pairing is authenticated automatically by this device

#include "BluetoothSerial.h"

String device_name = "ESP32-BT-Slave";

// Check if Bluetooth is available
#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

// Check Serial Port Profile
#if !defined(CONFIG_BT_SPP_ENABLED)
#error Serial Port Profile for Bluetooth is not available or not enabled. It is only available for the ESP32 chip.
#endif

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  SerialBT.begin(device_name);  //Bluetooth device name
  //SerialBT.deleteAllBondedDevices(); // Uncomment this to delete paired devices; Must be called after begin
  Serial.printf("The device with name \"%s\" is started.\nNow you can pair it with Bluetooth!\n", device_name.c_str());
}

void loop() {
  if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
  delay(20);
}
```

### Debug Message

```plain
C:\Users\taolo\AppData\Local\Temp\.arduinoIDE-unsaved2025110-15336-mdlic9.hcp6r\SerialToSerialBT\SerialToSerialBT.ino:19:2: error: #error Serial Port Profile for Bluetooth is not available or not enabled. It is only available for the ESP32 chip.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-02-10:
Yes, BluetoothSerial is for bluetooth classic, which is only available on esp32.  Newer models use BLE.  If you search on google, you can find libraries that will provide a similar protocol over BLE.

closed 2025-02-10 by me-no-dev

SuGlider, 2025-02-10:
@taoloo897 - Please note that in order to make BluetoothSerial to work it is necessary Bluetooth Classic profiles SPP (Serial Port Profile). This profile works with Bluetooth 2.0+EDR, Bluetooth 3.0+HS, and Bluetooth 4.0.

ESP32-C6 supports Bluetooth 5.0 (LE). Therefore, it will only run BLE (Bluetooth Low Energy) which is not compatible with the previous Bluetooth version and neither with SPP.

ESP32 (the first SoC launched in 2016) is the only one that supports SPP.

There are a few external (community) BLE Serial libraries. One of them is:
https://github.com/avinabmalla/ESP32_BleSerial

There are a few BLE 5.0 Modules, such as HC-42, in the market that could work with the ESP32-C6 in order to make a point to point communication. Another option would be using 2 ESP32-C6, one for each point, in order to achieve this communication.

This article that demonstrates the use of HC-42 and an ESP32-S3 with Serial BLE, may help.
https://embedded-things.blogspot.com/2022/05/ble-uart-communication-between-esp32-s3.html
ESP32-S3 also has BLE, therefore, it shall work for the ESP32-C6 as well.

Feel free to search for other libraries and modules that best suit your needs.

taoloo897, 2025-02-11:
Thanks a lot