espressif/arduino-esp32 #10958: any example for a 3 button 3 relays light switch ?

islamessam, 2025-02-11:
### Board

esp32 c6

### Device Description

ESP32 c6 dev kit

### Hardware Configuration

3 input
3 output
status indicator

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

WIN 11 and WIN 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

Hello,
I couldnt modify the on-off light bulb/on-off switch to create a 3 buttons 3 relays light switch
is there a way to do it ? but in a simple way on the arduino IDE ?

### Sketch

```cpp
original Code without modification
-------------------------------------
// Copyright 2024 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at

//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/**
 * @brief This example demonstrates simple Zigbee light bulb.
 *
 * The example demonstrates how to use Zigbee library to create a end device light bulb.
 * The light bulb is a Zigbee end device, which is controlled by a Zigbee coordinator.
 *
 * Proper Zigbee mode must be selected in Tools->Zigbee mode
 * and also the correct partition scheme must be selected in Tools->Partition Scheme.
 *
 * Please check the README.md for instructions and more detailed description.
 *
 * Created by Jan Procházka (https://github.com/P-R-O-C-H-Y/)
 */

#ifndef ZIGBEE_MODE_ED
#error "Zigbee end device mode is not selected in Tools->Zigbee mode"
#endif

#include "Zigbee.h"

/* Zigbee light bulb configuration */
#define ZIGBEE_LIGHT_ENDPOINT 10
uint8_t led = RGB_BUILTIN;
uint8_t button = BOOT_PIN;

ZigbeeLight zbLight = ZigbeeLight(ZIGBEE_LIGHT_ENDPOINT);

/********************* RGB LED functions **************************/
void setLED(bool value) {
  digitalWrite(led, value);
}

/********************* Arduino functions **************************/
void setup() {
  Serial.begin(115200);

  // Init LED and turn it OFF (if LED_PIN == RGB_BUILTIN, the rgbLedWrite() will be used under the hood)
  pinMode(led, OUTPUT);
  digitalWrite(led, LOW);

  // Init button for factory reset
  pinMode(button, INPUT_PULLUP);

  //Optional: set Zigbee device name and model
  zbLight.setManufacturerAndModel("Espressif", "ZBLightBulb");

  // Set callback function for light change
  zbLight.onLightChange(setLED);

  //Add endpoint to Zigbee Core
  Serial.println("Adding ZigbeeLight endpoint to Zigbee Core");
  Zigbee.addEndpoint(&zbLight);

  // When all EPs are registered, start Zigbee. By default acts as ZIGBEE_END_DEVICE
  if (!Zigbee.begin()) {
    Serial.println("Zigbee failed to start!");
    Serial.println("Rebooting...");
    ESP.restart();
  }
  Serial.println("Connecting to network");
  while (!Zigbee.connected()) {
    Serial.print(".");
    delay(100);
  }
  Serial.println();
}

void loop() {
  // Checking button for factory reset
  if (digitalRead(button) == LOW) {  // Push button pressed
    // Key debounce handling
    delay(100);
    int startTime = millis();
    while (digitalRead(button) == LOW) {
      delay(50);
      if ((millis() - startTime) > 3000) {
        // If key pressed for more than 3secs, factory reset Zigbee and reboot
        Serial.println("Resetting Zigbee to factory and rebooting in 1s.");
        delay(1000);
        Zigbee.factoryReset();
      }
    }
    // Toggle light by pressing the button
    zbLight.setLight(!zbLight.getLightState());
  }
  delay(100);
}
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-02-11:
Hi @islamessam, can you please better explain what should be the result?
1 device having 3 on/off lights?
1 device having 3 on/off switches?

islamessam, 2025-02-11:
> Hi [@islamessam](https://github.com/islamessam), can you please better explain what should be the result? 1 device having 3 on/off lights? 1 device having 3 on/off switches?

Hello,
I am expecting button one (B1) to toggle relay one (R1) , B2 to toggle R2 and B3 to toggle R3 with a status indicator 

something that could make a device like the one in the below link:

https://www.amazon.eg/-/en/Concave-Neutral-Required-gateway-Control/dp/B0D2DJY6B1

P-R-O-C-H-Y, 2025-02-12:
I have some work to be done on this to support different ways on how to do this, but for now there is a way. Pleas follow this issue as it's the same question https://github.com/espressif/arduino-esp32/issues/10867

I am closing it as a duplicate. Feel free to comment on that issue instead if you need any help. Thanks