espressif/arduino-esp32 #10966: ArduinoOTA - 'handle' quite expensive

dirkx, 2025-02-12:
### Board

All

### Device Description

All - with OTA enabled

### Hardware Configuration

All, with OTA enabled

### Version

latest master (checkout manually)

### IDE Name

any

### Operating System

any

### Flash frequency

any

### PSRAM enabled

yes

### Upload speed

any

### Description

ArduinoOTA requires a call to handle() in loop() for its house keeping. With the salient parts:

```
void ArduinoOTAClass::handle() {
    ......
    if(_udp_ota.parsePacket())
        _onRx();
    _udp_ota.flush(); // always flush, even zero length packets must be flushed.
}
```

it appears that _always_ doing a flush is relatively expensive; i.e. it takes 50-80 microseconds even when there is no packet seen. May be good to only do this when parsePacket() actually sees a packet (even if it is zerolenght or corrupted packet).

### Sketch

```cpp
#include <ArduinoOTA.h>
void setup() {
   ArduinoOTA.begin();
}
void loop() {
  ArduinoOTA.handle();
}
```

### Debug Message

```plain
na
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-03-07:
>  it takes 50-80 microseconds even when there is no packet seen.

It's hard to believe that. `clear()` deletes the buffer or does nothing if one is not available. There is nothing to take that much time, even if it has to be fetched from flash to be executed.

dirkx, 2025-03-08:
Hmm - off; so I am using a profiler like [this](https://github.com/MakerSpaceLeiden/AccesSystem/blob/a48778e2794899609c7eac32db12147733500400/lib-arduino/ACNode/src/ACNodeBase.cpp#L418):

```
   for (it =_handlers.begin(); it!=_handlers.end(); ++it) {
        unsigned long s = micros();
        (*it)->loop();
        unsigned long delta = micros() - s;
        
        if ((*it)->micros_in_loop == 0)
            (*it)->micros_in_loop = delta;

        (*it)->micros_in_loop = ((*it)->micros_in_loop * 5000 + delta)/5001;
        
        if (show)
            Debug.printf("   %12lu %08x %s\n",(*it)->micros_in_loop,(*it), (*it)->name());
    };
```
That basically does a rolling; tapered average; with the last 5000 being the bulk of the timing. This is done at a rate of a bit over 1kHz.  

The call to` loop()` is [exactly](https://github.com/MakerSpaceLeiden/AccesSystem/blob/a48778e2794899609c7eac32db12147733500400/lib-arduino/ACNode/src/OTA.cpp#L180):(

```
void OTAWithDisplay::loop() {
    ArduinoOTA.handle();
}
```

Yet I am seeing below - around 73 milliseconds spend in `ArduinoOTA.handle()`.

```
11:37:10.290:DBG: Profile (in microSeconds):
11:37:10.336:DBG:               5 3ffc46cc MachineState
11:37:10.390:DBG:               1 3ffbb150 RestAPI
11:37:10.440:DBG:               1 3ffbb1a8 ApprovalAPI
11:37:10.493:DBG:               2 3ffbb9ec ACBase
11:37:10.542:DBG:               1 3ffb8624 IODebounce
11:37:10.594:DBG:               1 3ffb86ec IODebounce
11:37:10.647:DBG:               1 3ffb8880 IODebounce
11:37:10.700:DBG:               6 3ffb8978 RFID-MFRC522
11:37:11.755:DBG:              72 3ffb8fe0 ArduinoOTA
11:37:11.811:DBG:               2 3ffcdcc0 IODebounce
11:37:11.863:DBG:               1 3ffcdd58 IODebounce
```

And if I comment this  `ArduinoOTA.handle()` the microSeconds spend drop to around 0.6 microsecond average.